dresden-elektronik/deconz-rest-plugin #6645: Tuya/Moer Zigbee Temperature Humidity Sensor _TZ3000_itnrsufe

stevel032, 2023-01-05:
## Device

- Product name: Moer Zigbee Temperature Humidity Sensor (Powered by Tuya)
- Manufacturer: _TZ3000_itnrsufe
- Model identifier: TS0201
- Device type : 
  - Sensor

## Screenshots

[image]

[image]

### Basic

[image]

### Identify

[image]

### Power Configuration

[image]

### Temperature Management

[image]

### Relative Humidity Management

[image]

BabaIsYou, 2023-01-08:
May be you can try with this DDF. But not sure about battery because your screenshot seems display 0 for attr 0x0021
```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_ywagc4rj,_TZ3000_itnrsufe",
  "modelid": "TS0201,TS0201",
  "vendor": "Moes",
  "product": "TS0201",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_TEMPERATURE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0402"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "refresh.interval": 3600,
          "read": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2",
            "fn": "zcl"
          },
          "default": 0
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/temperature",
          "awake": true,
          "default": 0
        }
      ]
    },
    {
      "type": "$TYPE_HUMIDITY_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0405"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "refresh.interval": 3600,
          "read": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2",
            "fn": "zcl"
          },
          "default": 0
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/humidity",
          "awake": true,
          "parse": {
            "at": "0x0000",
            "cl": "0x0405",
            "ep": 0,
            "eval": "Item.val = Attr.val * 10 + R.item('config/offset').val",
            "fn": "zcl"
          },
          "default": 0
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 60,
          "max": 3600,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0402",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x29",
          "min": 60,
          "max": 300,
          "change": "0x00000032"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0405",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x21",
          "min": 60,
          "max": 300,
          "change": "0x0000000A"
        }
      ]
    }
  ]
}
```

stevel032, 2023-01-09:
Thank you and I've love to try it out.

But... from the information I can find, it's a big hassle to try it. I have Home Assistant, with DeConz running as a container. There's special setup for me to gain root access first before I can look at containers, and then adjust the DeConz configuration. Do you guys know a way to try out DDF file in Home Assistant, without having root privilege?

Or is it a required step for me to try this out? if not, I'm happy to wait until you guys merge the code and issue the next release.

Thank you so very much!!

BabaIsYou, 2023-01-09:
> Or is it a required step for me to try this out? if not, I'm happy to wait until you guys merge the code and issue the next release.

As a new device someone has to test the DDF before submitting it as a Pull Request (PR) for change to be included in a future release.

May be [https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-home-assistant-using-text-editor/1839](https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-home-assistant-using-text-editor/1839) could help you ?

stevel032, 2023-01-29:
Sorry I missed this update, my apologies.

This sounds reasonable, and let me work on this and see if I can figure out a way to test this.

stevel032, 2023-02-04:
So I managed to do the test, after gaining root access to the Docker instance inside HA, and followed the instructions in the link above, but still not able to see the device in the PhosCon UI. 

First I tried to create a new device file with the content above named `/usr/share/deCONZ/devices/tuya/moes_zigbee.json`, and without restarting the docker instance (and wipe out this file). Then I went to the deConz UI (through VNC I think, provided by HA), and opened up the DDF editor, and clicked "Hot Reload".

I came back to the Phoscom UI, and would not see my new sensor.

I repeated the test above by creating a new file in the local location: `/data/.local/share/dresden-elektronik/deCONZ/devices/moes_zigbee.json`. But the result is the same.

So maybe my question is, how would I know that the system is picking up the new DDF file above? Thank you!

BabaIsYou, 2023-02-04:
My mistake, sorry.
Have to replace
  "manufacturername": "_TZ3000_ywagc4rj,_TZ3000_itnrsufe",
  "modelid": "TS0201,TS0201",
  
By
  "manufacturername": ["_TZ3000_ywagc4rj ", "_TZ3000_itnrsufe"],
  "modelid": ["TS0201 ", "TS0201"],

stevel032, 2023-02-05:
The thing still doesn't work. The DeConz UI still shows "0xA08F" in bold on top, different from other recognized devices. In fact, the DDF editor was able to read the malformed file and correct it. So I had the following:

[image]

I would like to try maybe debugging this further, and tried to figure out where the log messages are, from these files:

```
root@core-deconz:/# cd /
root@core-deconz:/# ls -lR /var/log
/var/log:
total 212
-rw-r--r-- 1 root root  10737 Jan 23 11:59 alternatives.log
drwxr-xr-x 1 root root   4096 Jan 23 11:59 apt
-rw-rw---- 1 root utmp      0 Nov 13 16:00 btmp
-rw-r--r-- 1 root root 147935 Jan 23 11:59 dpkg.log
-rw-r--r-- 1 root root   3264 Jan 23 11:57 faillog
-rw-r--r-- 1 root root   1382 Jan 23 11:56 fontconfig.log
-rw-rw-r-- 1 root utmp  30192 Jan 23 11:57 lastlog
drwxr-xr-x 1 root adm    4096 Jan 23 11:55 nginx
-rw-rw-r-- 1 root utmp      0 Nov 13 16:00 wtmp

/var/log/apt:
total 104
-rw-r--r-- 1 root root 15612 Jan 23 11:59 eipp.log.xz
-rw-r--r-- 1 root root 13799 Jan 23 11:59 history.log
-rw-r----- 1 root adm  72637 Jan 23 11:59 term.log

/var/log/nginx:
total 0
-rw-r----- 1 www-data adm 0 Jan 23 11:55 access.log
-rw-r----- 1 www-data adm 0 Jan 23 11:55 error.log
```

Ideally I wonder if there's a way to turn on the debug flag to get more messages? Or alternatively, there might be a better/proper way to debug this, I'd appreciate some pointers. Thanks!!

stevel032, 2023-02-05:
I actually figured out how to get debug messages, they are right there in the UI. It looks like my custom DDF file is recognized and loaded, but the DeConz UI still doesn't show the device name:

[image]

[image]

BabaIsYou, 2023-02-05:
Looks like something wrong elsewhere in the DDF because even if it find it the atteibuted seems not to be fully created (and listed in log after loading the DDF)

Will have a look later, sorry have no time today.

[Screenshot 2023-02-04 at 5 40 00 PM]<https://user-images.githubusercontent.com/55261196/216797319-278cabb4-445c-4a21-8457-5d477fd46ef8.png>

[Screenshot 2023-02-04 at 5 40 25 PM]<https://user-images.githubusercontent.com/55261196/216797332-d2254fb8-afa1-44b6-ac3e-78894d75bab2.png>

stevel032, 2023-02-05:
Thank you and there is no hurry...

But if you think this might be a DDF issue, I took the liberty of copying one of the existing ones (`/usr/share/deCONZ/devices/tuya/_TZ3000_i8jfiezr_temp_hum_sensor.json`) into the `/data/.local` directory (which is persistent after docker restart), and changed the manufacturer name and id, as I assume these are the two pieces of information used to match entries in the DDF.

In any case, the end result is the same, the DeCONZ UI is still not recognizing my device:

[image]

BTW, here's another quick note, my device is battery powered, and is not always connected to the Zigbee network. It seems that I have to poke the reset button briefly for it send over some data. Not sure if this affects/changes anything. Thanks a whole bunch!!

stevel032, 2023-02-05:
Success!

Without anything to lose, I restarted the docker container, my device still doesn't show up. I poked the reset button quickly one more time, and the whole thing springs to life:

[image]

Just to recap, I'm using the `_TZ3000_i8jfiezr_temp_hum_sensor.json` and only changed the "Manufacturer Name" in the beginning of the DDF file.

As you can see, there are still a bunch of problems with this DDF file:

1. The "Vendor" and "Version" fields are not correct.
2. The "Humidity" field is 1/10 of the actual value. My device it self shows 50% at the time.

We are getting very close, looking forward to getting this ironed out!!

Mimiix, 2023-02-05:
> The thing still doesn't work. The DeConz UI still shows "0xA08F" in bold on top, different from other recognized devices. In fact, the DDF editor was able to read the malformed file and correct it. [...]

Notice the silver status. Putting that on gold might help

BabaIsYou, 2023-02-05:
Thanks @Mimiix I missed that too ! ;-)

regarding to the manufacturer that’s the expected value taken from the manufacturer in the DDF. Doesn’t really matter.

for battery you can change the computed (eval) part, using DDF editor or directly in the DDF. It should look like this

Item.val = Attr.val * 10 + R.item('config/offset').val

I can look at the DDF you used at this time but will look further later.
You can multiply par 100 instead of 10 if you need it.

For version, your device seems to have no value in swversion, nor application version attributes in your screenshot then nothing to display

stevel032, 2023-02-06:
Thank you for the feedback, I was able to add the 10x factor and now I have correct humidity display like this:

[image]

The "vendor" field is still strange, even though the DDF says the vendor is "Moes", it doesn't seem to reflect to the DeCONZ UI. This is not super important I guess, but it'll just be nice if it can be fixed, see DDF snippet below:

[image]

BabaIsYou, 2023-02-06:
Vendor is taken from manufacturer then it’s correct.

[Screenshot 2023-02-05 at 4 32 16 PM]<https://user-images.githubusercontent.com/55261196/216856115-a4b12d92-816a-44fd-b5f5-37ceb5bd1215.png>

[Screenshot 2023-02-05 at 4 35 03 PM]<https://user-images.githubusercontent.com/55261196/216856256-89906a95-bc3c-454e-8711-adb6ad94d1b1.png>

stevel032, 2023-02-09:
Sorry I sat on this for a couple of more days, as my units are now working well for me (with Home Assistant integration). Are there anything I can/should do to help add this type of device into the code repo? Just to recap, here are the things I did:

1. I copied an existing DDF file into my own location (`/usr/share/deCONZ/devices/tuya/_TZ3000_i8jfiezr_temp_hum_sensor.json`) and renamed it
2. I  changed the "manufacturername" to fit my device, which is `_TZ3000_itnrsufe`
3. I adjusted the calculation of humidity level and added a x10 factor to it.

That's it!

BabaIsYou, 2023-02-09:
Thx. I did the PR to have it added in the next release.

stevel032, 2023-02-10:
That's wonderful, thank you!

I took a quick look at the PR, it seems that you did not adjust the x10 factor in the humidity sensor. It's not a big deal as I'm not using it myself. But I suspect that the original DDF was already incorrect. You might want to decide how best to handle this. Thanks again!!

BabaIsYou, 2023-02-10:
> you did not adjust the x10 factor in the humidity sensor.

factor *10 for humidity is in the DDF, do you mean that you had to do a factor *100 ?

stevel032, 2023-02-21:
Sorry no, I meant to say, compared to the original `_TZ3000_i8jfiezr_temp_hum_sensor.json` file, I added a *10 factor. In any case, I'm happy to try out the final release and let you guys know if this factor should be further adjusted, it's not a big deal.

vdiogo, 2023-03-15:
How could we know when this fix would be available? Apologies for noob Q

stevel408, 2023-03-18:
I don't know... I'm running the dev version in my instance and have been happy. I don't see any related commits. So maybe I'll just keep an eye out here and make sure it doesn't go stale again.

stevel032, 2023-03-29:
Peeking in here again to keep the issue open/alive...

stevel408, 2023-03-29:
Peeking in here to keep the issue alive...

stevel408, 2023-04-19:
Peeking again to keep this alive...

closed 2023-05-18 by github-actions[bot]