dresden-elektronik/deconz-rest-plugin #6646: Tuya ZigBee Smart Wireless Control Gas Water Valve

dsimop, 2023-01-05:
## Device

- Product name: Tuya ZigBee Smart Wireless Control Gas Water Valve
- Manufacturer: _TZ3000_iedbgyxt
- Model identifier: TS0001
- Device type : Please remove all unrelated device types. 
  - Other: Water / Gas Valve

## Screenshots

  - [image]
  - [image]

### Basic
[image]

### Identify
[image]

### Groups
[image]

### Scenes
[image]

### On/Off
[image]

dsimop, 2023-01-05:
After several minutes, the blue blinking light stops and the device gets unavailable.
If you press manually the button on the device in order to turn the valve on/off, the device gets re-paired - but after several minutes it gets disconnected as well.

dsimop, 2023-02-10:
Request to be reopened.

Mimiix, 2023-03-15:
Sure :)

@smanar this one can be added ?

Smanar, 2023-03-16:
Sure, can try this DDF to test ?

```
{
	"schema": "devcap1.schema.json",
	"manufacturername": ["_TZ3000_o4cjetlm", "_TZ3000_iedbgyxt", "_TYZB01_4tlksk8a"],
	"modelid": ["TS0001", "TS0001", "TS0001"],
	"product": "Tuya Smart water/gas valve",
	"sleeper": false,
	"status": "Gold",
	"subdevices": [{
		"type": "$TYPE_ON_OFF_OUTPUT",
		"restapi": "/lights",
		"uuid": [
			"$address.ext",
			"0x01"
		],
		"items": [{
				"name": "attr/id"
			},
			{
				"name": "attr/lastannounced"
			},
			{
				"name": "attr/lastseen"
			},
			{
				"name": "attr/manufacturername"
			},
			{
				"name": "attr/modelid"
			},
			{
				"name": "attr/name"
			},
			{
				"name": "attr/swversion",
				"parse": {
					"fn": "zcl",
					"ep": 1,
					"cl": "0x0000",
					"at": "0x0001",
					"script": "tuya_swversion.js"
				},
				"read": {
					"fn": "zcl",
					"ep": 1,
					"cl": "0x0000",
					"at": "0x0001"
				}
			},
			{
				"name": "attr/type"
			},
			{
				"name": "attr/uniqueid"
			},
			{
				"name": "state/on",
                                "refresh.interval": 500
			},
			{
				"name": "state/reachable"
			},
			{
				"name": "config/interfacemode",
				"refresh.interval": 86400,
				"read": {
					"fn": "zcl",
					"ep": 1,
					"cl": "0x0000",
					"at": [
						"0x0004",
						"0x0000",
						"0x0001",
						"0x0005",
						"0x0007",
						"0xfffe"
					]
				},
				"parse": {
					"fn": "zcl",
					"ep": 1,
					"cl": "0x0000",
					"at": "0xfffe",
					"eval": "Item.val = Attr.val"
				},
				"public": false,
				"default": 0
			}
		]
	}]
}
```

Perhaps this device need binding but I m not sure.

closed 2023-04-14 by github-actions[bot]

dsimop, 2023-05-19:
Could you please indicate a source of instructions on how to manually upload a DDF (Could not find concrete guidelines so far)? I am currently using the deCONZ add-on on Home Assistant (HassIO).
* Maybe @Mimiix you could re-direct me to the respective resources?

Mimiix, 2023-05-19:
@Smanar should be able to. Nevertheless, roaming around this GitHub and seeing other requests might give you some guidance too.

Smanar, 2023-05-19:
On home assitant, the short answer, you need to put the file in `/data/.local/share/dresden-elektronik/deCONZ/devices/` 
The long answer https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-home-assistant-using-text-editor/1839/20

dsimop, 2023-05-19:
Thank you both.