dresden-elektronik/deconz-rest-plugin #6647: Pairing Aqara Smart Wireless Switch WXKG07LM not working anymore with version 2.19.X

EmJay276, 2023-01-06:
Issue https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5599 is still persistent for Aqara WXKG07LM
With Version 2.12.06 I can pair the device, with version 2.19.03 it does not pair.

## Describe the bug

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment

 - Host system: (Raspberry Pi / PC / NAS)
 - Running method: (Raspbian / Ubuntu / Home Assistent deCONZ Add-on / Marthoc Docker container / Windows / Virtual Machine)
 - Firmware version: (26720700)
 - deCONZ version: (2.19.03)
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? If so: Yes, several, but no issue with older version

## deCONZ Logs

## Additional context

Mimiix, 2023-01-06:
Please fill the form in full.

EmJay276, 2023-01-06:
@Mimiix I filled the form except the Log, if it is mandatory I need to reset my switch again.

Mimiix, 2023-01-06:
the question here mostly is what you mean with can't be paired. How does it show in deCONZ?

EmJay276, 2023-01-06:
It does not show up in deconz

Mimiix, 2023-01-06:
Can you show some logs while pairing?

EmJay276, 2023-01-06:
Which debugging levels do you need? I will create an empty docker to avoid my other 50 deviced to spam the log

Mimiix, 2023-01-06:
Info, Info L2, Aps, APS l2, Error , Error L2.

EmJay276, 2023-01-06:
I think a lot of other devices tried to join the network when I paired
```
14:29:51:465 APS-DATA.request id: 24, addrmode: 0x03, addr: 0x00212effff06c332, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x00
14:29:51:465 	asdu (length: 2): 1707
14:29:51:483 APS-DATA.confirm id: 24, status: 0x00 SUCCESS
14:29:51:483 APS-DATA.confirm request id: 24 -> confirmed, timeout -1645226212
14:29:51:495 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 255, rssi: -72
14:29:51:495 	asdu: 17001a070132c306ffff2e21009436f50801881700a2ba120001fc
14:29:51:495 APS-DATA.indication request id: 24 -> finished
14:29:51:496 APS-DATA.request id: 24 erase from queue
14:29:51:872 poll node 00:17:88:01:08:33:2c:5c-0b
14:29:51:872 Poll light node Extended color light 1
14:29:52:381 APS-DATA.request id: 29, addrmode: 0x02, addr: 0xfffc, profile: 0x0000, cluster: 0x0036, ep: 0x00 -> 0x00 queue: 0 len: 3 tx.options 0x00
14:29:52:381 	asdu (length: 3): 050001
14:29:52:382 send permit join, duration: 0
14:29:52:417 APS-DATA.confirm id: 29, status: 0x00 SUCCESS
14:29:52:428 aps request id: 29 finished, erase from queue
14:29:52:734 poll node 00:17:88:01:08:33:2c:5c-0b
14:29:52:734 Poll light node Extended color light 1
14:29:52:782 read attributes of 0x0017880108332C5C cluster: 0x0006: [ 
14:29:52:782 0x0000 
14:29:52:782 ]
14:29:52:782 add task 210 type 19 to 0x0017880108332C5C cluster 0x0006 req.id 32
14:29:52:782 Poll APS request 32 to 0x0017880108332C5C cluster: 0x0006
14:29:52:845 Search sensors done
14:29:52:882 APS-DATA.request id: 32, addrmode: 0x03, addr: 0x0017880108332c5c, profile: 0x0104, cluster: 0x0006, ep: 0x01 -> 0x0B queue: 0 len: 5 tx.options 0x00
14:29:52:882 	asdu (length: 5): 1047000000
14:29:52:938 Poll APS confirm 32 status: 0x00
14:29:52:938 Erase task req-id: 32, type: 19 zcl seqno: 71 send time 0, profileId: 0x0104, clusterId: 0x0006
14:29:52:939 APS-DATA.confirm id: 32, status: 0x00 SUCCESS
14:29:52:939 APS-DATA.confirm request id: 32 -> erase from queue
14:29:52:961 APS-DATA.indication srcAddr: 0x51b1, srcEp: 0x0B dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 255, rssi: -71
14:29:52:961 	asdu: 1847010000001000
14:29:52:961 APS-DATA.request id: 32 erase from queue
14:29:52:961 Node data 0x0017880108332c5c profileId: 0x0104, clusterId: 0x0006
14:29:52:961 0x0017880108332C5C: added ZCL value 0x0B/0x0006/0x0000
14:29:52:962 Websocket 192.168.178.32:33218 send message: {"e":"changed","id":"65520","r":"groups","state":{"all_on":false,"any_on":false},"t":"event"} (ret = 93)
14:29:52:962 Websocket 192.168.178.41:51929 send message: {"e":"changed","id":"65520","r":"groups","state":{"all_on":false,"any_on":false},"t":"event"} (ret = 93)
14:29:53:205 APS-DATA.indication srcAddr: 0x676e, srcEp: 0x0B dstAddrMode: 2, profile: 0x0104, cluster: 0x0019, lqi: 255, rssi: -63
14:29:53:205 	asdu: 012d01004f12191401300117
14:29:53:205 APS-DATA.indication from unknown node 0xA4C138672DCA7FD7
14:29:53:256 read attributes of 0x0017880108332C5C cluster: 0x0300: [ 
14:29:53:256 0x0000 
14:29:53:256 0x0001 
14:29:53:256 0x0003 
14:29:53:256 0x0004 
14:29:53:256 0x0007 
14:29:53:256 0x0008 
14:29:53:256 0x4001 
14:29:53:256 ]
14:29:53:257 add task 213 type 19 to 0x0017880108332C5C cluster 0x0300 req.id 37
14:29:53:257 Poll APS request 37 to 0x0017880108332C5C cluster: 0x0300
14:29:53:281 APS-DATA.request id: 37, addrmode: 0x03, addr: 0x0017880108332c5c, profile: 0x0104, cluster: 0x0300, ep: 0x01 -> 0x0B queue: 0 len: 17 tx.options 0x00
14:29:53:282 	asdu (length: 17): 1048000000010003000400070008000140
14:29:53:306 Poll APS confirm 37 status: 0x00
14:29:53:306 Erase task req-id: 37, type: 19 zcl seqno: 72 send time 0, profileId: 0x0104, clusterId: 0x0300
14:29:53:307 APS-DATA.confirm id: 37, status: 0x00 SUCCESS
14:29:53:307 APS-DATA.confirm request id: 37 -> erase from queue
14:29:53:323 APS-DATA.indication srcAddr: 0x51b1, srcEp: 0x0B dstAddrMode: 2, profile: 0x0104, cluster: 0x0300, lqi: 255, rssi: -71
14:29:53:323 	asdu: 18480100000020200100002095030000219c7604000021506907000021790108000030020140003002
14:29:53:323 APS-DATA.request id: 37 erase from queue
14:29:53:323 Node data 0x0017880108332c5c profileId: 0x0104, clusterId: 0x0300
14:29:53:323 0x0017880108332C5C: added ZCL value 0x0B/0x0300/0x0000
14:29:53:324 0x0017880108332C5C: added ZCL value 0x0B/0x0300/0x0001
14:29:53:324 0x0017880108332C5C: added ZCL value 0x0B/0x0300/0x0003
14:29:53:324 0x0017880108332C5C: added ZCL value 0x0B/0x0300/0x0004
14:29:53:324 0x0017880108332C5C: added ZCL value 0x0B/0x0300/0x0007
14:29:53:324 0x0017880108332C5C: added ZCL value 0x0B/0x0300/0x0008
14:29:53:324 0x0017880108332C5C: added ZCL value 0x0B/0x0300/0x4001
14:29:53:325 Websocket 192.168.178.32:33218 send message: {"e":"changed","id":"1","r":"lights","state":{"alert":null,"bri":null,"colormode":"ct","ct":377,"effect":"none","hue":8256,"on":null,"reachable":true,"sat":149,"xy":[0.4633,0.4114]},"t":"event","uniqueid":"00:17:88:01:08:33:2c:5c-0b"} (ret = 234)
[... 511 lines cut ...]
14:30:11:581 APS-DATA.request id: 188, addrmode: 0x03, addr: 0x0017880108332c5c, profile: 0x0104, cluster: 0x0006, ep: 0x01 -> 0x0B queue: 0 len: 5 tx.options 0x00
14:30:11:581 	asdu (length: 5): 1059000000
14:30:11:712 Poll APS confirm 188 status: 0x00
14:30:11:712 Erase task req-id: 188, type: 19 zcl seqno: 89 send time 0, profileId: 0x0104, clusterId: 0x0006
[... cut ...]
14:30:12:317 APS-DATA.indication srcAddr: 0x51b1, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 255, rssi: -70
14:30:12:317 	asdu: 2e000d0c0132c306ffff2e2100fc3967feff20ba8467df25020fb6
14:30:12:318 APS-DATA.indication request id: 192 -> finished
14:30:12:318 APS-DATA.request id: 192 erase from queue
14:30:12:713 poll node 00:17:88:01:08:33:2c:5c-0b
14:30:12:713 Poll light node Extended color light 1
14:30:12:765 read attributes of 0x0017880108332C5C cluster: 0x0006: [ 
14:30:12:765 0x0000 
14:30:12:765 ]
```

Mimiix, 2023-01-06:
Can you please post on Pastebin?

Additionally, i don't see anything about a new device joining.

EmJay276, 2023-01-06:
Here is a new try with my existing network
https://pastebin.com/VXyyaPEj

EmJay276, 2023-01-06:
The device is still in deconz (I reset only the switch, and did not delete it in deconz).
But I had the same behaviour when deleting it in deconz first before trying to rejoin it.
In the previous versions it worked fine to enable join and set the device to pairing mode.

EmJay276, 2023-01-06:
> You mention it can't be paired , but it's already in there again?

Sorry it was a bit confusing
1. Device is connected to the network and working properly
2. Device is reset by pressing (accidentally) the buttons for 10 seconds
3. The Device itself thinks it is no longer connected to the network
4. In Deconz the device is still showing (gets not removed from deconz automatically)
5. I want to reconnect the device, so I enable permit join and set the device to pairing mode
6. Nothing happens in Deconz
7. I remove the device manually from deconz
8. I do step 5. again, enable permit jojn and set the device to pairing mode
9. Still nothing happens

EmJay276, 2023-01-06:
> To properly test this:
> 
> Delete device in Phoscon Reset Device Reboot deCONZ start logging search devices See if found Share logs

Ok that really weired, not the device connected properly?!
https://pastebin.com/Up614GMq

The only difference was a restart of my docker container after removing the device from Phoscon.
Is the reboot mandatory? In my opinion this is very unintuitive.

Mimiix, 2023-01-06:
The only thing i know, is that it takes 24hours for a device to be marked "away".  Probably something going on there.

Additionally, I am noticing 0xE1 in yours logs:
15:09:59:239 0x84BA20FFFE65517C error APSDE-DATA.confirm: 0xE1 on task
15:10:01:433 Bind response success for 0x84ba20fffe655249 ep: 0x01 cluster: 0x0006
15:10:03:281 0x0017880108B4611C error APSDE-DATA.confirm: 0xE1 on task

This also might impact the experience.

EmJay276, 2023-01-06:
"0xE1 on task" is most likely due to interferance / weak signal, right?
How can I find out which device it is to check?

Mimiix, 2023-01-06:
> "0xE1 on task" is most likely due to interferance / weak signal, right? How can I find out which device it is to check?

Your coordinator. It has to do with USB3.0 and the surroundings. Not with a specific device. Please read the wiki , there's a error codes page explaining a bit.

Nevertheless, i believe your issue is not a bug and more an issue with your environment, therefore i am closing the issue here.

closed 2023-01-06 by Mimiix

EmJay276, 2023-01-06:
Thank for your help! The advice with restarting the docker container instead of waiting 24h for the device to be set "away", did the trick for me to get it reconnecting.
Perhaps it should be mentioned somewhere.

Mimiix, 2023-01-06:
Feel free to adjust the wiki. However it's normal, because on startup the devices are being checked by the bootup procedure.