dresden-elektronik/deconz-rest-plugin #6656: reject aps request queue is full

donnib, 2023-01-09:
## Describe the bug

Suddenly devices do not respond anymore.

## Steps to reproduce the behavior

USe the product and suddenly you need to restart the gateway.

## Expected behavior

## Screenshots

## Environment

 - Host system: (Raspberry Pi /)
 - Running method: (Raspbian)
 - Firmware version: (26400500)
 - deCONZ version: (2.19.03)
 - Device: (RaspBee I)

## deCONZ Logs

07:17:00:543 Daylight now: dawn, status: 130, daylight: 0, dark: 1
07:17:07:043 sql exec SELECT conf FROM zbconf ORDER BY rowid desc limit 1
07:17:07:044 Idle timer triggered
07:17:10:543 Daylight now: dawn, status: 130, daylight: 0, dark: 1
07:17:20:543 Daylight now: dawn, status: 130, daylight: 0, dark: 1
07:17:25:008 add task 273347 type 14 to group 0x0005 cluster 0x0006 req.id 175
07:17:25:010 drop request 169 send time 61803, cluster 0x0000, after 296 seconds
07:17:25:016 Websocket 10.10.0.7:45152 send message: {"e":"changed","id":"6","r":"lights","state":{"alert":null,"bri":254,"colormode":"ct","ct":260,"on":true,"reachable":true},"t":"event","uniqueid":"00:0b:57:ff:fe:38:fc:b3-01"} (ret = 175)
07:17:25:020 Websocket 10.10.0.7:45152 send message: {"e":"changed","id":"5","r":"lights","state":{"alert":null,"bri":254,"colormode":"ct","ct":260,"on":true,"reachable":true},"t":"event","uniqueid":"00:0b:57:ff:fe:37:bb:67-01"} (ret = 175)
07:17:25:022 Websocket 10.10.0.7:45152 send message: {"e":"changed","id":"2","r":"lights","state":{"alert":null,"bri":254,"colormode":"ct","ct":260,"on":true,"reachable":true},"t":"event","uniqueid":"00:0b:57:ff:fe:24:d5:98-01"} (ret = 175)
07:17:25:024 Websocket 10.10.0.7:45152 send message: {"e":"changed","id":"3","r":"lights","state":{"alert":null,"bri":254,"colormode":"ct","ct":260,"on":true,"reachable":true},"t":"event","uniqueid":"00:0b:57:ff:fe:30:60:d3-01"} (ret = 175)
07:17:25:039 Websocket 10.10.0.7:45152 send message: {"e":"changed","id":"4","r":"lights","state":{"alert":null,"bri":254,"colormode":"ct","ct":260,"on":true,"reachable":true},"t":"event","uniqueid":"00:0b:57:ff:fe:37:dc:4b-01"} (ret = 175)
07:17:25:042 Websocket 10.10.0.7:45152 send message: {"e":"changed","id":"5","r":"groups","state":{"all_on":true,"any_on":true},"t":"event"} (ret = 32050320)
07:17:25:044 Websocket 10.10.0.7:45152 send message: {"e":"changed","id":"3","r":"groups","state":{"all_on":true,"any_on":true},"t":"event"} (ret = 32049840)
07:17:25:058 APS-DATA.request id: 175, addrmode: 0x01, addr: 0x0005, profile: 0x0104, cluster: 0x0006, ep: 0x01 -> 0xFF queue: 23 len: 3 tx.options 0x00
07:17:25:059 	asdu (length: 3): 11f301
07:17:28:542 timeout on poll APS confirm
07:17:28:543 read attributes of 0x000B57FFFE37DC4B cluster: 0x0008: [ 
07:17:28:544 0x0000 
07:17:28:544 ]
07:17:28:544 add task 273348 type 19 to 0x000B57FFFE37DC4B cluster 0x0008 req.id 176
07:17:28:545 Poll APS request 176 to 0x000B57FFFE37DC4B cluster: 0x0008
07:17:28:545 drop request 170 send time 61864, cluster 0x0006, after 239 seconds
07:17:28:643 APS-DATA.request id: 176, addrmode: 0x03, addr: 0x000b57fffe37dc4b, profile: 0x0104, cluster: 0x0008, ep: 0x01 -> 0x01 queue: 24 len: 5 tx.options 0x00
07:17:28:644 	asdu (length: 5): 10f4000000
07:17:30:543 Daylight now: dawn, status: 130, daylight: 0, dark: 1
07:17:35:885 add task 273350 type 14 to 0x00178801067FEB34 cluster 0x0006 req.id 178
07:17:35:886 drop request 171 send time 61925, cluster 0x0006, after 185 seconds
07:17:35:890 Websocket 10.10.0.7:45152 send message: {"e":"changed","id":"76","r":"lights","state":{"alert":null,"bri":254,"colormode":"ct","ct":307,"on":true,"reachable":true},"t":"event","uniqueid":"00:17:88:01:06:7f:eb:34-0b"} (ret = 176)
07:17:35:942 reject aps request queue is full (25)
07:17:35:944 enqueue APS request failed with error 2, drop
07:17:40:543 Daylight now: dawn, status: 130, daylight: 0, dark: 1
07:17:43:855 add task 273352 type 14 to group 0x0005 cluster 0x0006 req.id 180
07:17:43:856 reject aps request queue is full (25)
07:17:43:933 Websocket 10.10.0.7:45152 send message: {"e":"changed","id":"6","r":"lights","state":{"alert":null,"bri":254,"colormode":"ct","ct":260,"on":false,"reachable":true},"t":"event","uniqueid":"00:0b:57:ff:fe:38:fc:b3-01"} (ret = 176)
07:17:43:937 Websocket 10.10.0.7:45152 send message: {"e":"changed","id":"5","r":"lights","state":{"alert":null,"bri":254,"colormode":"ct","ct":260,"on":false,"reachable":true},"t":"event","uniqueid":"00:0b:57:ff:fe:37:bb:67-01"} (ret = 176)
07:17:43:941 Websocket 10.10.0.7:45152 send message: {"e":"changed","id":"2","r":"lights","state":{"alert":null,"bri":254,"colormode":"ct","ct":260,"on":false,"reachable":true},"t":"event","uniqueid":"00:0b:57:ff:fe:24:d5:98-01"} (ret = 176)
07:17:43:982 Websocket 10.10.0.7:45152 send message: {"e":"changed","id":"3","r":"lights","state":{"alert":null,"bri":254,"colormode":"ct","ct":260,"on":false,"reachable":true},"t":"event","uniqueid":"00:0b:57:ff:fe:30:60:d3-01"} (ret = 176)
07:17:43:986 Websocket 10.10.0.7:45152 send message: {"e":"changed","id":"4","r":"lights","state":{"alert":null,"bri":254,"colormode":"ct","ct":260,"on":false,"reachable":true},"t":"event","uniqueid":"00:0b:57:ff:fe:37:dc:4b-01"} (ret = 176)
07:17:43:989 Websocket 10.10.0.7:45152 send message: {"e":"changed","id":"5","r":"groups","state":{"all_on":false,"any_on":false},"t":"event"} (ret = 32050320)
07:17:43:991 Websocket 10.10.0.7:45152 send message: {"e":"changed","id":"3","r":"groups","state":{"all_on":false,"any_on":false},"t":"event"} (ret = 32049840)
07:17:43:993 reject aps request queue is full (25)
07:17:44:043 reject aps request queue is full (25)
07:17:44:142 reject aps request queue is full (25)
07:17:44:243 reject aps request queue is full (25)
07:17:44:270 add task 273354 type 14 to group 0x0005 cluster 0x0006 req.id 182
07:17:44:271 Replace task 273352 type 14 in queue cluster 0x0006 with newer task 273354 of same type. 4 runnig tasks
07:17:44:273 reject aps request queue is full (25)
[... 695 lines cut ...]
07:18:38:243 reject aps request queue is full (25)
07:18:38:343 reject aps request queue is full (25)
07:18:38:505 reject aps request queue is full (25)
07:18:38:543 reject aps request queue is full (25)
07:18:38:642 reject aps request queue is full (25)
07:18:38:743 reject aps request queue is full (25)
07:18:38:843 reject aps request queue is full (25)
07:18:38:943 reject aps request queue is full (25)
07:18:39:042 reject aps request queue is full (25)
07:18:39:143 reject aps request queue is full (25)
07:18:39:243 reject aps request queue is full (25)
07:18:39:342 reject aps request queue is full (25)
07:18:39:443 reject aps request queue is full (25)
07:18:39:545 reject aps request queue is full (25)
07:18:39:642 reject aps request queue is full (25)
07:18:39:743 reject aps request queue is full (25)
07:18:39:843 reject aps request queue is full (25)
07:18:39:942 reject aps request queue is full (25)
07:18:40:043 reject aps request queue is full (25)
07:18:40:143 reject aps request queue is full (25)
07:18:40:243 reject aps request queue is full (25)
07:18:40:362 reject aps request queue is full (25)
07:18:40:442 reject aps request queue is full (25)
07:18:40:543 Daylight now: dawn, status: 130, daylight: 0, dark: 1
07:18:40:544 reject aps request queue is full (25)

Smanar, 2023-01-13:
Hello, can you make a try with the version 2.20.X ?

mihaimaersk, 2023-01-14:
@Smanar sure i can but has there been done any changes which relates to this problem e.g can you confirm this problem that is been addressed ? @manup any input here ?

donnib, 2023-01-14:
@Smanar sure i can but has there been done any changes which relates to this problem e.g can you confirm this problem that is been addressed ? @manup any input here ?

Smanar, 2023-01-14:
If I m right aps issue are here

>Fix firmware connectivity problems and reduce delays when sending commands.

manup, 2023-01-15:
Hi, yes v2.20.0-beta has quite some changes related to this issue. With help of another developers network we could iron out a few gotchas, but needs more confirmation from more setups. Unfortunately it's one of these problems which can't be reproduced easily in my own networks.

closed 2023-02-13 by github-actions[bot]