dresden-elektronik/deconz-rest-plugin #6657: [Device Support Request] Aqara QBKG38LM - Smart Switch without neutral

xADDRx, 2023-01-09:
Hello there,
I recently ordered from AliExpress  AQARA QBKG38LM . Smart switch with without neutral wire.

When attempting to join them using Phoscon, the "Add Switch" function does not detect them under Phoscon.

Using Deconz view, the following node and cluster info are available:

[image]

## Device

- Product name: QBKG38LM 
- Manufacturer: LUMI.
- Model identifier: lumi.switch.b1lc04.
- Device type : Please remove all unrelated device types. 
  - Switch

## Screenshots

### Basic
[image]

### Identify
[image]

### Alarms

### Device Temperature

### Groups
[image]
[image]

### Scenes
[image]
[image]
[image]
[image]

### On/Off
[image]
[image]

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

xADDRx, 2023-01-21:
Hi

Now the switches are displayed as lights, so I can already control the switches. I'll add your example when I get a chance to display it correctly as a switch.

Thank you for your help!

Smanar, 2023-01-21:
It's more complicated ^^
Wired switch are light not switch, you send order to them like for light, you can't send information to a battery switch. The difference is an zigbee side too, for group use for exemple.

In reality your device can have both mode, if you use for exemple the "decoupled" mode. But not usefull for me.

xADDRx, 2023-01-21:
Yes, it's also the first time that I'm playing around in the DDF :).

So can I leave it like this or what you mean?

Smanar, 2023-01-21:
>So can I leave it like this or what you mean?

As you want, it's for you, not for me ^^.
If you like it as it, you can too make a PR for others users.

closed 2023-02-19 by github-actions[bot]