dresden-elektronik/deconz-rest-plugin #6658: PROTO: CRC error

azlux, 2023-01-09:
## Describe the bug
Sometime, the raspbee2 (via deconz and home assistant) is slow to modify a state.
I have many error into the log, check below

## Steps to reproduce the behavior
Not sure, I suspect it because it's near my internet box (with 2.4 wifi)
The channel have been change to not have overlap. I have better response since that time (but not all good).

Is there are a way to check is the CRC is between the deconz and the raspbee II, or between the raspbee II and the remotes zigbee devices ?

## Environment
 - Host system: Raspberry Pi 3
 - Running method: docker : Home Assistant + deconz
 - Firmware version: 26720700
 - deCONZ version: 2.18.02 (same for 2.19.03)
 - Device: RaspBee II

## deCONZ Logs
```
deconz           | 16:28:04:258 ZCL attribute report 0x0015BC001A01E691 for cluster: 0x0400, ep: 0x27, frame control: 0x18, mfcode: 0x0000
deconz           | PROTO: CRC error
deconz           | PROTO: CRC error
deconz           | PROTO: CRC error
deconz           | PROTO: CRC error
deconz           | 16:28:19:494 Device TTL 1981 s flags: 0x7
deconz           | PROTO: CRC error
deconz           | PROTO: CRC error
deconz           | PROTO: CRC error
deconz           | PROTO: CRC error
deconz           | PROTO: CRC error
deconz           | 16:29:16:679 0x00124B0023A6C6C2 error APSDE-DATA.confirm: 0xA7 on task
deconz           | 16:29:19:489 Device TTL 1921 s flags: 0x7
deconz           | PROTO: CRC error
deconz           | PROTO: CRC error
deconz           | 16:29:57:478 0x00124B0023A65C6F error APSDE-DATA.confirm: 0xA7 on task
deconz           | PROTO: CRC error
deconz           | PROTO: CRC error
deconz           | PROTO: CRC error
deconz           | PROTO: CRC error
deconz           | 16:30:19:487 Device TTL 1861 s flags: 0x7
```

Best regards
Azlux

manup, 2023-01-09:
Beside interference, it can be worth to check if the serial port is configured properly, also to make sure that no other process is using the UART (system console). Related steps from the docs at https://phoscon.de/en/raspbee2/install#raspbian

[image]

azlux, 2023-01-09:
Yes, this part is good.
And since I've replace all my Bluetooth device, I've also disable the `dtoverlay=miniuart-bt`

```
[azlux@hodor: ~]$ cat /boot/config.txt
...
[all]
enable_uart=1
#dtoverlay=miniuart-bt
dtoverlay=disable-bt
```

closed 2023-03-05 by github-actions[bot]