dresden-elektronik/deconz-rest-plugin #6664: AWOW Zigbee Heating Thermostat Radiator MS-SH4-Z

bruesche, 2023-01-11:
Hi there,

I tried to implement the thermostat from AWOW. 
The Node is shown in deconz gui but is not shown in normal deconz. I tried to generate a DDF file as mentioned here:

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet

[image]

So I can see it in deconz and in iobroker. But I did't get values from the thermostat. Everything is Zero. 

Here is my generated DDF:
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZE200_thbr5z34",
  "modelid": "TS0601",
  "product": "TS0601",
  "sleeper": false,
  "status": "Gold",
  "path": "/devices/ts0601.json",
  "subdevices": [
    {
      "type": "$TYPE_THERMOSTAT",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0201"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/heatsetpoint",
          "description": "Target temperature of a thermostat.",
          "default": 0
        },
        {
          "name": "config/mode",
          "description": "Mode of the device."
        },
        {
          "name": "config/offset",
          "description": "Relative offset to the main measured value.",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/on",
          "description": "True when device is on; false when off.",
          "refresh.interval": 5
        },
        {
          "name": "state/temperature",
          "description": "The current temperature in °C &times; 100.",
          "default": 0
        },
        {
          "name": "state/valve",
          "description": "The open percentage of the thermostat valve.",
          "default": 0
        }
      ]
    }
  ]
}

Can anyone help me?

If the Infos above here are all wrong, the here are the screenshots for device request...

## Device

- Product name: AWOW Zigbee Heating Thermostat Radiator MS-SH4-Z
- Manufacturer: _TZE200_thbr5z34
- Model identifier: ts0601
- Device type : Thermostat

## Screenshots

### Basic
[image]

[image]

[image]

[image]

[image]

[image]

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

If more info is needed, please let me know.

Thx

bruesche, 2023-01-11:
Hi,

ok, I have to wait for next Deconz version right? Habe you an Idea When it could work? Or is there a workaround?

Smanar, 2023-01-12:
>ok, I have to wait for next Deconz version right

There is a new 2.20 branch available, there is no official roadmap, but if all is fine the next step is TRV, so Tuya TRV too

bruesche, 2023-01-12:
I installed the new Version 2.20 right before i tried to connect the Thermostat. So i Hope the next step comes soon. I leave this request open

Smanar, 2023-01-13:
@bruesche by curiosity what is your OS ?
I will update the PR and I need tester with tuya TRV to test it. But need other thing than HA as OS, a real linux OS ?

bruesche, 2023-01-13:
My OS is bullseye on Raspberry Pi 4

Smanar, 2023-01-13:
Ha nice, so if I give you command line, you are able to compile the code to support your device ? (Will be c++ code to support a DDF) ?

bruesche, 2023-01-13:
I can‘t compile. I don‘t know how. I am only a consumer. But I can try when I get a detailed guide….

Smanar, 2023-01-14:
Don't worry, I will give you all commands, no need to be a dev.
The harder thing will be to create text file for DDF, but as you have a classic OS, I can found command line too if you have problem.

I will make the code change on the Week end.

bruesche, 2023-01-14:
I‘ll do my very best… 😉

Smanar, 2023-01-14:
So have updated the code and start a DDF for your device to test it.

You have the procedure explained here https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Compiling-the-REST-plugin-for-device-specific-testing

So for you, after having installed deconz (The code is based on realy last one, so need a recent version, no problem for you as you are already on 2.20):

The next command line are not dangerous, they just install, lot of files.
Them launch the compilation, If you have a small raspberry, it can freeze during compilation, my pi 3 do, so remove "-j2", can remove it too if the rapsberry is your production machine, to don't freeze it
And the last one replace your original file by the new one, so by security you can make a backup of this one in another folder (rename it don't work) /usr/share/deCONZ/plugins/libde_rest_plugin.so

```
sudo apt install deconz-dev
git clone --branch tuya_trv_clean https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

Then you can run deconz (you can close it before the file replacement)

It don't impact your zigbee database, but can use a lot the CPU (at least the first time, compilation will be faster after).

So now with the new compiled lib you are able to use DDF for Tuya TRV, but unfortunately you need to make all action manually (but dont worry, nothing desctructive)
- Install the file config_windowopen_set_item.json in devices/generic/items
- Install the 2 DDF in the devices/tuya folder
- Install the 2 JS file in the devices/tuya folder

You can see files and paths here too, easier to create file with copy/paste https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6054/files

No hard if you have a desktop. Else I can give you command line to do that. You can have right issue too.

Take lot of time, but like I have said previoulsy, there is nothing destructive.

Smanar, 2023-01-16:
@bruesche the code was updated. But the procedure is same.
Ask for me if you have problem or don't understand something.

bruesche, 2023-01-16:
Sorry for the Delay. Can you give me the command lines for the files and directory? Desktop is not mines…

Smanar, 2023-01-16:
Ok so this one copy the whole DDF part
I presume you already have used 

`cd deconz-rest-plugin `

so now you can use
`sudo cp -R devices /usr/share/deCONZ/`

Il will copy all the "devices" contain from the PR to your deconz folder. More desctructive, but still no impact on your database, and thoses files are replaced at all deconz update.

bruesche, 2023-01-17:
a little Problem: 
pi@raspberrypi:~ $ cd deconz-rest-plugin
pi@raspberrypi:~/deconz-rest-plugin $ qmake && make
qmake: could not find a Qt installation of ''
pi@raspberrypi:~/deconz-rest-plugin $ qmake && make -j2
qmake: could not find a Qt installation of ''
pi@raspberrypi:~/deconz-rest-plugin $

bruesche, 2023-01-17:
something missing?

Smanar, 2023-01-17:
You haven't error message with this one

`sudo apt install deconz-dev`

Your system is up to date ? the previous request install the QT stuff normally, else can use 

`sudo apt-get install qt5-default`

bruesche, 2023-01-19:
After I install a fresh image the compiling works fine. Could connect the thermostat with deconz web page without problems. 
Datapoints are actual in iobroker. Battery is Zero but thermostat is reachable an on. When I try to set Heatpoint to 18 degrees the thermostat change to 90 degrees. Tried heat point set to 9 degrees the thermostat changed to 45 degrees.

bruesche, 2023-01-19:
Thermostat response only in auto mode. In manuell mode nothing happens. Can lock the thermostat over iobroker and unlock again.

bruesche, 2023-01-19:
the internal thermometer seems to work and is updating

bruesche, 2023-01-19:
Updated. Works fine! Degrees in iobroker set correctly at thermostat.

Smanar, 2023-01-19:
Nice, so just have problem on config/mode and config/battery ?
Have you tried to set a "manu setheatpoint" (perhaps explained in the documentation) to test mode "manu" and "off" ?

bruesche, 2023-01-19:
read in the manual, that the auto mode is not so good. the program defined in the thermostat will override the value of heat point if time is arrived. so the heat point for manuell mode would be nicer for smart home.

Smanar, 2023-01-19:
so 
"dpid": 105 is for the heatpoint "auto"
"dpid": 16 is for the normal heatpoint

You can edit them in the DDF, then restart deconz to test.

Smanar, 2023-01-19:
>but can't do it from iobroker. only see the value

You have an error message ? Or nothing to change it ?

bruesche, 2023-01-19:
heatpoint change works fine. Now degrees can be change in Manu mode.

No Errormessage. When I change the mode from thermostat I can see the mode change in iobroker. Other way round I change the mode in iobroker nothing happens at thermostat.

Smanar, 2023-01-19:
Ok so i m reading the Z2M code and this device use probably special mode for this request.

I have a try to ask you.

Enable deconz log in deconz / help / debug view, with flag "info" then change the mode manualy on the device, you will have as logs something like "TY_DATA_%s: seq %u, dpid: 0x%02X, type: 0x%02X, length: %u, val: %d\n",

With the dpid = 2.

bruesche, 2023-01-19:
curl -H 'Content-Type: application/json' -X PUT -d '{"mode":"off"}' http://IP:PORT/api/KEY/sensors/ID/config

works fine. I'll be able to change with this command line

Smanar, 2023-01-19:
>works fine. I'll be able to change with this command line

Ok so the problem is not from deconz ^^, no need more test on config/mode

Possibles values are "off" , "heat" and "auto", and they are same from the start, idk why iobroker have issue with them.

bruesche, 2023-01-19:
the mode isn't really a problem. I think smart Home only need heat mode (Manu). the only thing is the battery. then I think everything is fine.

Smanar, 2023-01-19:
Ha yes, so for the battery can try same, if you have time.

Enable deconz log in deconz / help / debug view, with flag "info", then just wait ^^, not possible to force the device to send a battery notification, so can take soooo log time, you will have as logs something like "TY_DATA_%s: seq %u, dpid: 0x%02X, type: 0x%02X, length: %u, val: %d\

With the dpid = 34 = 0x22

You can too enable the flag "DDF"

Smanar, 2023-01-19:
`deconz / help / debug view`

On the deconz application menu (on the GUI, I haven't asked if you still have it ?), the "help" part.

bruesche, 2023-01-19:
headless installation

Smanar, 2023-01-19:
So need to use the command line, more stressing for your hardware, and not possible if you use an SD card.

On other project they are using the same dp, but sure it work. You can take a look here https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/deCONZ-command-line-parameters#available-debug-parameters

But you are not alone, thx for you help btw, if it's not hurry for you, can try for another user with the same issue and the GUI.

Or you can try to remove the "*20" in the DDF, to be sure it's not because the value is too big. From that I have read it's the voltage, so 5 V, * 20 can't be superior at 100. but I have never see the value.

bruesche, 2023-01-19:
Hope i could help.

bruesche, 2023-01-22:
Tried to remove *20 by dpid 34 battery. Nothing changed: Battery shows zero %

Smanar, 2023-01-22:
Ok so I don't think it's a "too big value" issue.

Perhaps they are wrong and the device don't send battery level but low battery warning ?
Can you try replace

the "config/battery" by a "state/lowbattery" ?

```
        {
          "name": "state/lowbattery",
          "parse": {"fn": "tuya", "dpid": 110, "eval": "Item.val = Attr.val;"},
          "read": {"fn": "none"}
        },
```

bruesche, 2023-01-23:
hi, added the state/lowbattery. in iobroker the state arrive but show value = (null)

Smanar, 2023-01-23:
Ok so I mean the device probably don't use this DP.
So the device use the dpid 34 but we use it badly, but seriously I don't know how with a so simple explression than "Item.val = Attr.val;", or it use another dpid in reality.
I have catch the 34 value here https://github.com/zigpy/zha-device-handlers/issues/978#issuecomment-957972450

BTW for the iobroker issue, you can ask them for the issue ?
You don't have log inside to see what is the command used by iobroker ? the problem is in the body so in `{"mode":"off"}`

bruesche, 2023-01-24:
Should I try 21 and 14 for dpid?

Smanar, 2023-01-24:
You can make a try, yes, easier than getting logs, just need to restart deconz after change (no need to re-include the device).
But remove the *20 as value can be > 100

        {
          "name": "config/battery",
          "parse": {"fn": "tuya", "dpid": 21, "eval": "Item.val = Attr.val;"},
          "read": {"fn": "none"}
        },

bruesche, 2023-01-26:
Hi, tried dpid 14 and 21. No value for battery...

Smanar, 2023-01-26:
Ok, so better give up, can't make in blind mode. Just need to wait for someone with a working device and the GUI.
You are not alone with this issue, and we have it on other TRV too, there is realy something strange here.

bruesche, 2023-02-17:
Activ

Smanar, 2023-02-17:
Don't worry, the device was in the PR https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6674
And it will be probably in the version 2.21.

If I m right you just miss the battery ATM ?

bruesche, 2023-02-18:
Thats correct. 
i am very happy the the device is coming 🥳

bruesche, 2023-03-12:
Active

Smanar, 2023-03-12:
The PR is valided, the code is already included on last version.

closed 2023-03-13 by bruesche

Mimiix, 2023-03-13:
@smanar Was this indeed finished? If the case, may i please remind you to link your PR to this issue in the future ;)?