dresden-elektronik/deconz-rest-plugin #6665: Model Id reported as hexadecimal and not string

BabaIsYou, 2023-01-11:
## Describe the bug
Fed up to have a thermostat with "sticky" 0xFF endpoints that always reappears I decided, (after upgrade to version 2.19.3 beta) to delete the device once again and rejoining it to have it cleared.

Surprisingly it identified itself not with a string as usual for "Model Identifier" in Basic cluster, but with haxadeciaml value ?!

i.e. instaed of model identifier **SMT402AD** i got litteraly **0x534d5434303241440000000000301800203944000803000000dd590008ffff**
and in Location description (in the same cluster), I got also a strange value 
[image]

Of course the dedicated DDF (SMT402ad.json) is not recognzied until I add **0x534d5434303241440000000000301800203944000803000000dd590008ffff** as modelid. Then, once reloaded, the thermsotat is recognized and all seems to work ... and I didn't have 0xFF endpoints anymore ! ;-)

## Steps to reproduce the behavior
I deleted the device several times and rejoined it with the same behavior. Even after doing a factory reset on the thermostat the behavior is the same.

## Expected behavior

To be recognized as a **SMT402AD** and not as **0x534d5434303241440000000000301800203944000803000000dd590008ffff**
## Screenshots

## Environment

 - Host system: Raspberry Pi /
 - Running method: Raspbian
 - Firmware version: 26720700
 - deCONZ version: 2.19.3 beta
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? If so: Which? RFXcomm and Z-wave dongle

## deCONZ Logs
Which kind of log could be useful to trace that ?

## Additional context
As we can notice **0x534d543430324144** is the hexadecimal value for **SMT402AD** but don't know what the rest of the value stands for.

Smanar, 2023-01-13:
So you mean the model id in deconz (on cluster 0x0000) was "SMT402AD" but on recent version "0x534d5434303241440000000000301800203944000803000000dd590008ffff" ?

BabaIsYou, 2023-01-13:
Exactly, just for this device. I deleted it and repairing a couple of time with same behavior !

manup, 2023-01-14:
Interesting, I'd propose we check the APS_L2 / ZCL logs when reading the modelid to see what the raw data contains.
A while ago I've added some code to the ZclAttribute parser to let invalid strings be represented as hex string (was needed for another device) to have access to all the data in the string.

When you enter the hex representation `534d5434303241440000000000301800203944000803000000dd590008ffff` in below site you see that the first part indeed contains "SMT402AD" and then comes some binary encoded stuff which can't be represented as string, the old code would just cut this part off at first `00` like a zero terminated C-string.

https://www.rapidtables.com/convert/number/hex-to-ascii.html

Not sure what's the best approach to tackle this, both ways have their pros and contras.

BabaIsYou, 2023-01-15:
> the first part indeed contains "SMT402AD"

Yes I already noticed it in the additional context while opening  this issue  ;-) 
We can notice too in the screenshot that the "location description" is also in hexadecimal !?

I'll give a try  to get logs as soon as I can ... Stay tune ;-)

BabaIsYou, 2023-01-17:
Grrr !!! After some Deconz restarts to test new DDFs,  then 0XFF endpoints are back again ... with the hexadecimal Model ID but the deleted old device names and cluster (0x0402)  that did not exist anymore in the actual DDF ... Completely crazy.

Think that sometimes DeConz relaod some configuration from kind of "cache" or "backup" database and mix it with current i"nfrastructure"
[image]

Smanar, 2023-01-17:
Strange, long time I haven't see this issue on recent version.

>After some Deconz restarts

Just restart ? If I m right deconz don't create new entries if the permit join is not enabled ?

BabaIsYou, 2023-01-17:
To be honest, yesterday I tried a lot of things, adding a new aquara light sensor (then permit join) that didn't work on first try, trying some DDF modifications on another presence sensor device  already joinded (then some retsart of DeConz too). 
At which time these old EP where coming back, I don't really trace that, I discovered them this morning via Domoticz plugin interface. ANd they exist also in API of course.

```
{
    "config": {
        "offset": 0,
        "on": true,
        "reachable": true
    },
    "etag": "b5568cb4c2f4d6d25fd7f5d51968c0a4",
    "lastannounced": null,
    "lastseen": "2023-01-17T19:37Z",
    "manufacturername": "Stelpro",
    "modelid": "0x534d54343032414400dd590008fffffffff39e0008fffffffff9ffffffffff",
    "name": "Humidité CdM",
    "state": {
        "humidity": 5000,
        "lastupdated": "2022-10-12T09:15:31.898"
    },
    "type": "ZHAHumidity",
    "uniqueid": "f8:f0:05:ff:ff:70:7e:95-ff-0405"
}
```

Each modification on these FF EP (renaming, deleting, etc ...) acts on the "regular" EP.

BabaIsYou, 2023-01-17:
It's quite clear for me that these old 0XFF EPs are coming from a sort of cache or backup because the Cluster 0x0402 doesn't exist on this device nor the associated DDF (and it doesn,t appear in the "regular" EP). I think that DecOnz crashes last afternoon during my several tests ... may be it's the root of this re-appearing ...

SwoopX, 2023-01-17:
@BabaIsYou I'm afraid the partial logs you shared above is not quite what holds the gems. The APS_L2 logs would give us the actual APS payload (missing above) and maybe some insights what the device sends (in addition). Would be great to give it another go with ZCL and APS_L2.

With regard to the endpoint(s), you might want to take a look at the database. Data of interest is stored in `sensors` table (legacy), but also in table `resource_items`. Eventually, data there is kinda out of sync and loaded/processed un an unfortunate way. That might also cause the changing modelIDs btw.

BabaIsYou, 2023-01-18:
> Would be great to give it another go with ZCL and APS_L2.

Checked ZCL and APS_L2 and did a read on Basic Cluster and specific Model ID attributes. Here it is,.
Device NWK is 0xA2C7

```
09:35:37:474 ZCL got data for node=0xC6F3, cl=0x0B04, at=0x0508, status=0x00, type=0x21
09:35:38:364 ZCL read attr 0x00158D00025C9BFF, ep: 0x01, cl: 0x0300, attr: 0x0008, mfcode: 0x0000, aps.id: 50, zcl.seq: 91
09:35:38:493 ZCL got data for node=0x6870, cl=0x0300, at=0x0008, status=0x00, type=0x30
09:35:38:494 ZCL read attr 0x00158D00025C9BFF, ep: 0x01, cl: 0x0300, attr: 0x4000, mfcode: 0x0000, aps.id: 54, zcl.seq: 92
09:35:38:587 ZCL read report config, ep: 0x01, cl: 0x0008, mfcode: 0x0000, aps.id: 61, zcl.seq: 93
09:35:38:590 ZCL got data for node=0x6870, cl=0x0300, at=0x4000, status=0x00, type=0x21
09:35:38:592 ZCL read attr 0x00158D00025C9BFF, ep: 0x01, cl: 0x0006, attr: 0x0000, mfcode: 0x0000, aps.id: 63, zcl.seq: 94
09:35:38:720 ZCL read report config, ep: 0x01, cl: 0x0300, mfcode: 0x0000, aps.id: 73, zcl.seq: 95
09:35:38:732 ZCL got data for node=0x6870, cl=0x0006, at=0x0000, status=0x00, type=0x10
09:35:38:817 ZCL configure reporting ep: 0x01, cl: 0x0300, mfcode: 0x0000, aps.id: 82, zcl.seq: 96
09:35:39:089 ZCL got data for node=0xC6F3, cl=0x0B04, at=0x0508, status=0x00, type=0x21
09:35:39:364 ZCL read attr 0x086BD7FFFE5B8688, ep: 0x01, cl: 0x0006, attr: 0x0000, mfcode: 0x0000, aps.id: 109, zcl.seq: 97
09:35:39:430 ZCL got data for node=0x749F, cl=0x0006, at=0x0000, status=0x00, type=0x10
09:35:40:363 ZCL read attr 0x500B91400002C436, ep: 0x01, cl: 0x0201, attr: 0x0012, mfcode: 0x0000, aps.id: 115, zcl.seq: 98
09:35:40:467 ZCL got data for node=0x9239, cl=0x0201, at=0x0012, status=0x00, type=0x29
09:35:40:469 ZCL read attr 0x500B91400002C436, ep: 0x01, cl: 0x0201, attr: 0x0025, mfcode: 0x0000, aps.id: 118, zcl.seq: 99
09:35:40:516 ZCL got unsupported status: 0x86 for mandatory attribute
09:35:40:517 ZCL got data for node=0x9239, cl=0x0201, at=0x0025, status=0x86, type=0x00
09:35:40:517 ZCL read attr 0x500B91400002C436, ep: 0x01, cl: 0x0006, attr: 0x0000, mfcode: 0x0000, aps.id: 122, zcl.seq: 100
09:35:40:573 ZCL got unsupported status: 0x86 for mandatory attribute
09:35:40:574 ZCL got data for node=0x9239, cl=0x0006, at=0x0000, status=0x86, type=0x00
09:35:40:596 ZCL read report config, ep: 0x01, cl: 0x0B04, mfcode: 0x0000, aps.id: 127, zcl.seq: 101
09:35:40:647 ZCL configure reporting ep: 0x01, cl: 0x0B04, mfcode: 0x0000, aps.id: 130, zcl.seq: 102
09:35:41:364 ZCL read attr 0x500B91400001CB9B, ep: 0x01, cl: 0x0201, attr: 0x0012, mfcode: 0x0000, aps.id: 135, zcl.seq: 103
09:35:41:451 ZCL got data for node=0x964D, cl=0x0201, at=0x0012, status=0x00, type=0x29
09:35:41:453 ZCL read attr 0x500B91400001CB9B, ep: 0x01, cl: 0x0201, attr: 0x0025, mfcode: 0x0000, aps.id: 138, zcl.seq: 104
09:35:41:460 ZCL read report config, ep: 0x01, cl: 0x0204, mfcode: 0x0000, aps.id: 140, zcl.seq: 105
09:35:41:511 ZCL got unsupported status: 0x86 for mandatory attribute
09:35:41:512 ZCL got data for node=0x964D, cl=0x0201, at=0x0025, status=0x86, type=0x00
09:35:41:513 ZCL read attr 0x500B91400001CB9B, ep: 0x01, cl: 0x0006, attr: 0x0000, mfcode: 0x0000, aps.id: 142, zcl.seq: 106
09:35:41:522 ZCL configure reporting ep: 0x01, cl: 0x0204, mfcode: 0x0000, aps.id: 144, zcl.seq: 107
09:35:41:586 ZCL got unsupported status: 0x86 for mandatory attribute
09:35:41:587 ZCL got data for node=0x964D, cl=0x0006, at=0x0000, status=0x86, type=0x00
09:35:42:364 ZCL read attr 0x500B91400002CA1C, ep: 0x01, cl: 0x0201, attr: 0x0012, mfcode: 0x0000, aps.id: 151, zcl.seq: 108
09:35:42:439 ZCL got data for node=0x9BE9, cl=0x0201, at=0x0012, status=0x00, type=0x29
09:35:42:441 ZCL read attr 0x500B91400002CA1C, ep: 0x01, cl: 0x0201, attr: 0x0025, mfcode: 0x0000, aps.id: 154, zcl.seq: 109
09:35:42:527 ZCL got unsupported status: 0x86 for mandatory attribute
09:35:42:527 ZCL got data for node=0x9BE9, cl=0x0201, at=0x0025, status=0x86, type=0x00
09:35:42:528 ZCL read attr 0x500B91400002CA1C, ep: 0x01, cl: 0x0006, attr: 0x0000, mfcode: 0x0000, aps.id: 158, zcl.seq: 110
09:35:42:628 ZCL got unsupported status: 0x86 for mandatory attribute
09:35:42:628 ZCL got data for node=0x9BE9, cl=0x0006, at=0x0000, status=0x86, type=0x00
09:35:42:638 ZCL read report config, ep: 0x01, cl: 0x0B04, mfcode: 0x0000, aps.id: 164, zcl.seq: 111
09:35:42:677 ZCL configure reporting ep: 0x01, cl: 0x0B04, mfcode: 0x0000, aps.id: 166, zcl.seq: 112
09:35:43:364 ZCL read attr 0x70AC08FFFE6980B9, ep: 0x01, cl: 0x0000, attr: 0x4000, mfcode: 0x0000, aps.id: 171, zcl.seq: 113
09:35:43:527 ZCL got unsupported status: 0x86 for mandatory attribute
09:35:43:528 ZCL got data for node=0x9BF3, cl=0x0000, at=0x4000, status=0x86, type=0x00
09:35:43:529 ZCL read attr 0x70AC08FFFE6980B9, ep: 0x01, cl: 0x0000, attr: 0x4000, mfcode: 0x0000, aps.id: 173, zcl.seq: 114
09:35:43:610 ZCL got unsupported status: 0x86 for mandatory attribute
09:35:43:610 ZCL got data for node=0x9BF3, cl=0x0000, at=0x4000, status=0x86, type=0x00
09:35:44:365 ZCL read attr 0xB0CE1814036BCF99, ep: 0x01, cl: 0x0000, attr: 0x4000, mfcode: 0x0000, aps.id: 181, zcl.seq: 115
09:35:44:442 ZCL got unsupported status: 0x86 for mandatory attribute
09:35:44:443 ZCL got data for node=0x9FD3, cl=0x0000, at=0x4000, status=0x86, type=0x00
09:35:44:445 ZCL read attr 0xB0CE1814036BCF99, ep: 0x01, cl: 0x0006, attr: 0x0000, mfcode: 0x0000, aps.id: 183, zcl.seq: 116
[... cut ...]
09:36:01:979 ZCL got data for node=0x6870, cl=0x0006, at=0x0000, status=0x00, type=0x10
09:36:01:980 ZCL read attr 0x00158D00025C9BFF, ep: 0x01, cl: 0x0300, attr: 0x0001, mfcode: 0x0000, aps.id: 158, zcl.seq: 158
09:36:02:123 ZCL got data for node=0x6870, cl=0x0300, at=0x0001, status=0x00, type=0x20
09:36:02:521 ZCL read attr 0x086BD7FFFE5B8688, ep: 0x01, cl: 0x0006, attr: 0x0000, mfcode: 0x0000, aps.id: 168, zcl.seq: 159
09:36:02:585 ZCL got data for node=0x749F, cl=0x0006, at=0x0000, status=0x00, type=0x10
```

BabaIsYou, 2023-01-18:
Well, seems that database is a mess ;-(
In` sensors` table I have some sensors with wrong EP (0x01) that was created when I tested the DDF but with the right modelid (SMT402AD), I also have some sensors with right EP (0x19) and right Modelid, and I also have sensors with right EP (0x19) but wrong Modelid (hxadecimal)
[image]

In ressource_tems is looks like that I don't have resources with EP -ff i have two with EP -19
[image]

How can I clean this ?

Smanar, 2023-01-18:
First, I will try to delete the device using phoscon and deconz.
Then check again the database to see what is still here. But yes it seem you can just remove the "line".

I think the "ressource" table will be clean, but you still have crappy stuff in the "sensors" one.

And BTW don't forget to make a backup before the try ^^.

BabaIsYou, 2023-01-18:
Well, after cleaning database and repairing the device I only have deux sensors for this thermostat with valid EP (0x19). That part seems ok ;-) Thx

But I still have the Modelid in hexadecimal ;-(

BabaIsYou, 2023-01-28:
The value seems not to be constant, sometimes it's `0x534d54343032414400dd590008fffffffff39e0008fffffffff9ffffffffff`  and sometimes `0x534d543430324144000000000000000000dd590008fffffffff39e0008ffff`

Smanar, 2023-01-29:
Lol, it's look like a buffer limit issue.
Only the first part is usefull, but we have random garbage character after.

BabaIsYou, 2023-02-24:
> Lol, it's look like a buffer limit result. The usefull part is up to the null character.
> Only the first part is usefull, but we have random garbage character after, it depend of what happen in the memory if the buffer is not "cleared"

Interesting … may be one can change the reading of this attribute to keep only value before first null … don’t know the incidence.

BabaIsYou, 2023-03-22:
I think that we can close this issue.
I had opportunity to test this device with Z2M and another dongle and even if it is recognized as Stelpro SMT402AD by Z2M, the model displayed stays weird
[image]
Then I guess that it's my thermostat that has a trouble even if it seems functionnal.

BabaIsYou, 2023-03-28:
See above

closed 2023-03-28 by BabaIsYou

opechunka, 2024-05-02:
I have a similar issue with Salus PS600 pipe temperature sensor.  The Model Id is reported as a hex number:

"0x5053363030000020f60120b199000090000000a7750000b7c500009000000090"

Actually the first part  "0x505336303000" is null a terminated representation of "PS600".
A worse thing is that the part after the "0x505336303000" is not stable, some of the numbers randomly change. So I cannot just replace the model id in  DDF file ss909zb_temp_sensor.json 
I have 4 of those sensors. The behavior are the same for all of them. Of cause the second part differs for every of 4 devices.