dresden-elektronik/deconz-rest-plugin #6666: New BW-SHP13 doesn't show energy sensors in Home Assistant

MiguelCosta, 2023-01-11:
## Describe the bug
I have Blitzwolf BW-SHP13 and BW-SHP15 plugs and they have power and energy sensors in my Home Assistant. 
I bought more BW-SHP13 but the sensors don't show up in my Home Assistant.

I added screenshots with all details.

Can you help?

## Steps to reproduce the behavior

  1. Go to deConz
  2. Pair new device
  3. Restart Home Assistant
  4. The device only have a on/off switch

[image]

## Expected behavior

[image]

## Screenshots

[image]

## Environment

 - Host system: Raspberry Pi 4 8GB
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: 26720700
 - deCONZ version: 2.19.03
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? If so: Which? no

## deCONZ Logs

## Additional context

Mimiix, 2023-01-11:
Hi,

They are called all SHP13, but internally they are not the same. Your newest device is not supported by deCONZ yet.

In your case, Please open a new device request. 

Kind regards,

Mimiix, 2023-01-11:
Thinking aobut it: Can you please post all screenshots here of the clusters and Basic cluster.  I'll change the topic label.

MiguelCosta, 2023-01-11:
hi @Mimiix 

thanks for your answer 💪 

Basic
[image]

Simple Metering
[image]

Electrical Measurement
[image]

On/Off
[image]

MiguelCosta, 2023-01-11:
@Mimiix 
Do you need more information?

BabaIsYou, 2023-01-12:
As this device seems to be closer to SHP15 than SHP13 I suggest to try this [DDF](https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/blitzwolf/bw_shp15_smart_plug.json) after replacing the lines : 
```
  "manufacturername": ["_TZ3000_mraovvmm", "_TZ3000_u5u4cakc"],
  "modelid": ["TS011F", "TS011F"],
```
by these : 
```
  "manufacturername": ["_TZ3000_mraovvmm", "_TZ3000_u5u4cakc","_TZ3000_pjcqjtev"],
  "modelid": ["TS011F", "TS011F","TS011F"],
```

Mimiix, 2023-01-12:
> @Mimiix Do you need more information?

Seems fine now. Thanks.

BabaIsYou, 2023-01-17:
@MiguelCosta did you try it ?

MiguelCosta, 2023-01-17:
@BabaIsYou I am using the Home Assistant Addon... how can I add the DDF file? I think the addon doesn't have permission to add DDF files...

BabaIsYou, 2023-01-18:
Could this help : https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-home-assistant-using-text-editor/1839 ?

MiguelCosta, 2023-01-18:
@BabaIsYou I am using HAOS, I don't have docker commands

Mimiix, 2023-01-18:
You do, home assistant uses docker.

MiguelCosta, 2023-01-18:
@Mimiix and @BabaIsYou 

HAOS doesn't allow to use docker commands

[image]

And portainer is also deprecated: 
https://github.com/hassio-addons/addon-portainer

https://community.home-assistant.io/t/hassio-add-on-deconz-edit-ddf-add-on-filesystem-access/449220

Mimiix, 2023-01-18:
In that case, check with HA how to proceed on this.

MiguelCosta, 2023-01-18:
@Mimiix

I think it is related with the existing issues:

https://github.com/dresden-elektronik/deconz-rest-plugin-v2/issues/14

https://github.com/home-assistant/addons/issues/2733

Mimiix, 2023-01-20:
First one is not a issue or bug, simply a feature log in where they discuss.

The second one I am not sure about . Nevertheless, this is a issue with the addon and not deCONZ. You really need to check with ha.

Zehir, 2023-01-23:
@MiguelCosta @BabaIsYou , you can't edit DDF in home assistant because they get replaced on the boot of the addon.

You will need to create a new one in the `users's DDF` directory. It's located at `/mnt/data/supervisor/addons/data/core_deconz/.local/share/dresden-elektronik/deCONZ/devices`
So copy the existing one and copy it in the directory above and then edit the file content.

To access this hard to reach directory you can use this technique :
https://community.home-assistant.io/t/deconz-support-management-of-ddf-files-through-ha-config/370331/21?u=zehir
Note: while you are running the docker command HA OS will complain about unsuported software running but you can ignore the message.

This require the access to the debug SSH of HA on port 22222, not the SSH addon, it's not the same.
You can use this addon to enable it : https://github.com/adamoutler/HassOSConfigurator

MiguelCosta, 2023-01-23:
@Zehir thanks, I will try your suggestion and I will post here the result 🙂 

@Mimiix I didn't say they are bugs, in github is called "issue", so I only used the same word...

```
# issue 14
https://github.com/dresden-elektronik/deconz-rest-plugin-v2/issues/14

# issue 2733
https://github.com/home-assistant/addons/issues/2733
```