dresden-elektronik/deconz-rest-plugin #6670: Negative Level & Brightness Philips Hue Light Strip

TimoGoerlich, 2023-01-12:
## Describe the bug

I am running Deconz on a Proxmox Server via iobroker using a ConBee II stick.
Since update to 2.20.00 I am getting an error now for a connected Philips Hue White & Color Ambiance Light Strip.
The value .level and .bri are somehow automatically set to -2, resp. -1, which is reported as a warning in the log.
I completely removed the lightstrip and re-added it, but that did not help. Also manually changing those values does not help, as they get set again automatically once i activate or deactivate the strip (e.g. via Alexa routine).

## Steps to reproduce the behavior
1. Add Philips Hue White & Color Ambience Light Strip to Deconz
2. Activate the Light Strip

## Screenshots
[image]

## Environment

 - Host system: PC / Proxmox Server
 - Running method: Virtual Machine
 - Firmware version: 26720700
 - deCONZ version: 2.20.00 / 19.9.2022
 - Device: ConBee II
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? no

## deCONZ Logs
[image]

Smanar, 2023-01-13:
>Since update to 2.20.00

So it was working previously ?
Can you show the device json ? From for exemple Phoscon/help/API INformation/lights

TimoGoerlich, 2023-01-13:
Hi,

thanks for trying to help. Below you find the json. I also noticed following log information regarding "gradient" when trying to change brightness or color, might be related to this?

[image]

`{
    "capabilities": {
        "alerts": [
            "none",
            "select",
            "lselect",
            "blink",
            "breathe",
            "okay",
            "channelchange",
            "finish",
            "stop"
        ],
        "bri": {
            "min_dim_level": 0.01
        },
        "color": {
            "ct": {
                "computes_xy": true,
                "max": 500,
                "min": 153
            },
            "effects": [
                "none",
                "colorloop",
                "candle",
                "fireplace",
                "loop",
                "sunrise"
            ],
            "gamut_type": "C",
            "gradient": {
                "max_segments": 9,
                "pixel_count": 16,
                "pixel_length": 1250,
                "styles": []
            },
            "modes": [
                "ct",
                "effect",
                "gradient",
                "hs",
                "xy"
            ],
            "xy": {
                "blue": [
                    0.1532,
                    0.0475
                ],
                "green": [
                    0.17,
                    0.7
                ],
                "red": [
                    0.6915,
                    0.3083
                ]
            }
        }
    },
    "colorcapabilities": 31,
    "config": {
        "bri": {
            "execute_if_off": true,
            "startup": "previous"
        },
        "color": {
            "ct": {
                "startup": 336
            },
            "execute_if_off": true,
            "gradient": {
                "reversed": false
            },
            "xy": {
                "startup": [
                    0,
                    0
                ]
            }
        },
        "groups": [],
        "on": {
            "startup": "previous"
        }
    },
    "ctmax": 500,
    "ctmin": 153,
    "etag": "4486cf2df54a0a21a54a0a5cca86cbc7",
    "hascolor": true,
    "lastannounced": null,
    "lastseen": "2023-01-13T19:18Z",
    "manufacturername": "Signify Netherlands B.V.",
    "modelid": "LCX004",
    "name": "Lightstrip Sideboard",
    "productid": "Philips-LCX004-1-GALSECLv1",
    "state": {
        "alert": "none",
        "bri": -2,
        "colormode": "xy",
        "ct": 383,
        "effect": "none",
        "gradient": {},
        "hue": 2182,
        "on": true,
        "reachable": true,
        "sat": 254,
        "xy": [
            -0.0014,
            0.3417
        ]
    },
    "swconfigid": "F03CAF4D",
    "swversion": "1.97.3",
    "type": "Extended color light",
    "uniqueid": "00:17:88:01:0b:90:01:34-0b"
}`

ebaauw, 2023-01-14:
Interesting.  Also note that `xy` has a negative value for x.  My guess would be that the values are interpreted as signed instead of unsigned on your hardware/VM.  I remember running into signed vs unsigned char issues in the button maps a long time ago.

Can you please check in the GUI what the values are for:
- The _Current Level_ attribuite in the _Level Control_ cluster?
- The _Current X_ attribuite in the _Color Control_ cluster?
- The _Light State_ attribute in the _Hue Effects_ cluster?

Next, could you please try and force-read each of these attributes and check what the API reports after that.  My guess is that after reading _Current Level_ and _Current X_, the API would show the correct values, but after reading _Light State_ it would show the negative values again.  If that's indeed the case, I fear this might be an issue with the Javascript engine on your platform.

Could you also check if the web socket notifications contain the negative values?

TimoGoerlich, 2023-01-14:
Thanks for trying to help. Unfortunately I am not a big expert here, I dont know how exactly to follow your first 2 requests

> 
> Can you please check in the GUI what the values are for:
> 
> * The _Current Level_ attribuite in the _Level Control_ cluster?
> * The _Current X_ attribuite in the _Color Control_ cluster?
> * The _Light State_ attribute in the _Hue Effects_ cluster?
> 
Sorry what you mean with GUI?

> Next, could you please try and force-read each of these attributes and check what the API reports after that. My guess is that after reading _Current Level_ and _Current X_, the API would show the [...]
Do you mean force-set values via iobroker? What happens that if i set either xy, bri or level, it takes the value and also controls the light correct, but randomly after like 1-30 seconds it creates those negative values again. 

> Could you also check if the web socket notifications contain the negative values?

The values in iobroker currently look like this:
[image]

and this is what is in the websocket:
`{
    "19:51:16:844": {
        "capabilities": {
            "alerts": [
                "none",
                "select",
                "lselect",
                "blink",
                "breathe",
                "okay",
                "channelchange",
                "finish",
                "stop"
            ],
            "bri": {
                "min_dim_level": 0.01
            },
            "color": {
                "ct": {
                    "computes_xy": true,
                    "max": 500,
                    "min": 153
                },
                "effects": [
                    "none",
                    "colorloop",
                    "candle",
                    "fireplace",
                    "loop",
                    "sunrise"
                ],
                "gamut_type": "C",
                "gradient": {
                    "max_segments": 9,
                    "pixel_count": 16,
                    "pixel_length": 1250,
                    "styles": []
                },
                "modes": [
                    "ct",
                    "effect",
                    "gradient",
                    "hs",
                    "xy"
                ],
                "xy": {
                    "blue": [
                        0.1532,
                        0.0475
                    ],
                    "green": [
                        0.17,
                        0.7
                    ],
                    "red": [
                        0.6915,
                        0.3083
                    ]
                }
            }
        },
        "e": "changed",
        "id": "60",
        "r": "lights",
        "t": "event",
        "uniqueid": "00:17:88:01:0b:90:01:34-0b"
    },
    "19:51:16:954": {
        "capabilities": {
            "alerts": [
                "none",
                "select",
                "lselect",
                "blink",
                "breathe",
                "okay",
                "channelchange",
                "finish",
                "stop"
            ],
            "bri": {
                "min_dim_level": 0.01
            },
            "color": {
                "ct": {
                    "computes_xy": true,
                    "max": 500,
                    "min": 153
                },
                "effects": [
                    "none",
                    "colorloop",
                    "candle",
                    "fireplace",
                    "loop",
                    "sunrise"
                ],
                "gamut_type": "C",
                "gradient": {
                    "max_segments": 9,
                    "pixel_count": 16,
                    "pixel_length": 1250,
                    "styles": []
                },
                "modes": [
                    "ct",
                    "effect",
                    "gradient",
                    "hs",
                    "xy"
                ],
                "xy": {
                    "blue": [
                        0.1532,
                        0.0475
                    ],
                    "green": [
                        0.17,
                        0.7
                    ],
                    "red": [
                        0.6915,
                        0.3083
                    ]
                }
            }
        },
        "e": "changed",
        "id": "60",
        "r": "lights",
        "t": "event",
        "uniqueid": "00:17:88:01:0b:90:01:34-0b"
    }
}`

snozzlebert, 2023-01-24:
I've a similar experience for a GU 10 Extended color light with firmware 1.101.7. 
Issue occured after updating the deCONZ to the latest version with deCONZ 2.20.1
Brightness, color temperature, and the x of xy are negative in the API response:

```
{
        "manufacturername": "Philips",
        "modelid": "5063331P7",
        "name": "",
        "productid": "Philips-LCG002-3-GU10ECLv2",
        "state": {
            "alert": "none",
            "bri": -2,
            "colormode": "ct",
            "ct": -69,
            "effect": "none",
            "hue": 7726,
            "on": true,
            "reachable": true,
            "sat": 197,
            "xy": [
                -0.0007,
                0.4153
            ]
        },
        "swconfigid": "D779D146",
        "swversion": "1.101.7",
        "type": "Extended color light",
...
    }
```

ebaauw, 2023-01-24:
No, it’s not the JavaScript engine itself, but a C++ helper function that implements `ZclFrame.at()` called from JavaScript.  It returns a `char`, which defaults to unsigned on some systems, but signed on others.  On these systems `bri` values > 127, or `ct`, raw `x`, or raw `y` values > 32767, are treated as negative.

snozzlebert, 2023-01-25:
I'm using the ubuntu/stable/deconz-2.20.01-qt5.deb version in a linux 5.15.80 x86_64 VM.
Restore to 2.18.2  (Qt Javascript engine) fixes the issue.

ebaauw, 2023-01-26:
The fix is because that version doesn’t yet use a DDF and corresponding JavaScript for Hue lights.

closed 2023-01-30 by manup