dresden-elektronik/deconz-rest-plugin #6671: Sonoff Zbminil2 extreme zigbee

MiAutomations, 2023-01-13:
- Product name: sonoff zbminil2 extreme zigbee
- Manufacturer: Sonoff
- Model identifier: ZBMINIL2
- Device type : Switch

[https://itead.cc/product/sonoff-zbminil2-extreme-zigbee-smart-switch-no-neutral-required/](url)

[image]

Can you please add this new device to be supported ? 

Thank you and best regards

Mimiix, 2023-01-13:
Hi,
Please add all requestes screenshots.

K

MiAutomations, 2023-01-13:
> l requestes screens

Sorry please the node Info also as requested 

[image]

Mimiix, 2023-01-13:
We need screenshots for every cluster.

MiAutomations, 2023-01-13:
> We need screenshots for every cluster.

Ok

[image]
[image]
[image]
[image]
[image]
[image]
[image]

Please let me know if you need any further info 

Thank you

Mimiix, 2023-01-13:
Please put them in the Original Post as we ask in the template.

Also make sure all attributes are read and have populated values.

MiAutomations, 2023-01-13:
> Please put them in the Original Post as we ask in the template.
> 
> Also make sure all attributes are read and have populated values.

Done 

Thank you

Smanar, 2023-01-13:
Hello, What this device have more than the model "ZBMINI-L" ?

Here it's the same DDF but adapted for your device

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "SONOFF",
  "modelid": "ZBMINIL2",
  "product": "ZBMINIL2",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_LIGHT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0006",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 5,
          "max": 1800
        }
      ]
    }
  ]
}
```

MiAutomations, 2023-01-13:
> his device have more than the model "ZBMINI-L" ?

Sorry I don't know the diferences

Regarding the DDF 

Where I can use it? 

Thank you

Smanar, 2023-01-13:
Depend of your OS.
If you have HA https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-home-assistant-using-text-editor/1839 (the site is out ATM)
Else https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet (to resume you just need to create a file called what_you_want.json with this contain in a "devices" folder.

ge8020, 2023-02-04:
also would like to have the zbmini-l2 working

Smanar, 2023-02-04:
Can test the DDF and give return.

tbrausch, 2023-02-04:
I copied the ddf for the Sonoff ZMMINIL and modified the model number to ZMMINIL2 and it worked perfectly.

Smanar, 2023-02-04:
@tbrausch ok, thx, so have made a change to uspport both device with the same DDF https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6726

closed 2023-03-05 by github-actions[bot]

Tjchispas, 2023-04-01:
[image]
this works for me

djgreko, 2023-04-05:
I have been waiting for the new release but this device is not supported yet. Moreover, it is not easy for an average user to be able to load DDF for unsupported devices. Are you saying there is any hope? I have had these devices for months and cannot use them.
Thank you

Smanar, 2023-04-05:
On the PR was marked for v2.22.0-beta.
But yes, you can create yourself the DDF, it's just a text file https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6726