dresden-elektronik/deconz-rest-plugin #6675: Danfoss Ally Hermostat eTRV0103

phlexss, 2023-01-16:
## Device

- Product name: Danfoss Ally  Thermostat
- Manufacturer: Danfoss
- Model identifier: eTRV0103
- Device type : Please remove all unrelated device types. 
  - Thermostat

## Screenshots

[image]
[image]

### Basic
[image]

### Identify
[image]

### Diagnostics
[image]

### Other clusters that are not mentioned above

### Power Configuration
[image]

### Poll Control
[image]

### Thermostat
[image]

### Thermostat User Interface Configuration
[image]

phlexss, 2023-01-16:
Additional info:

[image]

Smanar, 2023-01-16:
Hello, can you try this DDF ?
It's just a clone of the TRV100

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": ["Danfoss", "Danfoss"],
  "modelid": ["eTRV0100", "eTRV0103"],
  "vendor": "Danfoss",
  "product": "Ally thermostat",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_THERMOSTAT",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0201"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0301",
        "endpoint": "0x01",
        "in": [
          "0x0001",
          "0x0201",
          "0x0204",
          "0x0B05"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2;",
            "fn": "zcl"
          },
          "read": {
[... 451 lines cut ...]
      "cl": "0x000A",
      "report": [
        {
          "at": "0x0000",
          "dt": "0xE2",
          "min": 3600,
          "max": 10800
        },
        {
          "at": "0x0007",
          "dt": "0x23",
          "min": 3600,
          "max": 10800
        },
        {
          "at": "0x0008",
          "dt": "0xE2",
          "min": 3600,
          "max": 10800
        }
      ]
    }
  ]
}
```

phlexss, 2023-01-16:
I load this ddf in deconz and do a hot reload.....but then what is supposed to happen?

I'm a bit new to loading ddf files.

If I used the phoscon web app to pair a new sensor (the thermostat) it doesn't find it.....but there is a device added to deconz that I see in VNC....this is the one in the captures.

I want to use this device in Home Assistant but it doesn't see it.

Can I use this ddf so deconz will recognize it when I pair the thermostat?

Thank you for your help!

Smanar, 2023-01-16:
>I load this ddf in deconz and do a hot reload.....but then what is supposed to happen?

After that if you make a right clic on the node then "edit DDF" you need to see the same one. Else just put the file in a "devices" folder, restart deconz (deconz scan DDF only on start). Then you can re-include the device after.
And like have said mimic, better to check in the api (phoscon/help/api information) or HA, I don't think Phoscon support TRV.

phlexss, 2023-01-16:
Ok...I put de json file in the .../deCONZ/devices file and rebooted my HA......will try to add the thermostat when it's running again....

Should I add it from Phoscon web app or in deconz VNC?

phlexss, 2023-01-16:
Succes! I can now add the thermostat from phoscon webui!

Thanks a lot!

Smanar, 2023-01-16:
No issue to set heatsetpoint ?

I m looking the code and seing this hack

```
                    else if (sensor->modelId() == QLatin1String("eTRV0100") || sensor->modelId() == QLatin1String("TRV001") ||
                             sensor->modelId() == QLatin1String("eT093WRO"))
                    {
                        if (addTaskThermostatCmd(task, VENDOR_DANFOSS, 0x40, data.integer, 0))
                        {
                            updated = true;
                        }
```

Mimiix, 2023-01-16:
@phlexss Please don't close, i'd like to keep it open to make sure all works and we have a PR attached accordingly.

phlexss, 2023-01-16:
You may be right....the temperature is reported correctly and I can change the temp but the thermostat doesn's seem to open or close based on the temperature....

Smanar, 2023-01-16:
You have a display somewhere on the device to see the current setpoint value ?

And, no, I don't think it will be possible only with DDF, so need a complete PR https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6677

Are you able to compile the code to test it ? Else I think it will be fine without tests.

phlexss, 2023-01-16:
The thermostat has a display that shows the set temperature...that's how I know it's receiving the new temp from HA.....but it's not opening or closing based on this.

I have no way to compile the code. Is this a code-change in the deconz plugin or somewhere else?

Smanar, 2023-01-16:
Yep in the deconz plugin.
I can explain you how to do if you have a real OS ?

phlexss, 2023-01-16:
Please explain....but I'm running HA on pi4. I know my way around linux and windows but I guess I can't compile it on the pi4 in the HA os?

phlexss, 2023-01-16:
If it requires Linux I can install this on my VMware workstation....

phlexss, 2023-01-16:
I have installed Ubuntu so I have a real OS now ;-)

Smanar, 2023-01-16:
You have the procedure explained here https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Compiling-the-REST-plugin-for-device-specific-testing

So for you, after having installed deconz (The code is based on 2.20 version, but will probably work on old one):

The next command line are not dangerous, they just install, lot of files.
Them launch the compilation, If you have a small raspberry, it can freeze during compilation, my pi 3 do, so remove "-j2", can remove it too if the rapsberry is your production machine, to don't freeze it
And the last one replace your original file by the new one, so by security you can make a backup of this one in another folder (rename it don't work) /usr/share/deCONZ/plugins/libde_rest_plugin.so

```
sudo apt install deconz-dev
git clone --branch ally_8 https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

Then you can run deconz (you can close it before the file replacement)

It don't impact your zigbee database, but can use a lot the CPU (at least the first time, compilation will be faster after).

Smanar, 2023-01-17:
Ok so no need.
There is already a PR open for this issue, so just need to wait.

https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6668

phlexss, 2023-01-18:
Ok, thanks for the update.

I have for now removed the device from deconz because it keeps dropping off/reconnecting every few minutes.

I had the same problem with my eTRV0101 thermostats with an earlier version of deconz....I hope the coming update will also fix this for eTRV0103.

phlexss, 2023-01-19:
Here is what I mean with that it keeps reconnecting...

[image]

Smanar, 2023-01-19:
Wich one field is represented on the graph ?

phlexss, 2023-01-19:
The purple line is the target temp, the orange line is when the thermostat is on.

The gaps in the graph means no info could be retrieved.

Smanar, 2023-01-19:
it's "state/on" ? I don't see what you are calling "thermostat is on", it's not just the "heat period" ?

phlexss, 2023-01-19:
Orange is when the thermostat is heating. This one was set to 21 Celcius where the actual temp was aroud 15....so it was always heating.

But the issue is that it keeps losing connection.....which is shown by the gaps in the lines....so during the gaps in the lines HA is not receiving any information.

This image is from a eTRV010 thermostat (no gaps):

[image]

Smanar, 2023-01-19:
But I still don't understand, It can't be just because the device don't send information ?
The device send informaton when it want.

phlexss, 2023-01-19:
I understand what you say.....but it's also showing "unavailable" in my HA panel that shows the thermostat and I then can't control it and can't change it's settings.

See image: the top one is ok, the bottom one is the eTVR0103

[image]

Smanar, 2023-01-19:
So yes it's state/reachable, and for sensor it need to take 24h to happen, so there is a problem yes.
Wich one deconz version are you Using ?

And if in the DDF you set

`"sleeper": true,`

phlexss, 2023-01-19:
I have deconz addon 6.17.1 on HA which is deconz 2.19.3

I will try change in ddf but I can't test until later today or tomorrow.

Thanks for the help so far!

phlexss, 2023-01-25:
I no longer see the (dis)connect issue and it doesn't seem to make a difference if sleeper is set to true or false.

I still can't control the device from HA but I guess I need the fix for that.

Installed 2.20.1 yesterday but the fix is not in there it seems...

Smanar, 2023-01-25:
Nope the Swoops PR is still in waiting list, you can check  the PR to see when it will be included.
For the moment if you want to use the device you can use the library you have compiled.

But strange you can't control it with HA, without the disconnecting issue and with the DDF + the compiled library.

You have an error message ? Perhaps it's a synchronisation issue ? (if the device was added in HA before the good reconisation in the API)

phlexss, 2023-01-25:
I haven't compiled the library....I don't want to experiment to much with my HA installation.

Smanar, 2023-01-25:
Ha ok, so it's for that you can control it yet.

SwoopX, 2023-01-25:
The device only needs the addition of the modelID and manufacturer in the DDF to be fully functional. The change in the plugin is convenience.

Smanar, 2023-01-26:
I don't think.
Without the c++ code edition deconz will use
```

                        if (addTaskThermostatReadWriteAttribute(task, deCONZ::ZclWriteAttributesId, 0x0000, THERM_ATTRID_OCCUPIED_HEATING_SETPOINT, deCONZ::Zcl16BitInt, data.integer))
                        {
                            updated = true;
                        }
```

With the code edition

```
                        if (addTaskThermostatCmd(task, VENDOR_DANFOSS, 0x40, data.integer, 0))
                        {
                            updated = true;
                        }
```

I don't have the device but it seem without the c++ edition it don't work.

SwoopX, 2023-01-26:
I do not really get what you're trying to say. The Ally has roughly 10 configurable items and all of them work for the current modelID and will work for any new modelID which is added to the DDF. The code examples above basically say "either use this or use that", but both will end with the same result.

I have 8 Ally's at home and done pretty much all the implementation around it so one could say I'd know how the device works 🙂

phlexss, 2023-01-27:
@SwoopX: do you own one of the newer eTRV0103 devices? Do you use it/them in HA or some other home automation?

All I can say is that I have 3 eTRV0101 devices that work perfectly in HA and two newer eTRV0103 devices that do not work ok in HA: with the newer ones (that use the new DDF) I can read/set the temp on the device but it doesn't respond correctly to it: I hear the thermostat "move" for a fraction of a second after I set the target temp to a higher temp than it is measuring.....but it should fully open the thermostat in one go as the others do.

If I change the temp manually on the device itself it reacts as it should: it immediately fully opens or closes depending on the set target temp.

In the test I just did the measured temp is 22C and I set the target temp in HA to 30C.....I heard it move for a fraction of a second and in HA it shows up as "heating" (which is good). But if I now increase the temp by 0.5C on the device itself I do hear it go fully open.....which means it wasn't open before.

All this is telling me I can't control the device from HA and there is a significant difference with the eTRV0101. If this has to do with the code in deconz or maybe in HA I don't know....

I have tested the above with two difference eTRV0103 devices.

Smanar, 2023-01-27:
>I do not really get what you're trying to say

Deconz will don't send the same request when the user use the "config/heatpoint" with or without the C++ code edition, and it seem only 1 is working.

faustusdotbe, 2023-02-01:
> All I can say is that I have 3 eTRV0101 devices that work perfectly in HA and two newer eTRV0103 devices that do not work ok in HA: with the newer ones (that use the new DDF) I can read/set the temp [...]

I have one eTRV0103 and I can confirm this behaviour as well, using the DDF above.  Using the [template](https://community.home-assistant.io/t/danfoss-ally-1-18-set-external-temperature-to-radiator-covered-trv/380219/3) below in homeassistant/developer-tools/template never shows more than valve opening of 50%, even when asking ludicrous temperatures such as 30C when it's 18. 

I'm fairly new to HA so I suppose my potential for help will be quite limited, but I'm happy to help debug things if needed.

```  - platform: template
    sensors:
      living_room_trv_valve:
        friendly_name: "Living room TRV valve"
        value_template: "{{ state_attr('climate.cuisine', 'valve')|int }}"
        unit_of_measurement: "%"
        device_class: power_factor
        icon_template: >-
            {%set val= state_attr('climate.cuisine', 'valve')|int %}
            {%if val < 15 %}
              hass:numeric-0-circle-outline
            {%elif val < 25 %}
              hass:numeric-1-circle-outline
            {%elif val < 45 %}
              hass:numeric-2-circle-outline
            {%elif val < 65 %}
              hass:numeric-3-circle-outline
            {%elif val < 85 %}
              hass:numeric-4-circle-outline
            {%else%}
              hass:numeric-5-circle-outline
            {%endif%}
```

SwoopX, 2023-02-01:
Gents, I'm afraid this debate is going nowhere 🙂 

It all started off with the statement: Without C++ changes, the new device is not controllable
**This is just wrong**, everything is controllable solely with the DDF change (confirmed here twice and reflecting what I said).

Then it gets a bit more precise, we're with setting temperatures. The experiences described is a works-as-designed when setting temperatures via the `occupied heating setpoint` attribute. That is part of the official Danfoss documentation. Again, the device is controllable and does something, but expectations do not get met.

To now get a more crisp reaction on setting temperatures, yes, the C++ code must be slightly amended and then the device should meet the initial expectations. This is already part of the respective PR.

faustusdotbe, 2023-02-02:
Thanks for the update @SwoopX . To clarify, what's going to happen now? I see a PR (https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6668) has been merged, this changes a DDF as well as the c++ code. 

My questions would be the following:
- do you know how long this will take to percolate into the official docker image?
- once that's there, do we need to remove the manually-added DDF, or is everything going to be transparent?

Many thanks

SwoopX, 2023-02-09:
Well, typically it takes a day or two for new releases to pop up as docker image as well. Not sure when there'll be a new release, haven't heard anything in that direction yet.

If you need to remove the custom DDF depends a bit on yourself and if you've done anything else. The location where yous is residing now gets precedence over the default and deconz included files...
So, if you wanna go for standard, remove it. If you made some more changes, you probably want to keep it where it is (for the time being).

closed 2023-03-11 by github-actions[bot]

spektab, 2023-03-14:
I have buyed the device 103 mentioned above. I am using deconz 2.20.1

After installing the ddf the device is visible with the Rest API, but within openhab 3.4.2 the thing state is unknown and battery states are not visible.

is there something missing?