dresden-elektronik/deconz-rest-plugin #6676: Robb Smarrt On/Off ROB_200-017-0

phlexss, 2023-01-16:
## Device

- Product name: Robb smarrt smart plug 3680 Watt
- Manufacturer: ROBB smarrt
- Model identifier: ROB_200-017-0
- Device type : Please remove all unrelated device types. 
  - Switch

## Screenshots

### Basic
[image]

### Identify
[image]

### Groups
[image]

### Scenes
[image]

### On/Off
[image]

### Simple Metering
[image]

### Diagnostics
[image]

### Other clusters that are not mentioned above

### Electrical Measurement
[image]

### Temperature Measurement
[image]

Smanar, 2023-01-16:
Hello can you try this DDF
```
{
    "schema": "devcap1.schema.json",
    "manufacturername": "ROBB smarrt",
    "modelid": "ROB_200-017-0",
    "product": "Random tuya plug",
    "sleeper": false,
    "status": "Gold",
    "subdevices": [
      {
        "type": "$TYPE_ON_OFF_PLUGIN_UNIT",
        "restapi": "/lights",
        "uuid": [
          "$address.ext",
          "0x01"
        ],
        "items": [
          {
            "name": "attr/id"
          },
          {
            "name": "attr/lastannounced"
          },
          {
            "name": "attr/lastseen"
          },
          {
            "name": "attr/manufacturername"
          },
          {
            "name": "attr/modelid"
          },
          {
            "name": "attr/name"
          },
          {
            "name": "attr/swversion"
          },
          {
            "name": "attr/type"
          },
          {
            "name": "attr/uniqueid"
          },
          {
            "name": "state/on"
          },
          {
            "name": "state/reachable"
          }
        ]
      },
      {
        "type": "$TYPE_CONSUMPTION_SENSOR",
        "restapi": "/sensors",
        "uuid": [
          "$address.ext",
          "0x01",
          "0x0702"
        ],
        "items": [
          {
            "name": "attr/id"
          },
          {
            "name": "attr/lastannounced"
          },
          {
            "name": "attr/lastseen"
          },
          {
            "name": "attr/manufacturername"
          },
          {
            "name": "attr/modelid"
          },
          {
            "name": "attr/name"
          },
          {
            "name": "attr/swversion"
          },
          {
            "name": "attr/type"
          },
          {
            "name": "attr/uniqueid"
          },
          {
            "name": "config/on"
          },
          {
            "name": "config/reachable"
          },
          {
          "name": "state/consumption",
          },
          {
            "name": "state/lastupdated"
          }
        ]
      },
      {
        "type": "$TYPE_POWER_SENSOR",
        "restapi": "/sensors",
        "uuid": [
          "$address.ext",
          "0x01",
          "0x0b04"
        ],
        "items": [
          {
            "name": "attr/id"
          },
          {
            "name": "attr/lastannounced"
          },
          {
            "name": "attr/lastseen"
          },
          {
            "name": "attr/manufacturername"
          },
          {
            "name": "attr/modelid"
          },
          {
            "name": "attr/name"
          },
          {
            "name": "attr/swversion"
          },
          {
            "name": "attr/type"
          },
          {
            "name": "attr/uniqueid"
          },
          {
            "name": "config/on"
          },
          {
            "name": "config/reachable"
          },
          {
            "name": "state/current"
          },
          {
            "name": "state/lastupdated"
          },
          {
            "name": "state/power"
          },
          {
            "name": "state/voltage"
          }
        ]
      }
    ],
    "bindings": [
      {
        "bind": "unicast",
        "src.ep": 1,
        "cl": "0x0006",
        "report": [
          {
            "at": "0x0000",
            "dt": "0x10",
            "min": 1,
            "max": 1800
          }
        ]
      },
      {
        "bind": "unicast",
        "src.ep": 1,
        "cl": "0x0702",
        "report": [
          {
            "at": "0x0000",
            "dt": "0x25",
            "min": 1,
            "max": 300,
            "change": "0x0000000A"
          }
        ]
      },
      {
        "bind": "unicast",
        "src.ep": 1,
        "cl": "0x0B04",
        "report": [
          {
            "at": "0x0505",
            "dt": "0x21",
            "min": 1,
            "max": 500,
            "change": "0x00000003"
          },
          {
            "at": "0x0508",
            "dt": "0x21",
            "min": 1,
            "max": 300,
            "change": "0x0000000F"
          },
          {
            "at": "0x050B",
            "dt": "0x29",
            "min": 1,
            "max": 300,
            "change": "0x00000003"
          }
        ]
      }
    ]
  }
```

And tell me if some value are not working or need adjustement ?

phlexss, 2023-01-16:
When I load this in deconz I get "Failed to open /data/ddf/robbsmart.json"

...but the file is there.

Mimiix, 2023-01-16:
> When I load this in deconz I get "Failed to open /data/ddf/robbsmart.json"
> 
> ...but the file is there.

Remove the comma trailing on line 95.

Smanar, 2023-01-16:
And DDF need to be in the "devices" folder. Not sure if it work on other.

>Remove the comma trailing on line 95.

It was not already removed 2h ago ?

phlexss, 2023-01-16:
I copied the json to the ...\devices folder and restarted deconz....now I have the power and comsumption sensors!

Thanks a lot!

Smanar, 2023-01-16:
No convertions problem ? no need a *10 or a *100 ?

Mimiix, 2023-01-16:
@phlexss  Please don't close, i'd like to keep it open to make sure all works and we have a PR attached accordingly.

phlexss, 2023-01-16:
Ok....sorry I closed it.

So far it looks like I'm getting the correct power reading....but I'll have to do some testing to be sure....

phlexss, 2023-01-16:
Consumption value doesn't look right...it reports 138.027,725 kWh

Smanar, 2023-01-16:
You know what ? this device respect zigbee feature, and I lost the habit ^^
I m seing on your capture "dividor 3600000", but this value is realy hight, here a new DDF with the dividor

```
{
    "schema": "devcap1.schema.json",
    "manufacturername": "ROBB smarrt",
    "modelid": "ROB_200-017-0",
    "product": "Smart plug 3680 Watt",
    "sleeper": false,
    "status": "Gold",
    "subdevices": [
      {
        "type": "$TYPE_ON_OFF_PLUGIN_UNIT",
        "restapi": "/lights",
        "uuid": [
          "$address.ext",
          "0x01"
        ],
        "items": [
          {
            "name": "attr/id"
          },
          {
            "name": "attr/lastannounced"
          },
          {
            "name": "attr/lastseen"
          },
          {
            "name": "attr/manufacturername"
          },
          {
            "name": "attr/modelid"
          },
          {
            "name": "attr/name"
          },
          {
            "name": "attr/swversion"
          },
          {
            "name": "attr/type"
          },
          {
            "name": "attr/uniqueid"
          },
          {
            "name": "state/on"
          },
          {
            "name": "state/reachable"
          }
        ]
      },
      {
        "type": "$TYPE_CONSUMPTION_SENSOR",
        "restapi": "/sensors",
        "uuid": [
          "$address.ext",
          "0x01",
          "0x0702"
        ],
        "items": [
          {
            "name": "attr/id"
          },
          {
            "name": "attr/lastannounced"
          },
          {
            "name": "attr/lastseen"
          },
          {
            "name": "attr/manufacturername"
          },
          {
            "name": "attr/modelid"
          },
          {
            "name": "attr/name"
          },
          {
            "name": "attr/swversion"
          },
          {
            "name": "attr/type"
          },
          {
            "name": "attr/uniqueid"
          },
          {
            "name": "config/on"
          },
          {
            "name": "config/reachable"
          },
          {
          "name": "state/consumption",
          "read": {"at": "0x0000", "cl": "0x0702", "ep": 1, "fn": "zcl" },
          "parse": {"at": "0x0000", "cl": "0x0702", "ep": 1, "eval": "Item.val = Attr.val / 3600000"},
          },
          {
            "name": "state/lastupdated"
          }
        ]
      },
      {
        "type": "$TYPE_POWER_SENSOR",
        "restapi": "/sensors",
        "uuid": [
          "$address.ext",
          "0x01",
          "0x0b04"
        ],
        "items": [
          {
            "name": "attr/id"
          },
          {
            "name": "attr/lastannounced"
          },
          {
            "name": "attr/lastseen"
          },
          {
            "name": "attr/manufacturername"
          },
          {
            "name": "attr/modelid"
          },
          {
            "name": "attr/name"
          },
          {
            "name": "attr/swversion"
          },
          {
            "name": "attr/type"
          },
          {
            "name": "attr/uniqueid"
          },
          {
            "name": "config/on"
          },
          {
            "name": "config/reachable"
          },
          {
            "name": "state/current"
          },
          {
            "name": "state/lastupdated"
          },
          {
            "name": "state/power"
          },
          {
            "name": "state/voltage"
          }
        ]
      }
    ],
    "bindings": [
      {
        "bind": "unicast",
        "src.ep": 1,
        "cl": "0x0006",
        "report": [
          {
            "at": "0x0000",
            "dt": "0x10",
            "min": 1,
            "max": 1800
          }
        ]
      },
      {
        "bind": "unicast",
        "src.ep": 1,
        "cl": "0x0702",
        "report": [
          {
            "at": "0x0000",
            "dt": "0x25",
            "min": 1,
            "max": 300,
            "change": "0x0000000A"
          }
        ]
      },
      {
        "bind": "unicast",
        "src.ep": 1,
        "cl": "0x0B04",
        "report": [
          {
            "at": "0x0505",
            "dt": "0x21",
            "min": 1,
            "max": 500,
            "change": "0x00000003"
          },
          {
            "at": "0x0508",
            "dt": "0x21",
            "min": 1,
            "max": 300,
            "change": "0x0000000F"
          },
          {
            "at": "0x050B",
            "dt": "0x29",
            "min": 1,
            "max": 300,
            "change": "0x00000003"
          }
        ]
      }
    ]
  }
```

But I think the better way, is take values now, use the device during a day and compare with news values to be sure.

phlexss, 2023-01-16:
I'll keep an eye on it and connect something I know the powerconsumption of before applying the new DDF....

phlexss, 2023-01-19:
I'm seeing weird powerspikes in the history in Home Assistant.....I'm pretty sure this is not what's happening in reality.

[image]

I'm also pretty sure the reported power value is not correct but I still have to do more accurate testing.

Any idea what could be causing this?

Mimiix, 2023-01-19:
Broken device or firmware.

deCONZ doesn't do anything with the values other than doing a multiply or devide depending on the actual values reported.

Smanar, 2023-01-19:
Yeah, and it's strange because we generally have problem with 0 value, but your go from 200 to 800W.
BTW what is the device connected to it ? (from the capture need to divide by 10)

Try this DDF, the power will use a / 10 and skip totally the legacy code.

```
{
    "schema": "devcap1.schema.json",
    "manufacturername": "ROBB smarrt",
    "modelid": "ROB_200-017-0",
    "product": "Smart plug 3680 Watt",
    "sleeper": false,
    "status": "Gold",
    "subdevices": [
      {
        "type": "$TYPE_ON_OFF_PLUGIN_UNIT",
        "restapi": "/lights",
        "uuid": [
          "$address.ext",
          "0x01"
        ],
        "items": [
          {
            "name": "attr/id"
          },
          {
            "name": "attr/lastannounced"
          },
          {
            "name": "attr/lastseen"
          },
          {
            "name": "attr/manufacturername"
          },
          {
            "name": "attr/modelid"
          },
          {
            "name": "attr/name"
          },
          {
            "name": "attr/swversion"
          },
          {
            "name": "attr/type"
          },
          {
            "name": "attr/uniqueid"
          },
          {
            "name": "state/on"
          },
          {
            "name": "state/reachable"
          }
        ]
      },
      {
        "type": "$TYPE_CONSUMPTION_SENSOR",
        "restapi": "/sensors",
        "uuid": [
          "$address.ext",
          "0x01",
          "0x0702"
        ],
        "items": [
          {
            "name": "attr/id"
          },
          {
            "name": "attr/lastannounced"
          },
          {
            "name": "attr/lastseen"
          },
          {
            "name": "attr/manufacturername"
          },
          {
            "name": "attr/modelid"
          },
          {
            "name": "attr/name"
          },
          {
            "name": "attr/swversion"
          },
          {
            "name": "attr/type"
          },
          {
            "name": "attr/uniqueid"
          },
          {
            "name": "config/on"
          },
          {
            "name": "config/reachable"
          },
          {
          "name": "state/consumption",
          "read": {"at": "0x0000", "cl": "0x0702", "ep": 1, "fn": "zcl" },
          "parse": {"at": "0x0000", "cl": "0x0702", "ep": 1, "eval": "Item.val = Attr.val / 3600000"},
          },
          {
            "name": "state/lastupdated"
          }
        ]
      },
      {
        "type": "$TYPE_POWER_SENSOR",
        "restapi": "/sensors",
        "uuid": [
          "$address.ext",
          "0x01",
          "0x0b04"
        ],
        "items": [
          {
            "name": "attr/id"
          },
          {
            "name": "attr/lastannounced"
          },
          {
            "name": "attr/lastseen"
          },
          {
            "name": "attr/manufacturername"
          },
          {
            "name": "attr/modelid"
          },
          {
            "name": "attr/name"
          },
          {
            "name": "attr/swversion"
          },
          {
            "name": "attr/type"
          },
          {
            "name": "attr/uniqueid"
          },
          {
            "name": "config/on"
          },
          {
            "name": "config/reachable"
          },
          {
            "name": "state/current"
          },
          {
            "name": "state/lastupdated"
          },
        {
          "name": "state/power",
          "read": {
            "at": "0x050b",
            "cl": "0x0b04",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x050b",
            "cl": "0x0b04",
            "ep": 1,
            "eval": "Item.val = Attr.val / 10"
          },
          "refresh.interval": 400
        },
          {
            "name": "state/voltage"
          }
        ]
      }
    ],
    "bindings": [
      {
        "bind": "unicast",
        "src.ep": 1,
        "cl": "0x0006",
        "report": [
          {
            "at": "0x0000",
            "dt": "0x10",
            "min": 1,
            "max": 1800
          }
        ]
      },
      {
        "bind": "unicast",
        "src.ep": 1,
        "cl": "0x0702",
        "report": [
          {
            "at": "0x0000",
            "dt": "0x25",
            "min": 5,
            "max": 300,
            "change": "0x0000000A"
          }
        ]
      },
      {
        "bind": "unicast",
        "src.ep": 1,
        "cl": "0x0B04",
        "report": [
          {
            "at": "0x0505",
            "dt": "0x21",
            "min": 5,
            "max": 500,
            "change": "0x00000003"
          },
          {
            "at": "0x0508",
            "dt": "0x21",
            "min": 5,
            "max": 300,
            "change": "0x0000000F"
          },
          {
            "at": "0x050B",
            "dt": "0x29",
            "min": 5,
            "max": 300,
            "change": "0x00000003"
          }
        ]
      }
    ]
  }
```

phlexss, 2023-01-19:
deconz says it failed to open the file.

Smanar, 2023-01-19:
My bad, typo, have updated the DDF

phlexss, 2023-01-19:
I applied new ddf (hot reload)....so far it looks good and power consumption seems a realistic figure now....also no weird spikes yet but that might take more time to properly diagnose.

Will this ddf survive a restart of deconz?

Smanar, 2023-01-19:
>Will this ddf survive a restart of deconz?

If you are using docker and persistent folder yes, else yes.

There is one folder reset every time you update deconz, and one that never move.

phlexss, 2023-01-19:
It looks like it's still showing spikes but only smaller now (makes sense).....it's probably a fault in the hardware because I have another one of these plugs and that doesn't show spikes....

[image]

phlexss, 2023-01-19:
You may be right....there are some heat cords attached (to keep things from freezing)....so I guess they explain the spikes!

You can close it...can you make a PR? ....I have no idea how to do that....

Smanar, 2023-01-19:
No prb, PR done https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6681

Mimiix, 2023-01-19:
> You may be right....there are some heat cords attached (to keep things from freezing)....so I guess they explain the spikes!
> 
> You can close it...can you make a PR? ....I have no idea how to do that....

Heat devices have spikes. The reason is that they aren't on full time but are only on during heating. After it cools it heats again.

closed 2023-02-06 by manup