dresden-elektronik/deconz-rest-plugin #6682: Moes Smart Switch Module 3 Gang MS-104CZ / _TZE3000_pfc7i3kt

BoBiene, 2023-01-20:
## Device

- Product name: Moes Smart Switch Module 3 Gang MS-104CZ
- Manufacturer: _TZE3000_pfc7i3kt
- Model identifier: TS0003
- Device type : 
  - Switch

## Screenshots
[image]
[image]

### Basic
[image]

### Identify
[image]

### Groups
[image]

### Scenes
[image]

### On/Off
[image]

### Simple Metering
[image]

### Electrical Measurement
[image]

Smanar, 2023-01-20:
Just for information, this device need too the "tuya unlock sequence".

BoBiene, 2023-01-20:
Is there anything i can do?

Smanar, 2023-04-13:
If someone can test the new DDF with the tuya unlock sequence ? 

{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZE3000_pfc7i3kt",
  "modelid": "TS0003",
  "product": "Wired Switch 3 gangs (locked)",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_LIGHT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "refresh.interval": 86400,
          "parse": {
            "fn": "zcl",
            "ep": 1,
            "cl": "0x0000",
            "at": "0x0001",
            "script": "tuya_swversion.js"
          },
          "read": {
            "fn": "zcl",
            "ep": 1,
            "cl": "0x0000",
            "at": "0x0001"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        },
        {
          "name": "config/interfacemode",
          "refresh.interval": 86400,
          "read": {
            "fn": "zcl",
            "ep": 1,
            "cl": "0x0000",
            "at": [
              "0x0004",
              "0x0000",
              "0x0001",
              "0x0005",
              "0x0007",
              "0xfffe"
            ]
          },
          "parse": {
            "fn": "zcl",
            "ep": 1,
            "cl": "0x0000",
            "at": "0xfffe",
            "eval": "Item.val = Attr.val"
          },
          "public": false,
          "default": 0
        }
      ]
    },
    {
      "type": "$TYPE_ON_OFF_LIGHT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x02"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "refresh.interval": 86400,
          "parse": {
            "fn": "zcl",
            "ep": 1,
            "cl": "0x0000",
            "at": "0x0001",
            "script": "tuya_swversion.js"
          },
          "read": {
            "fn": "zcl",
            "ep": 1,
            "cl": "0x0000",
            "at": "0x0001"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_ON_OFF_LIGHT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x03"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "refresh.interval": 86400,
          "parse": {
            "fn": "zcl",
            "ep": 1,
            "cl": "0x0000",
            "at": "0x0001",
            "script": "tuya_swversion.js"
          },
          "read": {
            "fn": "zcl",
            "ep": 1,
            "cl": "0x0000",
            "at": "0x0001"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0006",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 5,
          "max": 1800
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 2,
      "cl": "0x0006",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 5,
          "max": 1800
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 3,
      "cl": "0x0006",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 5,
          "max": 1800
        }
      ]
    }
  ]
}

BoBiene, 2023-04-13:
@Smanar thank you. I can test this, do you have a link to the docs where i can read how to use the ddf?

Smanar, 2023-04-14:
Depend of OS

On normal OS https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet
On HA https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-home-assistant-using-text-editor/1839

To resume you just need to create a test file in a "devices" folder with this contain, call the file whatyouwant.json and restart deconz.

BoBiene, 2023-04-15:
Hi, 

i had a typo in my initial request. It should be ``_TZ3000_pfc7i3kt`` instead of ``_TZE3000_pfc7i3kt`` (removing the **``E``**).
After changing the DDF it worked for all three output - thank you very much :)

Smanar, 2023-04-16:
Thx, have created the DDF here https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6902

Have modified the unlock part.

closed 2023-04-17 by manup

sciorty, 2023-04-20:
Since i cannot update the HA add-on with a beta, i manually created the ddf as the one introduced with the pull https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6902.
I loaded the ddf on the device and did both an hot reload and complete system reboot, but my problem reported on https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6886 is still there.
Am I missing something?

Mimiix, 2023-04-20:
@sciorty Perhaps a re-pair?

sciorty, 2023-04-20:
@Mimiix Will try too repair once i get home.
@Smanar i put it on `/usr/share/hassio/addons/data/core_deconz/.local/share/dresden-elektronik/deCONZ/devices` . If i right click on the node->edit DDF, this is what appears, so it should have been loaded
[image]

this is instead the DDF control
[image]

Smanar, 2023-04-20:
On the DDF preview you can see this part ?

```
        {
          "name": "config/interfacemode",
          "refresh.interval": 86400,
          "read": {
            "fn": "zcl",
            "ep": 1,
            "cl": "0x0000",
            "at": [
              "0x0004",
              "0x0000",
              "0x0001",
              "0x0005",
              "0x0007",
              "0xfffe"
            ]
          },
          "parse": {
            "fn": "zcl",
            "ep": 1,
            "cl": "0x0000",
            "at": "0xfffe",
            "eval": "Item.val = Attr.val"
          },
          "public": false,
          "default": 0
        }
```

If yes, the re-inclusion can help, else the tuya unlock sequence is done every 24h

sciorty, 2023-04-20:
No, i copied the file inside the directory as it is, then loaded it. But once I show the DDF, it lacks of the rows you are mentioning, Why?
[image]

Re-pairing did not help.

Smanar, 2023-04-21:
On the HA deconz plugin , need to put the file in `/data/.local/share/dresden-elektronik/deCONZ/devices/`

https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-home-assistant-using-text-editor/1839

sciorty, 2023-04-21:
uhmm i don't have that path, should it start from my raspberry root? 
`/usr/share/hassio/addons/data/core_deconz/.local/share/dresden-elektronik/deCONZ/devices` starts from the raspberry root and is the path where i already put another DDF for another device that was not working and it did the trick (https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6517#issuecomment-1446293557)

Smanar, 2023-04-21:
It's an hidden folder, just check if it exist.

sciorty, 2023-04-21:
It does not exist, I only have these:
[image]
[image]
[image]

Smanar, 2023-04-22:
But you are not on HA ?
They are using docker, it's not as easy as you are doing ^^, it's a realy complicated sytem, take a look on https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-home-assistant-using-text-editor/1839

sciorty, 2023-04-22:
I opened the docker container as you said in https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-home-assistant-using-text-editor/1839/2.
Then went to `/data/.local/share/dresden-elektronik/deCONZ/devices` and this is what appears, it's the same as from HA in `/usr/share/hassio/addons/data/core_deconz/.local/share/dresden-elektronik/deCONZ/devices`:
[image]

At this point I suppose my DDF has been loaded correctly, even if I don't see the part @Smanar is looking for on https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6682#issuecomment-1516523420 ?
Anyway I'm here because @Mimiix closed https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6886, i.e. my problem about the switches going all on/off in group. So I can control the device but all the switches are somehow grouped together!

Smanar, 2023-04-22:
If you don't see the code part, mean something have failed. Docker are something virtual, it's don't work like normal OS, you can see something and after a reboot see other thing.

You need this code part to be able to manage switch independently.

sciorty, 2023-04-22:
Ok but what am i doing wrong, isn’t the folder that one? 😭

Smanar, 2023-04-22:
The folder is good, but you need to go in the docker and copy/paste the code in the created file in `/data/.local/share/dresden-elektronik/deCONZ/devices/`

sciorty, 2023-04-22:
I really don't know what to say:
- deleted the files from both paths
- deleted the node from the deCONZ nodes graph
- factory resetted the device
- rebooted the entire raspberry
- created the DDF .json file inside the deCONZ container `/data/.local/share/dresden-elektronik/deCONZ/devices/`
- paired the device via phoscon
- when entering the editor inside deCONZ, I can see the DDF has not been loaded (correctly):
[image]

- so I opened the new manually created DDF file:
[image]

- to see it has been loaded:
[image]

- but the informations you are saying should appear, do not appear:
[image]

- even if it's different from what I see with the nano editor:
[image]

- even did an hot reload, restart etc. but I'm still stuck.

...maybe it would be easier to erase everything again and install the plugin beta release somehow?

Smanar, 2023-04-23:
4,28 kb size for the file ?
On my side it take 4.82 kio

Still problem on the text file creation.