dresden-elektronik/deconz-rest-plugin #6684: TS0601 TZE200_qoy0ekdb :: Values not read & API not publishing to apps

Ltek, 2023-01-21:
## Describe the bug
1. API is not providing this Device to 3rd party apps (tested: Phoscon and Jowihue plugin in Homeseer)
2. Device values (Temperature and Humidity) are not being entered into deConz -- possibly not being captured by deConz at all for this model

DDF seems to loaded into deConz properly when device is added (https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6543)

## Steps to reproduce the behavior
Added 2 separate devices multiple times. My other model Humi/Temp devices work fine.
1. I'd expect to see the device in all 3rd party apps even if the device values are not populated.
2. 

## Expected behavior
1. Values to be read
5. API to provide sensor to Phoscon and other 3rd party apps

## Screenshots
all node info screens attached

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
 device.
 

## Environment
 - Host system: PC
 - Running method: Windows 11
 - Firmware version: 26400500
 - deCONZ version: 2.20.0 1 / 9/19/2022
 - Device: ConBee I
 - Do you use an USB extension cable: No
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs
Debug Log attached from when I add two of these devices.
[deConz_Debug_Log.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/10472432/deConz_Debug_Log.txt)

Ltek, 2023-01-21:
@Smanar Nice catch!  Looks like a Phoscon bug. 

Phoscon : seems to know about them but it does not display the devices in the control UI (menu : Devices -> Sensors)? I attached the complete info from phoscon / help / Api Information / sensor

Jowihue (Homeseer plugin) works :  I deleted the old devices in Homeseer (Jowihue) and refreshed (told it to rescan deConz) ... it now shows devices properly in Homeseer! (see attached screenshot)

[deConz_PhosconSensors.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/10472678/deConz_PhosconSensors.txt)

Phoscon showing the devices in the info screen (but not in the control UI)...
[image]

Homeseer / Jowihue zigbee plugin pulling from deConz...
[image]

Smanar, 2023-01-21:
It's not suprising for me ^^.
It s a long story, but device in the API can have many name, because 1 device can have 3 sensors.
Deconz use only 1 of them, there is a prioritary list, but you can say it's totaly random ^^.

When you are in phoscon/help/API information, you will the Full API, the same value are used by third app, and here it's ok, you have temperature 84 and temperature 86.

Values in deconz can change at next restart, it will use value in the API.

But honnestly I have non explanation why the device is not visible in phoscon, I don't think they use whitelist for a simple temperature sensor.
Can you share 1 complete json from a not visible device in phoscon ? I will make a post in the forum.

Mimiix, 2023-01-21:
> , I don't think they use whitelist for a simple temperature sensor.

They do.

Ltek, 2023-01-22:
@Smanar not sure what JSON you want me to provide. I'm just using the stock install and DDFs, nothing custom

Smanar, 2023-01-22:
Lol ok, so it seem @Mimiix already know this issue ^^. You know a link on forum where I can put the json ?

Phoscon dev use a kind of whitelist, so if your device don't valid their check it will be not displayed.
For me was logic to display all ZHATemperature ...

Where you are on your capture, on phoscon/help/Api Information/sensors just select one of them and you will have a json with the model id, the manufacture, the type, and ect ....
Honnestly I don't think it will change something as we have realy few return from phoscon devs, but if we say nothing, they can't correct something.

Mimiix, 2023-01-22:
> Lol ok, so it seem @Mimiix already know this issue ^^. You know a link on forum where I can put the json ?
>
> Phoscon dev use a kind of whitelist, so if your device don't valid their check it will be [...]

As far as i know phosocn use a whitelist for everything.

vdiogo, 2023-02-13:
I still have the issue

Smanar, 2023-02-13:
Hello, wich one ?
You can't found the device in phoscon/help/API Information/sensors ?

vdiogo, 2023-02-13:
> 

Correct, it does not show available in the Phoscon App

Smanar, 2023-02-14:
But can you see it in the API using phoscon/help/API Information/sensors ?
Phoscon don't display all possibles devices.

vdiogo, 2023-02-14:
Sorry, noob here. Yes I do see it in the API information 
[image]

Smanar, 2023-02-14:
So it's fine, the problem is from phoscon. You don't use other third app ?
You can try here https://forum.phoscon.de/c/phoscon/17 but not easy to have return from phoscon devs.

vdiogo, 2023-02-14:
Thank you so much for pointing a direction. 
I use Phoscon coz it's the recommended app to use with Home Assistant. 
I just checked a ticket raised for this bug from a while back (which I believe you have commented)  https://forum.phoscon.de/t/help-with-ts0201-ts0601-temp-humidity-sensors-not-showing-in-phoscon-need-ddfs/1998/7
Happy to hear for alternatives

Smanar, 2023-02-14:
>I use Phoscon coz it's the recommended app to use with Home Assistant

But the device is visible in HA ?

vdiogo, 2023-03-18:
> > I use Phoscon coz it's the recommended app to use with Home Assistant
> 
> But the device is visible in HA ?

No, I cannot see it in HA

vdiogo, 2023-03-18:
Can this ticket be opened again please?

Smanar, 2023-03-18:
Not possible a device can be visible in the API and not in HA.
The "type" is correct ? Can you show the complete device json ? ( like before from phoscon/help/API Information)

Mimiix, 2023-03-18:
> Can this ticket be opened again please?

Please try to engage more this time as smanar was simply ignored.

vdiogo, 2023-03-18:
> Not possible a device can be visible in the API and not in HA. The "type" is correct ? Can you show the complete device json ? ( like before from phoscon/help/API Information)

Indeed, you are correct! I can see it in HA with the following sensors (missing temperature):
[image]

Smanar, 2023-03-19:
Ha !
And you miss too the temperature in the API, no sensor with this mac adress and finishing by "-0402" ?
If no, can you try ro re-include the device without deleting it, or if you have the GUI, right clic, edit DDF, and on the menu "hot relaod"

Remember:
- phoscon need to be in permit join for new sensor be added ("add new sensor")
- There is 2 entries in the API, one for the temperature and one for the humidity

vdiogo, 2023-03-19:
> Ha ! And you miss too the temperature in the API, no sensor with this mac adress and finishing by "-0402" ? If no, can you try ro re-include the device without deleting it, or if you have the GUI, [...]

The more I dig following your guidance the more I learn :)
The temperature shows with a different name, but it's the same sensor
[image]

Smanar, 2023-03-19:
So the temperature sensor is visible on API too.
HA have probably missed one, they have different number 51 and 52, the inslusion was probably done in 2 part.
If I m right there is a way to synchronise HA with the API ?

Kane610, 2023-03-19:
I guess the temperature entity would be named sensor.temperature_52 in hass then

vdiogo, 2023-03-20:
> I guess the temperature entity would be named sensor.temperature_52 in hass then

Indeed, you guessed right! Thank you so much for helping me out here

Kane610, 2023-03-20:
> > I guess the temperature entity would be named sensor.temperature_52 in hass then
> 
> Indeed, you guessed right! Thank you so much for helping me out here

You're welcome.

manup, 2023-04-01:
@all looks like this issue is solved, so can be closed?

closed 2023-05-01 by github-actions[bot]