dresden-elektronik/deconz-rest-plugin #6685: URC4450BC0-X-R by Universal Electronics Inc (Xfinity Keypad)

BabaIsYou, 2023-01-21:
## Describe the bug

1. Same as decribed by closed issue #5695, that is for the xfinity keypad XHK1-UE I only get the temperature and and the ancillary control sensor showing up in API (and then in third party apps). May be it's normal but I also waited the presence sensor as described in request  #4758
2.  The alarm [system documentation](https://dresden-elektronik.github.io/deconz-rest-doc/endpoints/alarmsystems/#basics) states that "After pairing a keypad as sensor, it’s automatically added to the default alarm system with id “1”." In my case that's not what happened. The keypad was not added to the default alarm system
[image]
3. The sensors, nor the devices, are prsent in DeConz GUI, only the temperature sensor is present in Phoscon
[image]
[image]

## Steps to reproduce the behavior
Pairing the device to my Zigbee Network

## Expected behavior
Having device present in DeConz GUI, perhaps having the presence sensor alos added and having the keypas added in the default alarm system (or may be could I modify the documentation)
## Screenshots
Se above. I could add others  if needed.

## Environment

 - Host system: Raspberry Pi 
 - Running method: Raspbian 
 - Firmware version: 26720700
 - deCONZ version: 2.19.3 beta)
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? RFXcomm and Z-Wave dongle

## deCONZ Logs
Dont' have log to produce at this time, may be it's normal behavior but not certain of that.

## Additional context

Smanar, 2023-01-21:
Hello, so first phoscon will don't help you here, sorry.
More will be possible using domoticz (but untested, so probably not finished)

About the presence sensor, need to make some search, because your device was added before the DDF core, so I don't remember how it work, but from my memory it was removed https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5352#issuecomment-1186590115

>The alarm [system documentation](https://dresden-elektronik.github.io/deconz-rest-doc/endpoints/alarmsystems/#basics) states that "After pairing a keypad as sensor, it’s automatically added to the default alarm system with id “1”." In my case that's not what happened

Honnestly, I need to check too, but I never see that happen, moslty because users include the keypad before setting the code0 (so configured = false).
As workaround you can add it used the API.

BabaIsYou, 2023-01-21:
Thx !
Didn’t really know if that was bugs or obsolete description/doc …

Most strange for me is that it doesn’t appear in DeConz GUI at all … first time I see that behavior … Phoscon is not a problem for me because I really use it only to have a kind of GUI when I’m fed up with PUT and GET via REST-API or have trouble with DeConz-plugin in Domoticz ;-)

BabaIsYou, 2023-01-21:
__

closed 2023-01-21 by BabaIsYou