dresden-elektronik/deconz-rest-plugin #6686: Device name

yan14, 2023-01-21:
## Device

- Product name: tint LED-Panel Aris white+color.
- Manufacturer: mueller licht 
- Model identifier: no known
- Device type : 
  - Light

## Screenshots

No screenshot, can‘t  even connect

### Basic

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

Couldn‘t even connect, therefor it couldn’t be found in deconz.
With my hue Base it was found in couple of seconds.

product page: https://www.mueller-licht.de/produktinformationen/artikel/?q=404045

Problem was already posted in the board 2021
https://forum.phoscon.de/t/tint-aris-zigbee-by-muller-licht-system-integration/972

yan14, 2023-01-22:
@Mimiix how can I complete this issue, when I can‘t even connect it to deconz?

Mimiix, 2023-01-22:
> @Mimiix how can I complete this issue, when I can‘t even connect it to deconz?

I replied in the forum post you posted.

closed 2023-02-20 by github-actions[bot]