dresden-elektronik/deconz-rest-plugin #6688: Please update firmware in deCONZ package (specifically raspbian deconz-latest.deb)

raoulbhatia, 2023-01-22:
## Describe the bug

The latest RPi package downloaded via http://deconz.dresden-elektronik.de/raspbian/deconz-latest.deb contains old firmware (i.e. `deCONZ_ConBeeII_0x26720700.bin.GCF` from `2021-08-19` vs. `deCONZ_ConBeeII_0x26780700.bin.GCF` from `2022-05-15`, see https://deconz.dresden-elektronik.de/deconz-firmware/ )

## Steps to reproduce the behavior

  1. Download http://deconz.dresden-elektronik.de/raspbian/deconz-latest.deb
  2. Use `dpkg -c deconz-latest.deb | grep -i firmware/`
  3. Check packaged firmware versions against https://deconz.dresden-elektronik.de/deconz-firmware/ and https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Firmware-Changelog

## Expected behavior

Latest firmware is available

## Screenshots

Package:
[image]

Stable firmware download page https://deconz.dresden-elektronik.de/deconz-firmware/
[image]

Current Stable Versions from https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Firmware-Changelog **NEWER** than packaged.

[image]

## Environment

 - Host system: RPi
 - Running method: Docker image from https://github.com/deconz-community/deconz-docker/
 - Firmware version: 0x26720700
 - deCONZ version: (2.xx.yy)
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs

## Additional context

Related issue deconz-community/deconz-docker#153

Spangen, 2023-01-22:
Did a new installation with the premade Raspberry image yesterday. Same older deCONZ_ConBeeII_0x26720700.bin.GCF firmware there.

SwoopX, 2023-01-22:
While it is apparently true that the latest firmware is not contained in the packages, this is neither a bug nor is it an issue of the deconz REST API plugin, which this is the repository for. However, you're invited to post your request in the deconz forum.

closed 2023-01-22 by SwoopX

raoulbhatia, 2023-01-23:
Thanks, @SwoopX for your reply.

I wanted to report this issue and to make it discoverable for other surprised people.
This seemed like the most appropriate repository and being on Github, this made things easy.

However, I do not want to create yet-another account at a forum just for 1 post.
I hope you and everybody else will understand that I'll thus leave it with this report.
Happy if somebody else will take this up and drive it to a resolution.

Cheers,
Raoul