dresden-elektronik/deconz-rest-plugin #6691: Mini Smart Plug _TZ3000_gznh2xla

KoljaHenckel, 2023-01-23:
Hi,
I ordered a Tuya smart Plug 'with smart metering' and was wondering that there is a new chip in it - again- .. :( 
According to https://zigbee.blakadder.com/Tuya_TS011F.html it is compatible with DeConz, and it is wound, but electric measurement is not working.
Hope it is correct to reqeust it here, haven't found the request yet.

## Device

- Product name: Mini Smart Plug 
- Manufacturer: _TZ3000_gznh2xla
- Model identifier: TS011F
- Device type : 
  - Light

## Screenshots
Basic:
[image]

Node Info:
[image]

On/off Attributes:
[image]

Simple Metering:
[image]

Electric Measurement:
[image]

Not sure if this information is relevant:

## Environment
Host system: Raspberry Pi
Running method: HomeAssistant 
  Home Assistant 2023.1.7
  Supervisor 2022.12.1
  Operating System 9.4
  Frontend 20230110.0 - latest
  Firmware version: 266b0700
deCONZ version: 2.19.03 / 19.9.2022
Firmware: 26720700
Device: ConBee II
Do you use an USB extension cable: yes
Is there any other USB or serial devices connected to the host system? If so: no

BabaIsYou, 2023-01-23:
Can you try this DDF to begin with it ? Got it from #6066 

```
{
    "schema": "devcap1.schema.json",
    "manufacturername": "_TZ3000_gznh2xla",
    "modelid": "TS011F",
    "product": "Random tuya plug",
    "sleeper": false,
    "status": "Gold",
    "subdevices": [
      {
        "type": "$TYPE_ON_OFF_PLUGIN_UNIT",
        "restapi": "/lights",
        "uuid": [
          "$address.ext",
          "0x01"
        ],
        "items": [
          {
            "name": "attr/id"
          },
          {
            "name": "attr/lastannounced"
          },
          {
            "name": "attr/lastseen"
          },
          {
            "name": "attr/manufacturername"
          },
          {
            "name": "attr/modelid"
          },
          {
            "name": "attr/name"
          },
          {
            "name": "attr/swversion"
          },
          {
            "name": "attr/type"
          },
          {
            "name": "attr/uniqueid"
          },
          {
            "name": "state/alert",
            "description": "The currently active alert effect.",
            "default": "none"
          },
          {
            "name": "state/on"
          },
          {
            "name": "state/reachable"
          }
        ]
      },
      {
        "type": "$TYPE_CONSUMPTION_SENSOR",
        "restapi": "/sensors",
        "uuid": [
          "$address.ext",
          "0x01",
          "0x0702"
        ],
        "items": [
          {
            "name": "attr/id"
          },
          {
            "name": "attr/lastannounced"
          },
          {
            "name": "attr/lastseen"
          },
          {
            "name": "attr/manufacturername"
          },
          {
            "name": "attr/modelid"
          },
          {
            "name": "attr/name"
          },
          {
            "name": "attr/swversion"
          },
          {
            "name": "attr/type"
          },
          {
            "name": "attr/uniqueid"
          },
          {
            "name": "config/on"
          },
          {
            "name": "config/reachable"
          },
          {
            "name": "state/consumption",
            "read": {"at": "0x0000", "cl": "0x0702", "ep": 1, "fn": "zcl" },
            "parse": {"at": "0x0000", "cl": "0x0702", "ep": 1, "eval": "Item.val = Attr.val * 10"},
            "refresh.interval": 900
          },
          {
            "name": "state/lastupdated"
          }
        ]
      },
      {
        "type": "$TYPE_POWER_SENSOR",
        "restapi": "/sensors",
        "uuid": [
          "$address.ext",
          "0x01",
          "0x0b04"
        ],
        "items": [
          {
            "name": "attr/id"
          },
          {
            "name": "attr/lastannounced"
          },
          {
            "name": "attr/lastseen"
          },
          {
            "name": "attr/manufacturername"
          },
          {
            "name": "attr/modelid"
          },
          {
            "name": "attr/name"
          },
          {
            "name": "attr/swversion"
          },
          {
            "name": "attr/type"
          },
          {
            "name": "attr/uniqueid"
          },
          {
            "name": "config/on"
          },
          {
            "name": "config/reachable"
          },
          {
            "name": "state/current"
          },
          {
            "name": "state/lastupdated"
          },
          {
            "name": "state/power"
          },
          {
            "name": "state/voltage"
          }
        ]
      }
    ],
    "bindings": [
      {
        "bind": "unicast",
        "src.ep": 1,
        "cl": "0x0006",
        "report": [
          {
            "at": "0x0000",
            "dt": "0x10",
            "min": 1,
            "max": 1800
          }
        ]
      },
      {
        "bind": "unicast",
        "src.ep": 1,
        "cl": "0x0702",
        "report": [
          {
            "at": "0x0000",
            "dt": "0x25",
            "min": 1,
            "max": 300,
            "change": "0x0000000A"
          }
        ]
      },
      {
        "bind": "unicast",
        "src.ep": 1,
        "cl": "0x0B04",
        "report": [
          {
            "at": "0x0505",
            "dt": "0x21",
            "min": 1,
            "max": 500,
            "change": "0x00000003"
          },
          {
            "at": "0x0508",
            "dt": "0x21",
            "min": 1,
            "max": 300,
            "change": "0x0000000F"
          },
          {
            "at": "0x050B",
            "dt": "0x29",
            "min": 1,
            "max": 300,
            "change": "0x00000003"
          }
        ]
      }
    ]
  }
```

KoljaHenckel, 2023-01-24:
Thanks a lot.
I copied it over, removed the device and paired it again.
Now there is an power entity, updating every 6 minutes, and consumption entity, that updates every 20 minutes.
The consumption entry seems to be 10x less that it should be
(I have a 40W lamp as a test there for one hour now, according to the plug it consumes about 38-39 W.
The consumption entry now states after 1 hour 0,004 kWh, not 0,04 kWh as it should. Or do I have a misunderstanding?

BabaIsYou, 2023-01-24:
Hi. It seems you’re right. Now that DDF is loaded you can directly change the eval compute from GUI inside DDF editor, (save and hit reload) or you can change the compute formula into the DDF and restart DeConz.

you can just try to replace the consomption formula from `Item.val = Attr.val * 10 ` to `Item.val = Attr.val*100`

KoljaHenckel, 2023-01-25:
Hi,
Not sure why, but after a few reboots and problems with the DeConz container, but no change in the DDF, it seems to be working correct now.
Still wondering what went wrong, but it is ok for the moment.

closed 2023-02-24 by github-actions[bot]