dresden-elektronik/deconz-rest-plugin #6692: After updating to 2.20.1 , my xiaomi motion sensors stay in motion detection state

pergolafabio, 2023-01-24:
## Describe the bug
Hi, i'm using Home Assistant, today received a notification to update addon to 6.18.0 (deconz 2.20.1)...
I have a few Xiamo Motion sensor, and now they are staying in motion detected state? Even restart the addon doesnt help...

1 additional note, those 3 sensors , i applied the pencil hack, so motion detection goes to false after 5 sec

## Steps to reproduce the behavior
Reboot deconz, and notice motion detected state to true

## Screenshots
[image]

## Environment

 - Host system: HassOS
 - Running method: Assistent deCONZ Add-on 
 - Firmware version: (26390500)
 - deCONZ version: (2.20.1)
 - Device: (ConBee I )
 - Do you use an USB extension cable: yes

KennethLavrsen, 2023-01-24:
I do not use HA addon. I run Deconz on bare metal.
And I use Philips Motion sensors. And also Aqara FP1.
I have had same problem for a couple of Deconz versions.

Normally I can walk around the house and trigger the sensors and then they turn off after normal timeout.

It is random which sensors report motion after restart of Deconz

pergolafabio, 2023-01-24:
yeah, i'm not at home right now, but updated remotely
now after restart they detect motion

Maybe if i go home, if i do a real detection, they hopefully go to OFF state?

in your case? You need to detect real motion 1 time so they go to off state after a restart of deconz?

andys73, 2023-01-24:
I run HA OS and have Aqara motion sensors.. my environment looks ok. I have installed the update last night and I have not rebooted the pi 4 since then...

pergolafabio, 2023-01-24:
have you already walked before them? as soon as i restart deconz, they are in motion detected state
not at home, didnt got the the chance yet to walk before them, but never seen this behaviour before

do you also have the pencil hack applied to your sensors to trigger 5 sec instead of 1-2 min ?

ebaauw, 2023-01-24:
> Normally I can walk around the house and trigger the sensors and then they turn off after normal timeout.

Sensors don’t “turn off”; they simply report state.

Some motion sensors, like Hue, are stateful, sending a Zigbee message, when motion is detected, and a second one, when motion is no longer detected.  Typically `config/delay` configures how long the sensor waits to send the second message when it no longer detects motion.

Other motion sensors are stateless, and only send a Zigbee message, when they detect motion (after a cool down period).  For these, deCONZ resets `state/presence` after `config/duration` seconds.  If you restart deCONZ before that time has expired, the sensors resource will continue to report `state/presence`, until deCONZ resets that, `config/duration` seconds after motion is next detected.

Yet other motion sensors, like the IKEA, technically aren’t sensors, but controllers: they send an _On with Timed Off_ command to bound lights, which deCONZ uses to set `state/presence` and (read-only) `config/delay`.  deCONZ resets `state/presence` after `config/delay` seconds, when `config/duration` is 0.  Otherwise it uses `config/duration`.

pergolafabio, 2023-01-24:
When i did the deconz Addon update, i was not home, all sensors were not reporting motion
I did the update, restarted addon, and now 3 motion sensors are in motion state detected true , they still are

Never seen this behaviour before, already done a lots of updates

They cant normally report motion... since there is no one at home (i hope :-) )

Anyway, i report back when i'm home, hopefulle walking before them will turn then to no motion detected state after cooldown

andys73, 2023-01-24:
I do not understand why I am not impacted...

[image]

pergolafabio, 2023-01-24:
well, my sensors are still in detected state, restarted deconz / ha 4 hours ago...

[image]

KennethLavrsen, 2023-01-24:
Read this thread on Home Assistant forum. This issue has been a problem for many for several months. It is not related to the recent release. The problems started probably with the 2.19 series though I actually tried to downgrade to the last 2.18 release and I still have the problem.

The solution in HA land seems to be that more and more give up on Deconz and move to Zigbee2MQTT which does not have this problem. And I must admit I am getting there too.

pergolafabio, 2023-01-24:
what thread? :-)

But never seen this issue before, i always keep my deconz addon up to date :-)

anyway, will report back later when i'm home
i hope they go to "FALSE" once they detected real motion

KennethLavrsen, 2023-01-24:
Sorry - forgot to paste the link to the HA thread

https://community.home-assistant.io/t/restarting-deconz-motion-sensors-are-always-detected/507607/2

I upgraded to 2.20.1 this morning and just tried a restart.

This time no motion sensors were activated but i saw several lights reported on - even if they are not. And they correct themselves within a minute. The problem is that the motion sensors do not correct themselves unless they are triggered and time out again @ebaauw explains above.

I think the issue is that there is a bug in deconz with the saved states. The database of the state is not read correctly or get corrupted maybe? I do not think it has anything to do with the physical sensors. It is deconz that goof up the states of anything at start and then depending on what kind of device you have, it is a small glitch like a light reported on for 10 seconds or a motion sensor that appears on until it gets triggered and times out and goes in sync.

This is really annoying

pergolafabio, 2023-01-24:
ah , ok thnx for posting the link, seems indeed the same issue
still strange that i didnt see it before on previous release ...

i always have my deconz addon uptodate, and must be running 2.19.x too before

```
6.18.0
Bump deCONZ to 2.20.1
6.17.1
Bump deCONZ to 2.19.3
6.17.0
Bump deCONZ to 2.19.1
6.16.0
```

Lets hope they go to "off" again when i"m back home 

my light didt turn on, since i have my automation based on from "off" to "on"  
and now it was more like from "unavaible" to "on"  

so i can live with the issue, altough, still anoying, that means the first time , when deconz is restarted, the light will not work

pergolafabio, 2023-01-24:
I'home, it's back to normal, once they detected real detection, even after restart HA or Deconz....

Finne75, 2023-01-24:
Thanks! See also thread [here](https://forum.phoscon.de/t/xiaomi-devices-weird-behavior-after-deconz-restart/3016).

KennethLavrsen, 2023-01-24:
I am happy that we now have confirmed that there is a bug.

and on "decent motion sensors". Most of mine are Philips Hue. They should be the most decent there is. Also note that when I do the systemctl restart deconz none of them were detecting motion.

thogens, 2023-01-25:
I'm having the same issue with some, but not all of my Aqara motion sensors after using 2.20.1.
After rebooting, lights will be switched "on", as "presence" of these sensors is set to "true" and won't be set to "false" until I'll trigger the sensor once manually.
I've not seen this issue with 2.19x or before.

pergolafabio, 2023-01-25:
yeah, its indeed strange ... seems they need a real motion detection the first time

I now rebooted HA or Deconz several times, and dont notice anymore that they are back to True... 
Maybe i was just when updating ... or when rebooting HassOs completely ,...

rafuz, 2023-01-25:
@ebaauw May it be the same for xiaomi contact sensor? I have noticed that some of my windows sensors default to "open" after a restart of Deconz. 
It will switch to the right status only after opening/closing the sensor at least once.
It happens only on 3-4 sensors on 15 I have but it is quite annoying as one of the windows is in the ceiling.
[image]
This is on Home Assistant deconz plugin and it happens from version 2.19 at least.

ebaauw, 2023-01-25:
Does v2.20 expose the Aqara sensors through DDF or through legacy?  And v2.19?

Smanar, 2023-01-25:
There is DDF for lumi.sensor_motion.aq2

rafuz, 2023-01-25:
I don't know how to check for contact sensors (lumi.sensor_magnet)

Smanar, 2023-01-25:
I have issue on the contact sensor "lumi.sensor_magnet.aq2", but I thought was battery issue, I m trying to found the issue.

MrSolo570, 2023-01-26:
> Read this thread on Home Assistant forum. This issue has been a problem for many for several months. It is not related to the recent release. The problems started probably with the 2.19 series [...]

Can you recommend a way on how to migrate there? I am running deconz add-on with home assistant and I am always afraid of deconz updates. Quite regularly something breaks when updating deconz, just recently I figure illuminance detection for aqara motion sensors got worse...

andys73, 2023-01-26:
ok so I am not going crazy that the iluminance detection for aqara is not reporting correctly. I have some aqara motion sensors that are not reporting at all for a whole day and others that seems to work fine but they are not updating the values as it used to be. It was very reliable some releases ago.
I suspect that ddfs are the cause of this as those devices were moved in ddf. Maybe it is a common problem with the ddf template and not the actual ddf.
Who can debug the code?

andys73, 2023-01-26:
ok so who is going to save us from this instability issue... can be escalated with dresden electronics?

pergolafabio, 2023-01-26:
The easiest way is to revert back to previous version, it was done before

andys73, 2023-01-26:
I reverted to 6 months backup, the oldest I had and it is still 0 lux. more details in https://forum.phoscon.de/t/xiaomi-motion-sensors-stay-in-the-motion-triggered-state/3218/10

KennethLavrsen, 2023-01-26:
andys73. You are highjacking this bug report for an unrelated problem. This bug report is about motion sensors occationally reporting motion after a restart. And this bug was confirmed. The motion sensors work fine after they detect motion once. This bug report addresses the issue with reporting wrong state after restart... sometimes, not always. And I confirmed that I also see it with Philips Hue.

it takes focus away from the original problem when unrelated issues get discussed and we end up with nothing getting fixed

andys73, 2023-01-27:
agreed, sorry. I leave this thread alone and I will log a separate issue.
Thanks

ebaauw, 2023-01-27:
Ok, deCONZ v2.20.1, mutilated the `modelid` in the DDF, forcing the Hue motion sensor to be exposed by legacy code.  No `config/sensitivity`, etc (I did cleanup the legacy code after adding the DDF), but the ZHAPresence sensor is exposed alright.  Shutdown, patch database, restart, and `state/presence` is false, as desired.  I think the `database.cpp` code isn't executed for DDF-managed devices.

I still need to figure out where in the code the state is loaded from the database, but I think @manup would prefer to correct this in the DDFs instead.  Simply adding `"default": false` to the `state/presence` item does the trick.

rafuz, 2023-01-27:
@ebaauw If I understand correctly I think that this fix should be applied also to the DDF of the contact sensor. It displays the same behaviour at deconz reboot (it stays open until you trigger it)
[devices](https://github.com/dresden-elektronik/deconz-rest-plugin/tree/master/devices)/[xiaomi](https://github.com/dresden-elektronik/deconz-rest-plugin/tree/master/devices/xiaomi)/xiaomi_mccgq01lm_openclose_sensor.json

ebaauw, 2023-01-27:
Tricky.  Contact sensors are stateful and it would be far more common to restart deCONZ while a door or window is open, than while a motion sensor detects motion.  Also, contact sensors don't have `config/duration`, whose timeout could be missed due to restarting deCONZ.  I think if you would reset `state/open` on startup, we'll get people complaining that deCONZ doesn't report the door as open, until it's next closed/opened, or the next periodic attribute report arrives (which might be an hour for Xiaomi sensor using the special attribute report; or never for _IAS Zone_ sensors without _Supervision Reports_).

I'm afraid here, the only solution really would be "get a decent sensor" (i.e. one supporting periodic attribute reporting/supervision reports).  I think the Samsumg multisensor is the only one I've seen that does, and it eats batteries like crazy.

BabaIsYou, 2023-01-27:
> tuya/ZY-M100_human_breathing_presence.json

I'did this one, I've checked it. It already has the `"default": false` on `state/presence`

KennethLavrsen, 2023-01-27:
@ebaauw I agree 100% with your analysis on contact sensors like window/door sensors. You would never want to reset those to off after restart. And I never saw any of my aqara window/door sensors acting wrong after restart of deconz. It is only Motion sensors. And I believe I have seen occationally that light groups have been reported on for a second but they correct them selves to off (unless they are indeed on) before I have had a chance to think about it.

I have only had the problem with my Motion sensors. I only have two types in my Zigbee network. Philips Hue (both the indoor and outdoor version) and Aqara FP1 (the millimeter wave sensore which is super cool).

The problem comes and goes. I can reset 10 times and all OK and then suddenly it is not OK on random sensors. And this is in a house with noone home and he sensors all being in no motion state when I reset.

I cannot help thinking that either

- There is rubbish in the database before we reset
- The database is populated with rubbish during reset
- The database is fine but the initial code never reads it and the result is the garbage that happen to be in RAM which in 99% of the time is a value that is picked up as meaning no motion detected.

Surely resetting all motion detectors at boot is a good defensive programming strategy and should be done no matter what because if you reset Deconz after motion is detected, risk is high that the sensors sends its off message while Deconz is resetting and the message will be lost and you end up in a false on state. Your strategy of always setting it to off is correct.

But why do we have garbage in rare cases? And maybe in some specific situation the probability is not 1 % false but 99% false. Maybe it depends on other data in the memory location. I would really look very critically at that init code if I were you.

ebaauw, 2023-01-27:
@rafuz

> I don't understand the "get a decent sensor" as it was working before 

The legacy code reset `state/presence` (for motion sensors) on startup.  Today, this is suppose to be handled in the DDF.  The fact that it wasn't (isn't) is a bug, and adding that is a fix, not a mere workaround.

The legacy code has never reset `state.open`, nor `state.smoke` not any other `state` attribute on startup.  For other sensors, "it" was **not** working before.

> The inteded behaviour for the entity is to restore the previous state from the DB if I understand correctly

The intended behaviour is for deCONZ to reflect the actual sensor state.  The problem is that that's not always possible for all sensors.  That problem is caused by the sensor, not by deCONZ.

As I explain above, some sensors cannot be queried for their actual state, because they turn off their radio to preserve battery power.  Some "light sleepers" wake up every 5 seconds, and appear to be reachable, albeit with a little delay; "deep sleepers" effectively are unreachable, unless an (external) event (like detecting motion) causes them to wake up.  Obviously, light sleepers are more battery hungry than deep sleepers.  This has nothing to do with deCONZ; this is how Zigbee works for (battery-powered) devices without _Receiver On When Idle_.

To cope with this, deCONZ, or any other Zigbee gateway, needs to keep a cached copy of the sensor state, which is updated when the sensor sends state events.  The API reports the "sensor state" from this cached copy.

"Decent" sensors don't just send events on state changes, but also periodically, so deCONZ, or any other gateway can re-synchronise the cached state in case of missed events (e.g. while the gateway was down).  In addition to not sending periodic events, some problematic motion sensors don't even send a state change events when no longer detecting motion (hence the `config/duration` workaround to reset `state/presence` automatically).  For _IAS Zone_ devices, the _Zone Status_ indicates where it sends events when the alarm no longer occurs (_Restore Reports_), and whether it sends periodic reports (_Supervision Reports_).

deCONZ persists the cached copy of the state in the database, but I suppose that's not always flushed immediately.  Especially when deCONZ is killed or crashed, the latest cached state might not be stored correctly.  On controlled shutdowns, the database is flushed, as far as I can tell.

> I have multiple automations that rely on the state of contact sensors and from some months they trigger after a reboot of deconz even if the contact sensor has never moved. 

Please open a separate issue/bug report for these.  We would need to know the type of sensor (and what events it sends) and, if possible, the state in the database.  I suspect there might be cases that deCONZ could handle better (e.g. force-querying light sleepers on startup), but these have been present for a long time, from before the move to DDF, and are not related to this issue particularly with motion sensors.

ebaauw, 2023-01-27:
@KennethLavrsen 

> Philips Hue (both the indoor and outdoor version) and Aqara FP1 (the millimeter wave sensore which is super cool).

The Hue motion sensors are as decent as they get: they wake up every 5 seconds or so, and, in theory could be queries on startup.  deCONZ configures them to report periodically evert 5 minutes (just like the Hue bridge), and they do send "no more motion" events.
The FP1 actually has _Receiver On When Idle_, so is even better in that aspect.

I think when exposed through DDF, deCONZ will actually query the state on startup, unless the device is marked as _Sleeper_.

> I cannot help thinking that either
> - There is rubbish in the database before we reset

I've never seen that, but it might be prudent to change the service starting deCONZ to make a copy of the database for analysis before starting deCONZ.

> - The database is populated with rubbish during reset

The only cases when I've seen that is a corrupted uSD card on the Raspberry Pi.  In that case the copy of the database would also be corrupted, unfortunately.  I think this is where I should remind everyone (including myself) to make (and verify) backups regularly.

> - The database is fine but the initial code never reads it and the result is the garbage that happen to be in RAM which in 99% of the time is a value that is picked up as meaning no motion detected.

The value of a ResourceItem is initialised (to `null`, `"none"`, `0` or `false`) on creation.  So if the database restore wouldn't happen, you would expect these default values for the attributes, not garbage.

@Mimiix 

> I think the additional factor here is that part of deCONZ is not open source (Core) and only visible for DE Employees. That part is (if i am correct) also handling requests so this is not helping either

The (closed source) deCONZ core (with the RaspBee/ConBee firmware) handles the low-level stuff of each incoming and outgoing Zigbee message.  The cached state, database, and restore from database are all handled by the (open source) REST API plugin.  I've never seen any ghost messages originating from the core, that weren't present on the Zigbee network.  Even so, with the right debug settings (`--dbg-aps=2`), the REST API plugins logs every state message it receives from the core, so this would be visible in the log.

rafuz, 2023-01-27:
Thank you for the clarifications, I have partial understanding on how the various sensors works so this is useful for me to try to figure out where could be the problem.

I see that seems that I am the only one experiencing this issue with contact sensors so for sure I will do additional investigation on the status of my sensors (batteries, communication) as I have installed them 3 years ago and joined to deconz.

I will open a new issue when I am sure that the problem is not on my devices.

SwoopX, 2023-01-27:
I got quite some of the devices to be checked and will have a look.

SwoopX, 2023-01-29:
@ebaauw Had a look into that as well and used the frient MOSZB-140 for that. While I absolutely agree that there is not automatic reset to false upon start, as the respective code in database.cpp is not used for DDF powered devices, I cannot confirm your mentioned workaround here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6692#issuecomment-1406339763. I would be very surprised if that works, as it contradicts my understanding of the associated code

Extremely simplified, it should work like this:
- Load DDF
- If no or an invalid value for the 3 functions, default, static or refresh interval is in the DDF, **load it from the generic item**
- Get resource item value
- When a saved resource item value wasn't found and therefore no timestamp was set for the item, then set the default value

`state/presence` has a default value of `false` in the generic item, which should be applicable regardless if _IAS Zone_ or _Occupancy Sensing_ cluster. For step 2 in the example above, the code finds your added default value in the DDF and therefore does not set or overwrite the default value from the generic item.

I fear we need a different approach to force the reset to false, probably when the resource item is loaded? That would kind of mimic the legacy approach.

Just for the records: Develco/frient motion sensors also have supervision and restore reports, so also some decent devices 😉

ebaauw, 2023-01-29:
> I would be very surprised if that works, as it contradicts my understanding of the associated code

It seemed to do the trick for the Hue motion sensor in my testing.

1doubleDD1, 2023-01-30:
Just to add my few cents to the topic... 
I've been having the same problem lately, with my Osram motion senzor (CentraLite Motion Sensor-A is the official name with Firmware: 30.0.83.16), I also have Aqara and Philips Hue motion sensors and they don't appear to be affected with the issue.
After Raspberry restarts Osram motion senzor stays in "detected" state until it's triggered. Since RasPi restarts in the middle of the night, previous state (prior to restart) was definitely "not detected".
I haven't noticed this problem on deConz 2.18.2, not sure about 2.19.1, definitely present on 2.19.3 and now on 2.20.1 (I was on 2.19.1 for a relatively short period).
Hopefully a solution will be found soon.

pergolafabio, 2023-01-30:
Indeed , in my case the issue only occures on cold boot, if I just restart addon, then there is no issue

thogens, 2023-01-31:
confirmed on my end with multiple motion sensors. All show same behaviour.
Only after cold boot, I see my Xiaomi motion sensors being switched to "presence". This causes my lights turn on and won't be switched off again until a new motion gets detected and the state is overwritten.
Before my cold boot, I made sure that all detections were false.
After booting, I see the correct (old) states being shown over the API (using ioBroker plugin).
But after some time, this state gets a new timestamp and is being switched to true.
Interestingly, some devices aren't even connected by that time, when presence is switched to true. At least reachable state was false.
-> Using 2.20.1 with Conbee II, running on Synology NAS with docker.

ebaauw, 2023-01-31:
> After booting, I see the correct (old) states being shown over the API (using ioBroker plugin). 
> But after some time, this state gets a new timestamp and is being switched to true.

Is this ioBroker taking some time to refresh its state after deCONZ restarts, or does the deCONZ API actually change its reported state?  Does deCONZ send a web socket notification?  Which timestamp gets updated?  `lastupdated`?  `lastseen`?  `lastannounced`?  Or a timestamp in ioBroker?  Does it change to the current time (when the change occurs) or to a previous time?

thogens, 2023-01-31:
I did another experiment:
Just rebooted the Deconz container. 
Outcome: In phoscon all motion sensors were set to "presence detected", but "last updated tag still shows old timestamp". 
ioBroker Deconz plugin showed old "not present" data with correct timestamp.
->
After rebooting the Deconz plugin within ioBroker, all states were set to "presence=true".
"lastupdated" has the "real last update" entry (2h ago for example), but in that datapoint there is a new "lastupdated" timestamp stored. (rebooting time).
Does that help, or is it too confusing?

ebaauw, 2023-01-31:
So upon restart, deCONZ reports `presence` with the `lastupdated` value of when motion was last detected.  It would seem that the database doesn’t get updated when deCONZ resets `presence` because of `config/duration`.  That’s probably a bug in its own right, but went unnoticed due the reset in the legacy code upon load from the database.

And, I understand, the ioBroker plugin maintains its own timestamp when it refreshes its state from deCONZ.

thogens, 2023-01-31:
Correct!
Upon restart, it takes 5mins or so until (all of a sudden) all motion sensors are set to presence = true. It really takes a while. Like some sort of timeout?! - Before, they are shown as "not reachable".
The timestamp is indeed showing the timestamp from the last detection = true, not the timestamp of the detection = false. So basically "lastupdated" is overwritten by an old value. Just tried it again.

ebaauw, 2023-01-31:
> It would seem that the database doesn’t get updated when deCONZ resets `presence` because of `config/duration`. 

Indeed, checking the code a call to `sensor->setNeedSaveDatabase(true);` is missing from `checkSensorStateTimerFired()`.  Not just for `state/presence` for ZHAPresence sensors, but also for `state.vibration` for the `lumi.vibration` ZHAVibration sensor and when generating fake x001 hold events for `lumi.sensor_switch` and Friends of Hue switches.

That would explain why the issue occurs so much more than one would expect for just missing a `config.duration` timeout during a restart.

francescopeloi, 2023-02-08:
hi,

sorry to add more noise but after reading this thread I am not sure if the problem described here applies to me too. I've got many Hue motion sensors in my setup and a couple of them started not detecting motion anymore, they say always "Motion: none" on the webapp, even though they show a refreshed timestamp of not to long ago (seconds). Could this be the same issue described in this thread?

SwoopX, 2023-02-09:
@ebaauw Nice catch! Unfortunately, however, I do not believe that solely that change resolves the observed behavior. Based on the tests over here, I've seen quite some presence sensors loading into a `true` state directly after deconz start (while they had a default of `false`, verified with some extended logging).

We should probably discuss with manup when he's back if we want to mimic the force to `false` also in the DDF related code. Shouldn't be too hard after all.

@francescopeloi No, that is something different not belonging here.

rafuz, 2023-02-17:
So, I have some fresh information about this same issue on magnet contact sensor from xiaomi.
This is the situation before reboot:
[image]
Sensors are closed.
This is after:
[image]
The most interesting part is that the sensors that are reading "open" have a last updated timestamp referring to the last time it was opened (yesterday).
[image]

SwoopX, 2023-02-19:
With regard to the timestamp, I guess there's some room for improvement. What happens when a device value is updated is, simplified, that its value will be saved to DB (certain conditions apply to have some throttling). However, the resource item `lastupdated` is not saved in that course of action. That can leas to such a "nonsense" timestamp like you have it here.

For my taste, your issue with the open state is something different that presence, as the mechanism behind it is considerably different. Weird though, no doubt about it.

I had a more closer look on the DB save actions the last days. In my opinion, things work a bit unfortunate regarding order and what's saved. I also feel the save action supposed to kick in upon closure of deconz is not working anymore. I'd have some comparably easy fixes in mind for some issues here, but we need to have manup involved here as well to support spotting the places where stuff goes wrong...

JanLp, 2023-03-04:
Conbee II
Version 2.20.01
Firmware 26780700

I have a problem with RTCGQ14LM ([lumi.motion.ac02](https://dresden-elektronik.github.io/deconz-rest-doc/devices/xiaomi/xiaomi_rtcgq14lm_p1_presence_sensor/#aqara-p1-presence-sensor-rtcgq14lm))

I put e.g. duration to 120

```
{
"duration": 120,
}
```

or sensitivity to 1

```
{
"sensitivity": 1,
}
```

Shortly afterwards, these values are reset to the standard `"duration": 30` and `"sensitivity": 3`, the same applies to `ledindication`

```
{
    "config": {
        "battery": 100,
        "duration": 30,
        "ledindication": false,
        "on": true,
        "reachable": true,
        "sensitivity": 3
    },
    "ep": 1,
    "etag": "a996cf7fa7d89b88ab9ec03b120823fc",
    "lastannounced": null,
    "lastseen": "2023-03-04T11:38Z",
    "manufacturername": "LUMI",
    "modelid": "lumi.motion.ac02",
    "name": "Motion 1",
    "state": {
        "lastupdated": "2023-03-04T11:39:20.044",
        "presence": true
    },
    "swversion": "0.0.0_0006",
    "type": "ZHAPresence",
    "uniqueid": "54:ef:44:10:00:52:c4:d2-01-0406"
}
```

SwoopX, 2023-03-04:
@JanLp Please do not hijack the issue with something which doesn't belong here. You're welcome to ask your question in the forum. Thanks!

JanLp, 2023-03-04:
I thought it was related to 😊 https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6692#issuecomment-1410886917

thogens, 2023-03-08:
I thought there was a fix planned for  2.21.0-beta. But after updating to that version, it still show same behaviour with Aqara/Xiaomi motion sensors. Any update?

JanLp, 2023-03-08:
> I thought there was a fix planned for 2.21.0-beta. But after updating to that version, it still shows same behaviour with Aqara/Xiaomi motion sensors. Any update?

Same wonder here

francescopeloi, 2023-03-08:
and here

manup, 2023-04-01:
There were a bunch of fixes related to motion sensors. Does the problem still occur with current version?

JanLp, 2023-04-01:
> There were a bunch of fixes related to motion sensors. Does the problem still occur with current version?

Yes

[7 more replies]

closed 2023-04-19 by pergolafabio