dresden-elektronik/deconz-rest-plugin #6697: SONOFF SNZB-02D

1Rolf, 2023-01-25:
## Device

- Product name: SONOFF SNZB-02D [LINK](https://itead.cc/product/sonoff-snzb-02d-zigbee-lcd-smart-temperature-humidity-sensor/)
- Manufacturer: SONOFF
- Model identifier: SNZB-02D
- Device type :  - Sensor

## Screenshots

### Basic

[image]

### Identify

[image]

### Power Configuration

[image]

### Poll Control

[image]

### Temperature measurement

[image]

### Relativ humidity measurement

[image]

### On/Off

[image]

### Node Info

[image]

BabaIsYou, 2023-01-25:
May be you can try to start with DDF of SNZB-02 device  ([here](https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/sonoff/snzb-02-multisensor.json)) after modifying lines Manufacturer and Modelid like below

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": ["eWeLink","SONOFF"]
  "modelid": ["TH01","SNZB-02D"]
  "vendor": "SONOFF",
  "product": "SNZB-02",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_TEMPERATURE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0402"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0302",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0402"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "refresh.interval": 86400,
          "read": {
            "at": "0x0006",
            "cl": "0x0000",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0006",
            "cl": "0x0000",
            "ep": 1,
            "eval": "Item.val = Attr.val",
            "fn": "zcl"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "read": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2"
          },
          "default": 0
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/temperature",
          "awake": true,
          "default": 0
        }
      ]
    },
    {
      "type": "$TYPE_HUMIDITY_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0405"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0302",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0405"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "refresh.interval": 86400,
          "read": {
            "at": "0x0006",
            "cl": "0x0000",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0006",
            "cl": "0x0000",
            "ep": 1,
            "eval": "Item.val = Attr.val",
            "fn": "zcl"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "read": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2"
          },
          "default": 0
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/humidity",
          "awake": true,
          "default": 0
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 60,
          "max": 3600,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0405",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x21",
          "min": 300,
          "max": 1800,
          "change": "0x000001F4"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0402",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x29",
          "min": 300,
          "max": 1800,
          "change": "0x00000064"
        }
      ]
    }
  ]
}
```

Smanar, 2023-01-25:
It will depend of your OS.
If you have HA https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-home-assistant-using-text-editor/1839
Else you just need to create a text file called what_you_want.json with this contain in a "devices" folder.

> 
All DDF provided directly with deCONZ typically reside in /usr/share/deCONZ/devices/ on a Linux system and are loaded first. However, files residing in the home directory of the user running deCONZ (e.g. /home/<DECONZUSER>/.local/share/dresden-elektronik/deCONZ/devices) will override the pre-packaged files to allow users to amend and keep their own files if desired.

Then restart deconz and re-include your device

1Rolf, 2023-01-25:
I don't think it should look like that, should it?

[image]

BabaIsYou, 2023-01-26:
Clearly not. Once DDF will be loaded the name of the device will change from 0x)B7*E to something more human readable ;-)
Where did put the DDF file I suggested ?

1Rolf, 2023-01-26:
The DDL file now works, I have recreated it from the linked original file.

The device is now recognised and I also get values BUT I have a very strange update behaviour via the rest interface.

Temperature is displayed but only updated very rarely (every few minutes), even in case of very large changes +/- 5°C.

The same applies to the humidity.

But I get the battery level, the offset and the accessibility updated several times per second.

According to this [video](https://youtu.be/u0nB1olg72w) the update behaviour should be different for Temp/Humidity

BabaIsYou, 2023-01-26:
Hi,

What is expected is a change in humidity when variation is about 5 degrés (0x1f4) or max every 1800 seconds.

For humidity it’s with a variation of 1% (0x64) or maxi every 1800 seconds too.

1Rolf, 2023-01-26:
Sorry im really not familiar with the whole thing. i thougt updates are triggered from the device?!

【High Accuracy & 5s Fast Refresh】Built-in Swiss-made smart hygrometer sensor that gives you more accurate readings maintaining ±0.2°C/±0.4°F accuracy in temperature and ±2%RH accuracy in humidity, refreshes readings every 5 seconds.

So do i need to change this? where?
Thanks!

BabaIsYou, 2023-01-26:
On the binding part of the DDF. Now that the DDF is loaded you can modify it thru the DDF editor in DeConz GUI directly (save and hot reload). Or in the DDF text file with a text editor.

the values to modify are 0x64 that is 500 so 5 degrees for this attribute and 0x64 that is 100 so 1% in that context. 

But keep in mind that the more the device reports, the more it drains the battery.

1Rolf, 2023-01-26:
I think i found it, i get more frequent updates of temperature and humidity but i still get flooded with battery, offset,.. updates (3-4 times every second).
How to stop this? I only get it for this device, others work fine.

BabaIsYou, 2023-01-26:
Well done. You can also play with "min" and "max" value in bind. If threshold (change) does not occur, it givies the intervall at which minimaly or maximaly the attribute must be reported in seconds.

For Battery it's this bind
```
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 60,
          "max": 3600,
          "change": "0x00000001"
        }
```

May be you can play with the 3 last lines by ajusting delai and change. I suggest to begin with : 
          "min": 3600,
          "max": 86400,
          "change": "0x00000002"

> 【High Accuracy & 5s Fast Refresh】Built-in Swiss-made smart hygrometer sensor that gives you more accurate readings maintaining ±0.2°C/±0.4°F accuracy in temperature and ±2%RH accuracy in humidity, refreshes readings every 5 seconds.

BTW this gives you the accuracy of your device, then the minimum value for signifiacant change in bind (not less than 0,2 (0x14) for temp and 2% (0xC8) for humidity). And then min value you could have for report that is 5 seconds. Value below these are useless for this device.

1Rolf, 2023-01-27:
Still flooding,...

Input Debug for 1sec 

27.01.2023, 09:05:03 |             Received | {"DataID":"{D7B089F0-6AFD-8861-2226-07B675D951B1}","Buffer":"{\"config\":{\"battery\":100,\"offset\":0,\"on\":true,\"reachable\":true},\"e\":\"changed\",\"id\":\"24\",\"r\":\"sensors\",\"t\":\"event\",\"uniqueid\":\"30:fb:10:ff:fe:74:81:7e-01-0405\"}"}
27.01.2023, 09:05:03 |             Received | {"DataID":"{D7B089F0-6AFD-8861-2226-07B675D951B1}","Buffer":"{\"config\":{\"battery\":100,\"offset\":0,\"on\":true,\"reachable\":true},\"e\":\"changed\",\"id\":\"23\",\"r\":\"sensors\",\"t\":\"event\",\"uniqueid\":\"30:fb:10:ff:fe:74:81:7e-01-0402\"}"}
27.01.2023, 09:05:03 |             Received | {"DataID":"{D7B089F0-6AFD-8861-2226-07B675D951B1}","Buffer":"{\"config\":{\"battery\":100,\"offset\":0,\"on\":true,\"reachable\":true},\"e\":\"changed\",\"id\":\"24\",\"r\":\"sensors\",\"t\":\"event\",\"uniqueid\":\"30:fb:10:ff:fe:74:81:7e-01-0405\"}"}
27.01.2023, 09:05:03 |             Received | {"DataID":"{D7B089F0-6AFD-8861-2226-07B675D951B1}","Buffer":"{\"config\":{\"battery\":100,\"offset\":0,\"on\":true,\"reachable\":true},\"e\":\"changed\",\"id\":\"23\",\"r\":\"sensors\",\"t\":\"event\",\"uniqueid\":\"30:fb:10:ff:fe:74:81:7e-01-0402\"}"}
27.01.2023, 09:05:03 |             Received | {"DataID":"{D7B089F0-6AFD-8861-2226-07B675D951B1}","Buffer":"{\"config\":{\"battery\":100,\"offset\":0,\"on\":true,\"reachable\":true},\"e\":\"changed\",\"id\":\"24\",\"r\":\"sensors\",\"t\":\"event\",\"uniqueid\":\"30:fb:10:ff:fe:74:81:7e-01-0405\"}"}
27.01.2023, 09:05:03 |             Received | {"DataID":"{D7B089F0-6AFD-8861-2226-07B675D951B1}","Buffer":"{\"config\":{\"battery\":100,\"offset\":0,\"on\":true,\"reachable\":true},\"e\":\"changed\",\"id\":\"23\",\"r\":\"sensors\",\"t\":\"event\",\"uniqueid\":\"30:fb:10:ff:fe:74:81:7e-01-0402\"}"}
27.01.2023, 09:05:03 |             Received | {"DataID":"{D7B089F0-6AFD-8861-2226-07B675D951B1}","Buffer":"{\"config\":{\"battery\":100,\"offset\":0,\"on\":true,\"reachable\":true},\"e\":\"changed\",\"id\":\"24\",\"r\":\"sensors\",\"t\":\"event\",\"uniqueid\":\"30:fb:10:ff:fe:74:81:7e-01-0405\"}"}

BabaIsYou, 2023-01-27:
Hmmm can you try with higher value in change for battery binding, like 0x1f4 ?
On DDF editor, preview gives you the last DDF changes ?

1Rolf, 2023-01-27:
I don't know why but after the last change you suggestet the DDL file seemd to be broken so i restarted from scratch.

I had a look around and found this (in two places).

[image]

I don't know what it is for, but it was set as 0, i changed it to 3600.
Looks like the flooding is stopped (at least for the moment).

BabaIsYou, 2023-01-27:
> I don't know what it is for, but it was set as 0, i changed it to 3600.

It's better to have it in "bind" part when you can combine treshold and min/max interval in place of "read" section where you just force a read at a specified interval whatever value the attribute has. And it dosen't replace bind as far as I know.

1Rolf, 2023-01-27:
The Bind part is in the DDF, i also could see the three entries in the reporting tab in the DDF editor.
But changes there (either in the file or in the ddf editor) had no affect, the flooding continued.

The only thing that stopped the flooding so far was changing the value above (interval)

Here's my current DDF
`
{
  "schema": "devcap1.schema.json",
  "manufacturername": "SONOFF",
  "modelid": "SNZB-02D",
  "vendor": "SONOFF",
  "product": "SNZB-02D",
  "sleeper": true,
  "status": "Gold",
  "path": "/devices/SNZB-02D.json",
  "subdevices": [
    {
      "type": "$TYPE_TEMPERATURE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0402"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0302",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0402"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "refresh.interval": 3600,
          "read": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2"
          },
          "default": 0
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/temperature",
          "awake": true,
          "default": 0
        }
      ]
    },
    {
      "type": "$TYPE_HUMIDITY_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0405"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0302",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0405"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "refresh.interval": 3600,
          "read": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2"
          },
          "default": 0
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/humidity",
          "awake": true,
          "default": 0
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 3600,
          "max": 7200,
          "change": "0x00000002"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0405",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x21",
          "min": 5,
          "max": 1800,
          "change": "0x00000064"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0402",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x29",
          "min": 5,
          "max": 1800,
          "change": "0x0000000A"
        }
      ]
    }
  ]
}
`

BabaIsYou, 2023-01-28:
It seems that you let the change value in the bind part for the battery to 0x00000002 that could perhaps be very low if the sensor interpret it as 0,2 or 0,02 degree. Could you delete the interval in read and change this value to something higher like 0x1f4 as suggested above ?

Smanar, 2023-01-28:
Yes ,the bind seem fine for me

```
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 3600,
          "max": 7200,
          "change": "0x00000002"
        }
      ]
    },
```

Perhaps unsupported value by the device, so it have switched to poll mode (that use , and need, the "refresh.interval")

closed 2023-02-26 by github-actions[bot]

sevrinrobstad, 2023-04-01:
Was there made a PR for this device? I need it as well :)

Smanar, 2023-04-01:
Hu ?
This device already have a DDF, it don't work for you ? wich one deconz version have you ?

sevrinrobstad, 2023-04-01:
I googled and found this thread before actually trying.
Updated deconz and tried now, everything OK! Thanx.

antocpu, 2025-02-14:
Hello,

I am writing regarding an issue with a Sonoff SNZB-02D sensor.

I followed the pairing procedure through the Phoscon deCONZ interface, following the instructions to enable pairing mode. Although the device was immediately recognized by the system, I am experiencing a malfunction: both temperature and humidity sensors are only displaying null values (0).

 
[image]

I'm using a conbee II with 2.29.1

Does someone have an idea on how I could make it works ?

note that I tried to delete the device, remove the battery and the pair it again, which did not fixed the problem.

Many thanks,

Tanddant, 2025-04-22:
@antocpu did you manage to get your issue resoleved?

antocpu, 2025-04-23:
Hello @Tanddant 

Yes, in fact, I did 2 things :

1. I a some layer of paper in the battery compartment. Because, I thing that the cover is too thin and thus the contact with the batteries is sometimes lost.
2. I re-paired the device getting closer to my conbee II stick

Cleraly, as this is a end device, the range is not so wide, espacially if you do not have a device that is also a router near it.

This 2 actions solved the issue and all is working like a charm by now.