dresden-elektronik/deconz-rest-plugin #6700: Aqara original motion sensor not reporting correctly illuminance in some sensors

andys73, 2023-01-27:
## Describe the bug
The aqara original motion sensor does not seems to report illuminance level correctly, reporting 0 or very low values. Some of the same sensor do report correct values which is quite odd.

## Steps to reproduce the behavior
1. Remove and repair motion sensor that showed 0 all day in phoscon. It still reports 0lux.
2. Repair the same motion sensor in an aqara hub ( I used a G2H camera). It reports correctly 

## Expected behavior
Consistent reporting of lux level as is used in automations.

## Screenshots
Phoscon reporting
[image]
Home assistant reporting
[image]
Aqara hub reporting
[image]
## Environment

 - Host system: Raspberry Pi
 - Running method: Home Assistant deCONZ Add-on
 - Firmware version: (26780700)
 - deCONZ version: (2.20.01)
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? USB SSD, Bluetooth dongle (extension away from conbee 2), UPS USB interface

## deCONZ Logs

## Additional context
Large network a combination of Aqara (motion and temperature), Lidl (plugs and extensions), Osram (plugs), Sonos ZBMINI, Heiman (plugs and smoke sensors),  Philips Hue (One motion sensor)

BabaIsYou, 2023-01-27:
Hmmm ... difficult to make the slightest diagnosis if we don't even know what equipment it is ... model Identifier, manufacturer anme ... some more precise information than just "Aquara Sensor" ;-)

andys73, 2023-01-27:
[image]

Smanar, 2023-01-27:
I m making tries on my side ATM, same model "lumi.sensor_motion.aq2" , no problem for the moment, still on 2.19 version.
Will try tommorow on 2.20

andys73, 2023-01-28:
I just cannot figure out why only some sensors are not reporting illuminance or very low, 1 lx. 
I thought that maybe is the one that I had to re-pair over the months but I have re-paired one that reports correctly and it still does after re-pairing... 
Maybe the ddf is shared between the two variants of this sensors, the one that has illuminance sensor and the one who does not and at re-pairing some sensors do not detect that they have the photodiode..

There is nothing wrong with the sensor that I have tested as it reported in the aqara app..

andys73, 2023-01-28:
I am trying to find a pattern and all I can say that on all sensors the reporting is erratic with some that reported some light now is 1lx despite the light in the room.
[image]

BabaIsYou, 2023-01-28:
Just to be sure. Do they all already report correctly in your environnement by the past ?

Smanar, 2023-01-28:
Still no issue on side for lightlevel, even with 2.20 version and the "lumi.sensor_motion.aq2"

andys73, 2023-01-28:
@BabaIsYou I have automations based on these motion sensors based also on contion on lux levels for 2 years now and used to be spot on. For a while I have noticed that the automations are not as reliable as before, i.e. light on the hallway will turn on even if is quite a lot of light or light will not turn on at night ... I thought is something wrong in HA but in fact the automation is not firing as the condition on above lux level is not met.

@MrSolo570  reported the same thing in this thread https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6692#issuecomment-1406339763

I am not sure what to provide or do to help.

andys73, 2023-01-28:
[https://pastebin.com/reHm4MBg](url)
the motion sensor that has a problem is Motion Sensor 10 Kitchen 2 maybe someone can read better than me on what is going on. I set the debug to info / info 2 and error and expose the sensor to a higher light level .. It still shows as 0 everywhere.

andys73, 2023-01-28:
ok I have done extensive testing and I can confirm that is all in my mind...
All works as expected, just the place where the sensor I was testing with it reports 0 lux and the same is in the aqara light.
I just need to tune a bit the automation light levels as in the winter is darker in the house.. 
thank you all for your help and support

closed 2023-01-28 by andys73

andys73, 2023-01-28:
for other interested I used the following condition on the sensor that is relevant i.e. not in a dark corner... for Home Assistant.
`condition:
  - condition: state
    entity_id: sensor.lightlevel_87
    attribute: dark
    state: "true"`

MrSolo570, 2023-01-29:
Can you elaborate on that?
I am using the same aqara motion sensor and it should turn on the lights only if the condition is met that the light level is below the value of 3. Are you saying I should just use the condition "dark"?

andys73, 2023-01-29:
yes, experimenting with the  conditions when lights should turn on I found it best to cover all cases below and I used dark attribute rather than playing with lux values.. and added the sun condition as a fail-over in case that attribute remains true at night for whatever reason.. Seems to work fine by now, I am sure I will fine tune it in the future :)
`  - condition: or
    conditions:
      - condition: sun
        before: sunrise
        after: sunset
      - condition: state
        entity_id: sensor.lightlevel_39
        attribute: dark
        state: true`

ebaauw, 2023-01-29:
Best practice is to base automations on `state/dark` and `state/daylight`, and calibrate the light level sensors though `config/tholddark` and `config/tholdoffset`.  Note that the sensor only reports the light level, dark and daylight are derived by the REST API plugin.

Also note that either `dark` or `daylight` can be `true` are any given time, but both can be `false` at the same time.  This leads to a kind of twilight state, where you don’t want to change the lights, or they keep flipping on and off.  So, you want to turn lights on when `dark` becomes `true` and turn them off when `daylight` becomes `true`.

For my Hue motion sensors (all models), I use `tholdark` of 12000.  Depending on the light level the sensor reports with only artificial light, `tholdoffset` is between 4000 and 12000.  I find it’s not always possible to position a combined sensor, so it doesn’t look directly into a light, as it also needs to see motion.

andys73, 2023-01-29:
excellent information. Thanks!
Not sure if aqara supports config/tholddark but for hallway & kitchen the previous code that I posted seems to hold ok for now.

MrSolo570, 2023-01-29:
Thank you for this information. I will try that approach too and report back.

Can anyone explain what is meant by "through config/tholddark and config/tholdoffset"? I am running home assistant and deconz via plugin... Thank you all