dresden-elektronik/deconz-rest-plugin #6704: Hama Smart Radiator Thermostat (TS0601 _TZE200_yw7cahqs) no values

MrMarkus42, 2023-01-28:
## Describe the bug

If've got 7 working Hama Radiator Thermostats.
Now I've connected three additional ones. They have the same firmeware version as the existing ones.
However all the recently added have the same issues:
- they are paired correctly. They are visible within Rest-API oder deCONZ graphical UI
- I can't control them: setting temp, on/off, ...
- they don't report any values (temp, etc.) within the Rest-API
## Steps to reproduce the behavior

If upgraded deCONZ to 2.20.1
If delete the devices several times and start a new pairing.

All without success.
## Expected behavior

## Screenshots

## Environment

 - Host system: Raspi
 - Running method: Raspbian
 - Firmware version: 26780700
 - deCONZ version: 2.20.1
 - Device: ConBee II 
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? no

## deCONZ Logs

 Log from Rest AP:

**Device with issue:**
#                    {
    "config": {
        "heatsetpoint": null,
        "locked": true,
        "mode": "auto",
        "offset": 0,
        "on": true,
        "preset": null,
        "reachable": true,
        "schedule": {
            "W124": [
                {
                    "heatsetpoint": 0,
                    "localtime": "T00:00"
                },
                {
                    "heatsetpoint": 0,
                    "localtime": "T00:00"
                }
            ],
            "W3": [
                {
                    "heatsetpoint": 0,
                    "localtime": "T00:00"
                },
                {
                    "heatsetpoint": 0,
                    "localtime": "T00:00"
                },
                {
                    "heatsetpoint": 0,
                    "localtime": "T00:00"
                },
                {
                    "heatsetpoint": 0,
                    "localtime": "T00:00"
                },
                {
                    "heatsetpoint": 0,
                    "localtime": "T00:00"
                },
                {
                    "heatsetpoint": 0,
                    "localtime": "T00:00"
                },
                {
                    "heatsetpoint": 0,
                    "localtime": "T00:00"
                },
                {
                    "heatsetpoint": 0,
                    "localtime": "T00:00"
                },
                {
                    "heatsetpoint": 0,
                    "localtime": "T00:00"
                },
                {
                    "heatsetpoint": 0,
                    "localtime": "T00:00"
                }
            ]
        },
        "schedule_on": false
    },
    "ep": 1,
    "etag": "a344728f016902f3b361aa0b7cb51cfe",
    "lastannounced": "2023-01-28T06:49:44Z",
    "lastseen": "2023-01-28T06:49Z",
    "manufacturername": "_TZE200_yw7cahqs",
    "modelid": "TS0601",
    "name": "Thermostat 155",
    "state": {
        "lastupdated": "2023-01-28T06:49:47.482",
        "lowbattery": null,
        **"on": null,
        "temperature": 0,
        "valve": 0**
    },
    "type": "ZHAThermostat",
    "uniqueid": "cc:86:ec:ff:fe:c2:00:b9-01-0201"
}
                    
Working device:
                
                    {
    "config": {
        "heatsetpoint": 2000,
        "locked": true,
        "mode": "heat",
        "offset": 0,
        "on": true,
        "preset": null,
        "reachable": true,
        "schedule": {
            "W124": [
                {
                    "heatsetpoint": 19,
                    "localtime": "T19:19"
                },
                {
                    "heatsetpoint": 0,
                    "localtime": "T19:19"
                }
            ],
            "W3": [
                {
                    "heatsetpoint": 19,
                    "localtime": "T19:19"
                },
                {
                    "heatsetpoint": 19,
                    "localtime": "T19:19"
                },
                {
                    "heatsetpoint": 19,
                    "localtime": "T19:19"
                },
                {
                    "heatsetpoint": 19,
                    "localtime": "T19:19"
                },
                {
                    "heatsetpoint": 19,
                    "localtime": "T19:19"
                },
                {
                    "heatsetpoint": 19,
                    "localtime": "T19:19"
                },
                {
                    "heatsetpoint": 19,
                    "localtime": "T19:19"
                },
                {
                    "heatsetpoint": 19,
                    "localtime": "T19:19"
                },
                {
                    "heatsetpoint": 19,
                    "localtime": "T19:19"
                },
                {
                    "heatsetpoint": 0,
                    "localtime": "T00:00"
                }
            ]
        },
        "schedule_on": false
    },
    "ep": 1,
    "etag": "0050537242aeac2d0f12e8ef4d0805a6",
    "lastannounced": "2023-01-27T23:02:59Z",
    "lastseen": "2023-01-28T07:02Z",
    "manufacturername": "_TZE200_yw7cahqs",
    "modelid": "TS0601",
    "name": "Heizung EG Bad",
    "state": {
        "lastupdated": "2023-01-16T00:25:13.656",
        "lowbattery": false,
        "on": false,
        "temperature": 1990,
        "valve": 0
    },
    "type": "ZHAThermostat",
    "uniqueid": "cc:86:ec:ff:fe:c2:01:04-01-0201"
}
## Additional context

Smanar, 2023-01-28:
Hello, as you are on Raspbian, you can make a try with an older version ?
Can you share some logs with "info"+"info_l2"+"error"+"error_l2" ?

MrMarkus42, 2023-01-29:
Hello Smanar,
I've tried several 2.19.x versions without success.

Here are some logs concerning this device:

08:48:23:426 Tuya debug Request : Address 0xCC86ECFFFEC200B9, Endpoint 0x01, Command 0x01, Payload 000367020004000000a0
08:48:23:428 Tuya debug 4 : Address 0xCC86ECFFFEC200B9 Payload 000367020004000000a0
08:48:23:429 Tuya debug 5 : Status: 0 Transid: 3 Dp: 615 (0x02,0x67) Fn: 0 Data 160

08:48:27:376 Tuya debug Request : Address 0xCC86ECFFFEC200B9, Endpoint 0x01, Command 0x01, Payload 000367020004000000a0
08:48:27:377 Tuya debug 4 : Address 0xCC86ECFFFEC200B9 Payload 000367020004000000a0
08:48:27:378 Tuya debug 5 : Status: 0 Transid: 3 Dp: 615 (0x02,0x67) Fn: 0 Data 160

Smanar, 2023-01-29:
>08:48:23:429 Tuya debug 5 : Status: 0 Transid: 3 Dp: 615 (0x02,0x67) Fn: 0 Data 160

This is one is the heatpoint temperature, for your device so 15 degree, why the value is not updated.
There is a check with the product id

```
                        //can be setpoint for some device
                        if (productId == "Tuya_THD SEA801-ZIGBEE TRV" ||
                            productId == "Tuya_THD Smart radiator TRV" ||
```

But this one is hidden, so idk how to see it in logs, I know it appear sometime, for exemple in HA

> 2023-01-12 13:08:43.893 DEBUG (MainThread) [pydeconz.websocket] {"attr":{"id":"55","lastannounced":"2023-01-10T21:00:56Z","lastseen":"2023-01-12T12:08Z","manufacturername":"_TZE200_yw7cahqs","modelid":"TS0601","name":"Thermostat 55","productid":"Tuya_THD Smart radiator TRV","swversion":null,"type":"ZHAThermostat","uniqueid":"54:0f:57:ff:fe:c9:23:fc-01-

Perhaps on log with enabling "info" and "info_l2", you will see websocket event.

It's possible you haven't the good value for 
`"productid":"Tuya_THD Smart radiator TRV"`

scr1p7, 2023-02-03:
Hello,
looks like their currently not fully supported (if you got a "V3": https://forum.phoscon.de/t/cannot-connect-hama-thermostat-ts0601/3100). There is already an open request: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6635

Smanar, 2023-02-03:
The V3 is a "_TZE200_h4cgnbzg", so need to wait for the tuya TRV PR was merged. Not possible adding new TRV without it.

MrMarkus42, 2023-02-04:
I've got: _TZE200_yw7cahqs (see logs).

I appologize: I'm to new to DeconZ... I really don't know what to do next...

Smanar, 2023-02-04:
@MrMarkus42 yours, is on the c++ code, so it need to work.

```
08:48:23:428 Tuya debug 4 : Address 0xCC86ECFFFEC200B9 Payload 000367020004000000a0
08:48:23:429 Tuya debug 5 : Status: 0 Transid: 3 Dp: 615 (0x02,0x67) Fn: 0 Data 160
```

After thoses lines, the device need to display the heatpoint temperature in the API, 16 degree on this sample.

You don't have the debug lines after this one ?
The only thing I see that can provoke this issue is a bad productId, but this one is not visible on the API, only on logs.

MrMarkus42, 2023-02-08:
Here are some more logs. I've checked all debug options within DeconZ GUI Debug View.

Also I've done a reset of the Device and started a new pairing.

10:59:07:046 Node 0xCC86ECFFFEC200B9 is known by 2 neighbors, last seen 359592 s

10:59:09:061 Tuya debug Request : Address 0xCC86ECFFFEC200B9, Endpoint 0x01, Command 0x01, Payload 00016501000100
10:59:09:062 Tuya debug 4 : Address 0xCC86ECFFFEC200B9 Payload 00016501000100
10:59:09:062 Tuya debug 5 : Status: 0 Transid: 1 Dp: 357 (0x01,0x65) Fn: 0 Data 0
10:59:09:064 Websocket 10.:38703 send message: {"config":{"heatsetpoint":null,"locked":true,"mode":"off","offset":0,"on":true,"preset":null,"reachable":true,"schedule":{"W124":[{"heatsetpoint":0,"localtime":"T00:00"},{"heatsetpoint":0,"localtime":"T00:00"}],"W3":[{"heatsetpoint":0,"localtime":"T00:00"},{"heatsetpoint":0,"localtime":"T00:00"},{"heatsetpoint":0,"localtime":"T00:00"},{"heatsetpoint":0,"localtime":"T00:00"},{"heatsetpoint":0,"localtime":"T00:00"},{"heatsetpoint":0,"localtime":"T00:00"},{"heatsetpoint":0,"localtime":"T00:00"},{"heatsetpoint":0,"localtime":"T00:00"},{"heatsetpoint":0,"localtime":"T00:00"},{"heatsetpoint":0,"localtime":"T00:00"}]},"schedule_on":false},"e":"changed","id":"155","r":"sensors","t":"event","uniqueid":"cc:86:ec:ff:fe:c2:00:b9-01-0201"} (ret = -1094184784)
10:59:09:065 Websocket 10.102.250.31:54291 send message: {"config":{"heatsetpoint":null,"locked":true,"mode":"off","offset":0,"on":true,"preset":null,"reachable":true,"schedule":{"W124":[{"heatsetpoint":0,"localtime":"T00:00"},{"heatsetpoint":0,"localtime":"T00:00"}],"W3":[{"heatsetpoint":0,"localtime":"T00:00"},{"heatsetpoint":0,"localtime":"T00:00"},{"heatsetpoint":0,"localtime":"T00:00"},{"heatsetpoint":0,"localtime":"T00:00"},{"heatsetpoint":0,"localtime":"T00:00"},{"heatsetpoint":0,"localtime":"T00:00"},{"heatsetpoint":0,"localtime":"T00:00"},{"heatsetpoint":0,"localtime":"T00:00"},{"heatsetpoint":0,"localtime":"T00:00"},{"heatsetpoint":0,"localtime":"T00:00"}]},"schedule_on":false},"e":"changed","id":"155","r":"sensors","t":"event","uniqueid":"cc:86:ec:ff:fe:c2:00:b9-01-0201"} (ret = -1094184784)
10:59:09:068 Websocket 10.:38703 send message: {"e":"changed","id":"155","r":"sensors","state":{"lastupdated":"2023-02-08T09:59:09.063","lowbattery":null,"on":null,"temperature":0,"valve":0},"t":"event","uniqueid":"cc:86:ec:ff:fe:c2:00:b9-01-0201"} (ret = -1094184784)
10:59:09:070 Websocket 10:54291 send message: {"e":"changed","id":"155","r":"sensors","state":{"lastupdated":"2023-02-08T09:59:09.063","lowbattery":null,"on":null,"temperature":0,"valve":0},"t":"event","uniqueid":"cc:86:ec:ff:fe:c2:00:b9-01-0201"} (ret = -1094184784)
10:59:09:071 APS-DATA.request id: 170, addrmode: 0x02, addr: 0x505c, profile: 0x0104, cluster: 0xEF00, ep: 0x01 -> 0x01 queue: 1 len: 5 tx.options 0x04
10:59:09:072 	asdu (length: 5): 10000b0100
10:59:09:073 MS CORE_APS_IND (processPacked:1339) took 21 ms
10:59:09:105 HTTP API GET /api/CF1097716D/sensors/155 - 10.
10:59:09:106 ApiMode: 0
10:59:09:108 {"config":{"heatsetpoint":null,"locked":true,"mode":"off","offset":0,"on":true,"preset":null,"reachable":true,"schedule":{"W124":[{"heatsetpoint":0,"localtime":"T00:00"},{"heatsetpoint":0,"localtime":"T00:00"}],"W3":[{"heatsetpoint":0,"localtime":"T00:00"},{"heatsetpoint":0,"localtime":"T00:00"},{"heatsetpoint":0,"localtime":"T00:00"},{"heatsetpoint":0,"localtime":"T00:00"},{"heatsetpoint":0,"localtime":"T00:00"},{"heatsetpoint":0,"localtime":"T00:00"},{"heatsetpoint":0,"localtime":"T00:00"},{"heatsetpoint":0,"localtime":"T00:00"},{"heatsetpoint":0,"localtime":"T00:00"},{"heatsetpoint":0,"localtime":"T00:00"}]},"schedule_on":false},"ep":1,"etag":"c2ec9079421a614ead7138bcb3309471","lastannounced":"2023-02-08T09:58:51Z","lastseen":"2023-02-08T09:58Z","manufacturername":"_TZE200_yw7cahqs","modelid":"TS0601","name":"Thermostat 155","state":{"lastupdated":"2023-02-08T09:59:09.063","lowbattery":null,"on":null,"temperature":0,"valve":0},"type":"ZHAThermostat","uniqueid":"cc:86:ec:ff:fe:c2:00:b9-01-0201"}
10:59

10:59:10:249 	asdu: 815c50b900c2feffec86cc80
10:59:10:250 ZDP device announce: 0xCC86ECFFFEC200B9, 0x505C, 0x80
10:59:10:250 ZDP add 0xCC86ECFFFEC200B9, nwk: 0x505C to discover queue
10:59:10:253 R stats, str: 18, num: 0, item: 22
10:59:10:257 DeviceAnnce of SensorNode: 0xCC86ECFFFEC200B9 [1]
10:59:10:259 DEV Tick.Join: event/device.anounce
10:59:10:259 DEV Tick: fast poll 0xCC86ECFFFEC200B9, mac capabilities: 0x80
10:59:10:261 Websocket 10.:38703 send message: {"attr":{"id":"155","lastannounced":"2023-02-08T09:59:10Z","lastseen":"2023-02-08T09:58Z","manufacturername":"_TZE200_yw7cahqs","modelid":"TS0601","name":"Thermostat 155","productid":null,"swversion":null,"type":"ZHAThermostat","uniqueid":"cc:86:ec:ff:fe:c2:00:b9-01-0201"},"e":"changed","id":"155","r":"sensors","t":"event","uniqueid":"cc:86:ec:ff:fe:c2:00:b9-01-0201"} (ret = -1094184784)
10:59:10:262 Websocket 10.:54291 send message: {"attr":{"id":"155","lastannounced":"2023-02-08T09:59:10Z","lastseen":"2023-02-08T09:58Z","manufacturername":"_TZE200_yw7cahqs","modelid":"TS0601","name":"Thermostat 155","productid":null,"swversion":null,"type":"ZHAThermostat","uniqueid":"cc:86:ec:ff:fe:c2:00:b9-01-0201"},"e":"changed","id":"155","r":"sensors","t":"event","uniqueid":"cc:86:ec:ff:fe:c2:00:b9-01-0201"} (ret = -1094184784)
[... cut ...]
10:59:10:896 ZCL read attr 0xCC86ECFFFEC200B9, ep: 0x01, cl: 0x0201, attr: 0x0012, mfcode: 0x0000, aps.id: 190, zcl.seq: 106
10:59:10:899 APS-DATA.request id: 190, addrmode: 0x03, addr: 0xcc86ecfffec200b9, profile: 0x0104, cluster: 0x0201, ep: 0x01 -> 0x01 queue: 1 len: 5 tx.options 0x00
10:59:10:902 	asdu (length: 5): 106a001200
10:59:10:934 skip device discovery while end devices is added
10:59:10:961 APS-DATA.confirm id: 187, status: 0x00 SUCCESS
10:59:10:962 APS-DATA.confirm request id: 187 -> erase from queue

Smanar, 2023-02-09:
`"productid":null`

The problem can come from that. But IDK what happen on this device, have you set a special shedule on it ?

You delete it before the new inclusion ?
Because you always have a full "schedule" stuff just after detection even the device haven't send notification about it.

And I don't see how DDF can help, as it's not possible to lock a productid using it.

MrMarkus42, 2023-02-19:
I've got a completely new device. Using this device the same error occours. It is recognized, but there are only "0" as values:

{
    "15:49:38:424": {
        "config": {
            "heatsetpoint": null,
            "locked": false,
            "mode": "off",
            "offset": 0,
            "on": true,
            "preset": null,
            "reachable": true,
            "schedule": {},
            "schedule_on": false
        },
        "e": "changed",
        "id": "156",
        "r": "sensors",
        "t": "event",
        "uniqueid": "8c:f6:81:ff:fe:c8:03:b7-01-0201"
    },
    "15:49:38:478": {
        "e": "changed",
        "id": "156",
        "r": "sensors",
        "state": {
            "lastupdated": "2023-02-19T14:49:38.338",
            "lowbattery": null,
            "on": null,
            "temperature": 0,
            "valve": 0
        },
        "t": "event",
        "uniqueid": "8c:f6:81:ff:fe:c8:03:b7-01-0201"
    },
    "15:49:39:462": {
        "config": {
            "heatsetpoint": null,
            "locked": true,
            "mode": "off",
            "offset": 0,
            "on": true,
            "preset": null,
            "reachable": true,
            "schedule": {},
            "schedule_on": false
        },
        "e": "changed",
        "id": "156",
        "r": "sensors",
        "t": "event",
        "uniqueid": "8c:f6:81:ff:fe:c8:03:b7-01-0201"
    },
    "15:49:40:491": {
        "config": {
            "heatsetpoint": null,
            "locked": true,
            "mode": "auto",
            "offset": 0,
            "on": true,
            "preset": null,
            "reachable": true,
            "schedule": {},
            "schedule_on": false
        },
        "e": "changed",
        "id": "156",
        "r": "sensors",
        "t": "event",
        "uniqueid": "8c:f6:81:ff:fe:c8:03:b7-01-0201"
[... 539 lines cut ...]
        "id": "156",
        "r": "sensors",
        "t": "event",
        "uniqueid": "8c:f6:81:ff:fe:c8:03:b7-01-0201"
    },
    "15:52:59:484": {
        "attr": {
            "id": "156",
            "lastannounced": null,
            "lastseen": "2023-02-19T14:52Z",
            "manufacturername": "_TZE200_yw7cahqs",
            "modelid": "TS0601",
            "name": "Thermostat 156",
            "productid": null,
            "swversion": null,
            "type": "ZHAThermostat",
            "uniqueid": "8c:f6:81:ff:fe:c8:03:b7-01-0201"
        },
        "e": "changed",
        "id": "156",
        "r": "sensors",
        "t": "event",
        "uniqueid": "8c:f6:81:ff:fe:c8:03:b7-01-0201"
    }
}

Smanar, 2023-02-19:
So there is probably a new version for this device if the old one are still working, but I realy have no clue where you issue is from.

I think you can't compile c++  code to make test ? So the better thing is waiting for the tuya PR to restart from scratch and make a DDF for the device instead of using the legacy code. 

https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6674

Smanar, 2023-02-20:
@MrMarkus42 The PR was merged.
Have started to make a DDF for your device

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZE200_yw7cahqs",
  "modelid": "TS0601",
  "vendor": "Tuya",
  "product": "Tuya TRV saswell",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_THERMOSTAT",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0xef00"
      ],
      "meta": {
        "values": {
          "config/mode": {"heat": 1, "off": 0}
        }
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/heatsetpoint",
          "parse": {"fn": "tuya", "dpid": 103, "eval": "Item.val = Attr.val * 10;"},
          "write": {"fn": "tuya", "dpid": 103, "dt": "0x2b", "eval": "Item.val / 10;"},
          "read": {"fn": "tuya"}
        },
        {
          "name": "config/locked",
          "parse": {"fn": "tuya", "dpid": 40, "eval": "Item.val = Attr.val;"},
          "write": {"fn": "tuya", "dpid": 40, "dt": "0x10", "eval": "Item.val;"},
          "read": {"fn": "none"}
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "config/mode",
          "values": [
              ["heat", 1], ["off", 0]
          ],
          "parse": {"fn": "tuya", "dpid": 2, "eval": "if (Attr.val == 0) { Item.val = 'off' } else { Item.val = 'heat' }"},
          "write": {"fn": "tuya", "dpid": 2, "dt": "0x10", "eval": "if (Item.val == 'heat') { 1 } else { 0 }"},
          "read": {"fn": "none"}
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/temperature",
          "parse": {"fn": "tuya", "dpid": 102, "eval": "Item.val = Attr.val * 10;"},
          "read": {"fn": "none"}
        }
      ]
    }
  ]
}
```

There is not yet all feature, but the heatpoint and the local temperature need to work.

closed 2023-03-21 by github-actions[bot]