dresden-elektronik/deconz-rest-plugin #6709: turn off with transition (fade out) stopped working in home assistant deconz addon

prof7bit, 2023-01-28:
## Describe the bug

turn off with transition (fade out) stopped working in home assistant. I don't know what happens when i use the rest api directly because I have downgraded the deconz addon again to avoid the bug. I have not found any way in home assistant to restore previous behavior (fade off a zigbee light) without downgrading the deconz addon.

## Steps to reproduce the behavior
in Home Assistant call the ```turn_off``` service with 
```
"data": {
  "transition": 10
}
```

## Expected behavior
Light will fade to black during 10 seconds, after this the light is off and the stored brightness in the bulb is 1 (if you turn it on the next time)

## Actual behavior
Light will switch off immediately and the stored brightness remains at 255

## Environment
 - Host system: proxmox + Home Assistant (2022.11.5) + Deconz-Addon 2.20.0 or 2.20.1 (not sure, have downgraded again)

I could not find anything suspicious in the change log, but the turn-off with transition behavior has definitely changed, With the afforementioned version I am completely unable to do a fade-off transition.

Fade from one brightness to another works, but fade-off seems impossible, at least from within home assistant.

prof7bit, 2023-01-28:
Since I did not find anything in the change logs, maybe one of the devs can remember whether there have been some changes in the way the transitions are handled, maybe this gives a clue as to whether the home assistent integration needs a change (or has already been silently changed while I did not update HA for other reasons).

Mimiix, 2023-01-28:
I would like to ask you to test the rest api first.

The Home assistant integration is not the scope of this repository.

prof7bit, 2023-01-28:
But has there been a breaking change? I did not update anything HA-related for almost two months, but the bug appeared 4 days ago when I upgraded deconz, and it was fixed again after I downgraded deconz. So **some** kind of change must have happened in the handling of transitions.

Mimiix, 2023-01-28:
Not that I know of. Again, when a bug report is placed we are asking to test issues as far as possible. Your posting a bug with home assistant and haven't tested the rest api. Now we can't determine where the issue might be.

dof250, 2023-02-03:
I'm also having the same issue. Transition to off doesn't work anymore. Issue is there when using Node-Red deCONZ plugin or Home Assistant

esbenlydiksen, 2023-02-05:
I see the same issue on my system.

ebaauw, 2023-02-05:
What lights does this concern?  Are these exposed through DDF or through legacy code?  Was the DDF updated in v2.20, to include `cap/on/off_with_effect`?

Note that you cannot specicy a _Transition Time_ to the Zigbee _Off_ command.  Instead, you need to use _Off with Effect_.  However not all devices support this command, notably some plugs, but also some exotic lights.

By default the legacy code sends _Off with Effect_ when `"on": false` is PUT.  When `"transitiontime": 0` is included, it sends _Off_.  It also sends _Off_ for some whitelisted devices that don't support _Off with Effect_.

As of v2.20, the DDF code sends _Off_, unless `cap/on/off_with_effect` is included in the DDF.

1doubleDD1, 2023-02-05:
> What lights does this concern? Are these exposed through DDF or through legacy code? Was the DDF updated in v2.20, to include `cap/on/off_with_effect`?

'Tis a good question, not sure how to check and reply to that. Most of my bulbs are old Osram E27 Dimmable or Tunable white. In the release notes for 2.20.x however I haven't seen anything related to Osram.

dof250, 2023-02-05:
I'm using only Philips (Signify) Hue lights. I maybe have an example that helps. I have a group off lights above the dining table named **Eettafel**. It's contains 6 lights (Eettafel 1 to 6). While using deCONZ in node-red and sending a transition 0 or 60 (6seconds) using the group it works and fades out or turns off instantly. When sending the same command to Eettafel 1 it always uses a 1 seconds transition to off and ignores my input.

dimitripb, 2023-02-12:
I second this issue. It can be reproduced with the Phoscon app. Create a scene with one bulb and brightness 0. Enter a transition of 10 s and turn on the bulb. Then activate the created scene. The bulb will enter off state within 1 second instead of 10.
So it's not a problem in Home Assistant or Node Red.

dimitripb, 2023-03-08:
Is there another issue addressing this bug? This one will be closed in several days but I can not find a related issue. Although this issue was created with a HA and Node Red context it IS reproducible within Phoscon (as described in my earlier post).

Mimiix, 2023-03-08:
> As there has not been any response in 21 days, this issue has been automatically marked as stale. **At OP: Please either close this issue or keep it active** It will be closed in 7 days if no further activity occurs.

As you replied now, Stale gets removed tonight automatically.

lindhe, 2023-03-31:
Please keep open.

prof7bit, 2023-04-23:
Still not fixed.

dimitripb, 2023-04-24:
> As of v2.20, the DDF code sends _Off_, unless `cap/on/off_with_effect` is included in the DDF.

Does this mean there is no bug and we need to change de DDF for the bulbs which encounter this problem?

ebaauw, 2023-04-24:
Tbh, I don't have enough info to assess whether there's a bug in deCONZ or not.  Also the "me too" remarks don't help, with similar symptoms in a completely different setup (e.g. recalling a ZIgbee scene, or updating the `action` of a `groups` resource).

As I said before, Zigbee doesn't take a transition time with an _Off_ command, so you would have to use a _Off with Effect_ command, or a _Move to Level (with On/Off)_.  To make matters worse, not all ligts implement these commands correctly, which resulted in lot of ugly whitelisting in the legacy code.

To fully analyse this issue, I would need to know:
1. What lights does this concern?  Ideally screen dump of the _Basic_ cluster and listing of the `lights` resource;
2. Check in the GUI how the light reacts to:
  - _Off_;
  - _Off with Effect_;
  - _Move to Level (with On/Off)_ with a level of 0 and a transition time of, say 20 (2s).
3. Check whether the light is exposed by legacy code or through DDF, in which case, attach the DDF.
4. The precise body to the PUT of the `state` of the `lights` resource.
5 The Zigbee commands sent by deCONZ as a result of the PUT (either from the deCONZ log or from a Zigbee sniffer).

If the behaviour changes between versions, we need to understand whether that's because 4) has changed (in which case the API client does something else), or because 5) has changed.

> Does this mean there is no bug and we need to change de DDF for the bulbs which encounter this problem?

Possibly, when:
- 2) shows that the lights behaves as intended on _Off with Effect_, but not with _Off_;
- 3) shows that the device is exposed by legacy or by a DDF without `off_with_effect`;
- 4) shows that the PUT body sent by HA hasn't changed;
- 5) shows that the Zigbee commands sent by deCONZ did change (from _Off with Effect_ to _Off_).

dimitripb, 2023-04-25:
Thanks! That helps a lot. I will checkout your suggestions. I am not fully into Zigbee so I hope you can help me when I am stuck.

dimitripb, 2023-05-06:
Answer on question 1: I only have some IKEA light and they are all affected. Below the screen dumps and resources:
[image]
```
  "14": {
    "capabilities": {
      "alerts": ["none", "select", "lselect"],
      "color": {
        "ct": {
          "max": 454,
          "min": 250
        },
        "modes": ["ct"]
      }
    },
    "colorcapabilities": 16,
    "config": {
      "groups": ["0"]
    },
    "ctmax": 454,
    "ctmin": 250,
    "etag": "c1b85accd33e089c1a2c8783545be534",
    "hascolor": true,
    "lastannounced": "2023-05-06T18:15:12Z",
    "lastseen": "2023-05-06T18:45Z",
    "manufacturername": "IKEA of Sweden",
    "modelid": "TRADFRIbulbE27WSglobeopal1055lm",
    "name": "Lamp trappenhuis",
    "state": {
      "alert": "none",
      "bri": 209,
      "colormode": "ct",
      "ct": 344,
      "on": false,
      "reachable": true
    },
    "swversion": "1.0.012",
    "type": "Color temperature light",
    "uniqueid": "50:32:5f:ff:fe:dc:7c:91-01"
  }
```
[image]
 ```
 "13": {
    "capabilities": {
      "alerts": ["none", "select", "lselect"],
      "color": {
        "ct": {
          "max": 65279,
          "min": 0
        },
        "modes": []
      }
    },
    "colorcapabilities": 0,
    "config": {
      "groups": ["0", "5"]
    },
    "ctmax": 65279,
    "ctmin": 0,
    "etag": "44b7a656c1552204650814abaa890555",
    "hascolor": true,
    "lastannounced": "2022-10-22T15:27:50Z",
    "lastseen": "2023-05-06T18:44Z",
    "manufacturername": "IKEA of Sweden",
    "modelid": "TRADFRI bulb GU10 WS 400lm",
    "name": "Spot 1 huiskamer",
    "state": {
      "alert": "none",
      "bri": 254,
      "colormode": "ct",
      "ct": 250,
      "on": false,
      "reachable": true
    },
    "swversion": "2.3.087",
    "type": "Color temperature light",
    "uniqueid": "00:0b:57:ff:fe:e7:ac:03-01"
  }

```
[image]

```
  "1": {
    "capabilities": {
      "alerts": ["none", "select", "lselect"],
      "color": {
        "ct": {
          "max": 65279,
          "min": 0
        },
        "modes": []
      }
    },
    "colorcapabilities": 0,
    "config": {
      "groups": ["0", "12"]
    },
    "ctmax": 65279,
    "ctmin": 0,
    "etag": "9ce9474d00d0e1430aeeea7f9c9e6185",
    "hascolor": true,
    "lastannounced": "2023-02-02T07:19:44Z",
    "lastseen": "2023-05-06T18:42Z",
    "manufacturername": "IKEA of Sweden",
    "modelid": "TRADFRI bulb E27 WS opal 1000lm",
    "name": "Lamp eettafel",
    "state": {
      "alert": "none",
      "bri": 254,
      "colormode": "ct",
      "ct": 250,
      "on": false,
      "reachable": true
    },
    "swversion": "2.3.095",
    "type": "Color temperature light",
    "uniqueid": "68:0a:e2:ff:fe:90:21:8b-01"
  }

```
[image]
```
  "3": {
    "capabilities": {
      "alerts": ["none", "select", "lselect"]
    },
    "config": {
      "groups": ["0"]
    },
    "etag": "06a3ab54773788115e6145548c587746",
    "hascolor": false,
    "lastannounced": "2023-04-16T20:23:22Z",
    "lastseen": "2023-05-06T18:39Z",
    "manufacturername": "IKEA of Sweden",
    "modelid": "TRADFRI bulb E27 WW 806lm",
    "name": "Lamp halletje",
    "state": {
      "alert": "none",
      "bri": 254,
      "on": false,
      "reachable": true
    },
    "swversion": "2.3.093",
    "type": "Dimmable light",
    "uniqueid": "58:8e:81:ff:fe:11:8a:70-01"
  }
```

dimitripb, 2023-05-06:
Question 2:
TRADFRIbulbE27WSglobeopal1055lm

**Off:** 
result: fading to off in about 1 second

**Off with effect:**
Effect identifier: Dying light
Effect variant: 20% dim up in 0.5s then off in 1s
result: working as expected

Effect identifier: Delayed all off
Effect variant: Fade off in 0.8s
result: working as expected

Effect identifier: Delayed all off
Effect variant: No fade
result: working as expected

Effect identifier: Delayed all off
Effect variant: 50% dim down in 0.8s and then fade off in 12s
result: working as expected

**Move to level:**
result: working as expected. The lights will fading out to off during the given time.

Question 3:
TRADFRIbulbE27WSglobeopal1055lm seems to use DDF.

@ebaauw : how do I know device is actually using legacy code or DDF.

1doubleDD1, 2023-05-07:
> 
> To fully analyse this issue, I would need to know:
> 
> 4. The precise body to the PUT of the `state` of the `lights` resource.
> 5. The Zigbee commands sent by deCONZ as a result of the PUT (either from the deCONZ log or from a Zigbee sniffer).
> 
> If the behaviour changes between versions, we need to understand whether that's because 4) has changed (in which case the API client does something else), or because 5) has changed.
> 

I'd like to help as well, as the issue also affects me and would like to see it properly resolved.
v2.19.03 was the last deCONZ version where my OSRAM lights faded to OFF as expected, afterwards obviously something changed.
Now, 1, 2 & 3 I was able to figure out how to find, but I'll need a bit of help on where to look for 4 & 5, so I can paste them in as well.

Thanks!

ebaauw, 2023-05-07:
> how do I know device is actually using legacy code or DDF.

In the GUI, click on the node, next right-click on the node and select _Edit DDF_ from the popup menu.  It should popup the DDF editor.  Check that the status is _Gold_.  And double-check that _DDF Mode_ in the _DDF_ tab of the _Control_ panel enables at least _Gold_ DDFs.

If there is a DDF, check that `on/off_with_effect` and `bri/move_with_onoff` have been listed under `Capabilities`.

> where to look for 4 & 5

4: in the log of the API client, or in the deCONZ log, with `HTTP` logging enabled;
5: As I mentioned before: in a Zigbee sniffer, or in the deCONZ log with `APS` and `APS_L2` logging enabled.

dimitripb, 2023-05-07:
@ebaauw Thanks! Fixed!

Adding `on/off_with_effect` and `bri/move_with_onoff` to the `capabilities` did the job. Three of four bulbs have DDF, so that's easy to fix. I will create pull requests for these.

The one without DDF needs some more research. Or can I just create a bronze or silver DDF with these capabilities?

dimitripb, 2023-05-10:
Adding "on/off_with_effect" and "bri/move_with_onoff" to the DDF fixed all my problems with the Ikea Tradfri bulbs.

@ebaauw Thanks for your help with this!
What is the difference between Capabilities items and Config items (in the DDF editor)? Everywhere in the documentation only Config items are explained or shown. Cap items do not seem to exist in the documentation.
When I check DDF files on github lots of them have Config items. In my Deconz setup they have only Cap items.

ebaauw, 2023-05-10:
See https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6316.

dimitripb, 2023-05-11:
I created these pull requests to update the DDFs for my bulbs.

#6966 
#6965 
#6964 
#6963 
#6962

1doubleDD1, 2023-05-12:
Well, I also tried creating DDF for my Osram lights and adding "on/off_with_effect" and "bri/move_with_onoff", but it still doesn't work. :(
Of course, it might be due to not really knowing what I'm doing, but I'm still quite sure that DDF is correct and should work. Can it be done on the fly just via "hot reloading" the changed DDF or do I need to also restart/reboot something else?
Thanks!

Mimiix, 2023-05-12:
@1doubleDD1  Best to ask on the forums in that case ;)

1doubleDD1, 2023-05-12:
@Mimiix When I know what to ask and how to formulate the question, then I generally do ask on Discord. This topic is however quite above me, and I don't want to be that guy with the question "How do I create DDF" because I'm -honestly- not really interested how the sausage is made, I just want to see it working (I am however happy to provide helpful data to someone with better understanding of this whole thing).
Secondly, I'm still not really sure what the issue is, as in current and previous versions there was no DDF for my Osram bulbs, however before they were working properly and now they are not. Something somewhere changed between 2.19x and 2.20.x that impacted this change and I really do not know how to get that information.

dimitripb, 2023-05-12:
> Can it be done on the fly just via "hot reloading" the changed DDF or do I need to also restart/reboot something else?

Hot reload schould be enough. Be sure the status is not Draft anymore. Set it to gold to test it on your own environment.

Probably something changed in the legacy code. That caused a lot of bulbs not functioning properly anymore. It's better to create well functioning DDF files instead of changing the legacy code again. At least in my opinion.

1doubleDD1, 2023-05-12:
Thanks @dimitripb for your help!
I just love this sort of issues... Same DDF that I created and wasn't working a couple od hours ago, I just tried again and now it's working perfectly. SMH Apparently the same one as in life also applies here, that time heals everything...
The status was/and still is in Bronze, apparently it doesn't affect it.

Mimiix, 2023-05-12:
> Thanks @dimitripb for your help! I just love this sort of issues... Same DDF that I created and wasn't working a couple od hours ago, I just tried again and now it's working perfectly. SMH Apparently the same one as in life also applies here, that time heals everything... The status was/and still is in Bronze, apparently it doesn't affect it.
> 
> UPDATE: Apparently it's not healed by the time or our Lord and Savior himself, status is the answer. When it's set to Bronze, it doesn't work properly. When I change it to Gold, suddenly it's working as it should. If I switch it to Bronze afterwards, it continue to works as it should... So apparently change to Gold does some magic inside. And yes, only Hot reload is enough.

Yes, but when you reboot it probably stops working.

1doubleDD1, 2023-05-12:
> 
> Yes, but when you reboot it probably stops working.

You mean about the Bronze status? Exactly, that's how I figured that it's the culprit.

Mimiix, 2023-05-12:
> > Yes, but when you reboot it probably stops working.
> 
> You mean about the Bronze status? Exactly, that's how I figured that it's the culprit.

So keep it on gold. No problem with that.

dimitripb, 2023-05-12:
@1doubleDD1  of you have a tested and working DDF for your bulbs please create a pull request te integrate it into Deconz:
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/How-to-create-a-pull-request-to-submit-a-new-Device-Description-File

1doubleDD1, 2023-05-12:
> @1doubleDD1 of you have a tested and working DDF for your bulbs please create a pull request to integrate it into Deconz: https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/How-to-create-a-pull-request-to-submit-a-new-Device-Description-File

Will do, have 2 more to do and test tomorrow 👍 Thanks

dimitripb, 2023-05-13:
@ebaauw 

One more question (for my understanding): In de current DDF for `devices/ikea/gu10_ws_400lm_light.json` ( https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/ikea/gu10_ws_400lm_light.json ) is listed:

```

  {
                    "name": "config/colorcapabilities",
                    "static": 16
                },
                {
                    "name": "config/ctmin",
                    "static": 250
                },
                {
                    "name": "config/ctmax",
                    "static": 454
                },
```
I cannot find any `Config` items called `ctmin`, `ctmax` and `colorcapabilities`. Should these actually be `Cap` items and is the current DDF wrong? Or do I misunderstand something?

1doubleDD1, 2023-05-13:
My guess would be that prior to 2.20 only Config was used, and then Capabilities were introduced (if you check bottom of my big post above with generic DDFs). So I would guess that older DDFs probably should be updated...
Also under Cap there are 2 instances of some items like ctmax, one is deprecated and one active.

1doubleDD1, 2023-05-14:
Apologies if this is already spamming on the topic, but I'm contemplating the usefulness of "on/off_with_effect" in the DDFs.
Once it's included, there is no way to turn the bulb instantly off (without any effect). And quite similar result to off with effect is usage of command off with transition of 1-2sec (which is part of "bri/move_with_onoff" that is already part of DDF). Generally I'm using on/off commands from HA via "Call service" action where it's easy enough to define whether you want or not the transition (I'm aware that this option is not available if one is using "Device command"). And on some lights I prefer the off with transition, on some instant off.

To summarize, if "on/off_with_effect" is included in DDF, user has **_no possibility_** to instantly switch the bulb off. However, if it's omitted, user **_can decide_** on the desired behaviour (although maybe a bit more complicated). Or am I missing something?

So, should it be included or not? Thanks for any input.

dimitripb, 2023-05-14:
@1doubleDD1 In my case (with IKEA bulbs) it IS a capability so it should be added to the DDF. I can turn off my bulbs immediately with "on/off_with_effect" in the DDF. No problem.
Maybe with Osram bulbs it's different. Before adding capabilities to the DDF you should test whether the device has actually that capability.

1doubleDD1, 2023-05-14:
Maybe I wasn't clear enough, when I test from deconz gui there are on, off, toggle and on/off with effect commands and they all work properly on the bulb (there are 3 off effects available and they all work). However if I include that capability in the DDF, then from HomeAssistant off command behaves as off with effect (fades to off in about 1s). It's like including the capability overrides instant off command.

dimitripb, 2023-05-14:
Yes I do understand, but when a bulb has a specific capability, it should be listed in the DDF in my opinion. Omitting could lead to other problems maybe. In my case the bulbs always went off in 1 second instead of immediately when I use HA. I don't know what is best practice in this case.
When you buy an Ikea hub and Ikea bulbs turning off is never immediately so I think they are meant to turn off with a transition of 1s.

prof7bit, 2023-05-16:
> However if I include that capability in the DDF, then from HomeAssistant off command behaves as off with effect (fades to off in about 1s). It's like including the capability overrides instant off command.

This is then probably a problem with the home assistant deconz integration and should be fixed there. But first of all the off-with-effect should be properly supported and usable on all bulbs that support it, then making home assistant use the proper Deconz API for all kinds of bulbs and all kinds of intended behavior would be a home assistant bug.

closed 2023-06-15 by github-actions[bot]