dresden-elektronik/deconz-rest-plugin #6710: Changes to `execute_if_off` sometimes not honoured

ebaauw, 2023-01-28:
## Describe the bug

When setting `config/color/execute_if_off` through the API, deCONZ sometimes fails to write _Options_.  It seems the change is ignored by state engine: no messages in the log (except the wen socket notification), no Zigbee _Write Attributes_ in the sniffer.  Sometimes this happens for all lights, sometimes for some.  Sometimes it works after restarting deCONZ.

## Steps to reproduce the behavior
PUT a light `config` with body of `{"color": {"execute_if_off": true}}`

## Expected behavior
Should work always.

## Screenshots
n/a

## Environment

 - Host system: Raspberry Pi 
 - Running method: Raspbian
 - Firmware version: 26780700
 - deCONZ version: 2.20.1
 - Device: ConBee II
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? no

## deCONZ Logs

When it works, the DDF log seems normal:
```
Jan 28 21:15:16 pi5 deCONZ[20998]: 21:15:12:017 SC tick --> StateCallFunction
Jan 28 21:15:16 pi5 deCONZ[20998]: 21:15:12:018 writeZclAttribute, ep: 0x0B, cl: 0x0300, attr: 0x000F, type: 0x18, mfcode: 0x0000, expr: Item.val ? 0x01 : 0x00
Jan 28 21:15:16 pi5 deCONZ[20998]: 21:15:12:018 00:17:88:01:0b:1c:ff:75-0b/config/color/execute_if_off expression: Item.val ? 0x01 : 0x00 --> 1
...
Jan 28 21:15:30 pi5 deCONZ[20998]: 21:15:21:522 SC tick --> StateRead config/color/execute_if_off, 00:17:88:01:0b:1c:ff:75-0b
Jan 28 21:15:30 pi5 deCONZ[20998]: 21:15:21:635 00:17:88:01:0b:1c:ff:75-0b/config/color/execute_if_off expression: Item.val = (Attr.val & 0x01) !== 0 --> true
Jan 28 21:15:30 pi5 deCONZ[20998]: 21:15:21:636 SC config/color/execute_if_off: synced
Jan 28 21:15:30 pi5 deCONZ[20998]: 21:15:21:636 SC --> StateFinished
Jan 28 21:15:30 pi5 deCONZ[20998]: 21:15:21:637 SC state change finished: 00:17:88:01:0b:1c:ff:75-0b
```

## Additional context

I noticed a while ago that changes for my Mueller light didn't stick, but didn't have the time to investigate.  Now I also see reports from one of my users, see https://github.com/ebaauw/homebridge-deconz/issues/100#issuecomment-1407475554.  And more extensive testing shows it sometimes doesn't work for my Hue lights either.  Not sure if relevant, but `execute_if_off` has default `true`, so it should be written on startup.

SwoopX, 2023-02-09:
@ebaauw Pew, that makes me nervous... However, it might be related to #6323, a state change is kept alive "just" for 3 mins. You might want to retest with the added attribute and raise the timeout.

ebaauw, 2024-09-15:
@SwoopX, I'm not sure if it's related; this is on a light, an always-on device.

I ran into (presumably) the same issue yesterday, where changing `config/sensitivity` on a Hue motion sensor wouldn't stick.  Thinking it could be related to the manufacturer-specific attribute, I tried `config/duration`, but that wouldn't stick either.  Then I tried `config/bri/execute_if_off` on a light, which wouldn't stick either.  As far as I could figure out from the log, deCONZ doesn't even try to send the _Write Attributes_.

Curiously, a simple restart of deCONZ fixed all issues at the same time.  I suspect it might be a race condition during initialisation of the DDF logic, see also #7927.

SwoopX, 2024-09-15:
I devinitively recall that we had some reports from people claiming changes do not stick and sensors partially not reporting what they're doing. Eventually, this all fits together 🤔 In any case, that is a nasty one and good that we now have some traces to follow up.

ebaauw, 2024-12-02:
I think this is another symptom of https://github.com/dresden-elektronik/deconz-rest-plugin/issues/7935.  The issue seems to be triggered by pairing a new device (causing the DDFs to be reloaded), and a restart of deCONZ seems to solve it.