dresden-elektronik/deconz-rest-plugin #6713: Invalid JSON after adding Philips Hue Go

shin10, 2023-01-31:
## Describe the bug

A Philips Hue Go (v1, not sure about others) causes some issues. For me, after updating the docker container, the device got lost in the phoscon pwa and Hue essentials currently isn't able to connect to the bridge anymore. According to the findings of @ulilicht [here](https://community.hueessentials.com/t/cannot-connect-to-bridge-invalid-json-after-adding-hue-go/5067), he's able to add it to phoscon, but also loses the bridge connection (in Hue Essentials) then.

Errors in Hue Essentials look like this:
`The error message is: Invalid JSON at 11183: Expected string or non-null literal.`

He tracked the issue down to the following commit: https://github.com/dresden-elektronik/deconz-rest-plugin/commit/c27e6fde80682e2f87744379df73ce4d4fca9a4d
and I can confirm the problem in various docker containers:

deconzcommunity/deconz:2.18.01 :white_check_mark:
deconzcommunity/deconz:2.18.02 :white_check_mark:
deconzcommunity/deconz:2.18.00 :x: (rebuild 3 months ago?)
deconzcommunity/deconz:2.19.00 :white_check_mark:
deconzcommunity/deconz:2.19.01 :white_check_mark:
deconzcommunity/deconz:2.19.02 :white_check_mark:
deconzcommunity/deconz:2.19.03 :white_check_mark:
deconzcommunity/deconz:2.20.00 :x:
deconzcommunity/deconz:beta :x:
deconzcommunity/deconz:2.20.01 :x:
deconzcommunity/deconz:stable :x:
deconzcommunity/deconz:latest :x:

## Steps to reproduce the behavior

Add a Hue Go device to deconz in affected versions,
or update deconz with a connected Hue Go to 2.20.00 or above.

## Expected behavior

If `swconfigid` isn't defined, it shouldn't return `null`, but remove the key, to avoid mentioned issues in phoscon/hue essential etc.

## Screenshots

## Environment

 - Host system: Raspberry Pi
 - Running method: Debian, docker deconzcommunity/deconz
 - Firmware version: 
 - deCONZ version: see above
 - Device: RaspBee II
 - Do you use an USB extension cable: no -- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? If so: Which?

## deCONZ Logs

## Additional context

ebaauw, 2023-01-31:
If I understand correctly, the JSON is perfectly valid, just not what Hue Essentials expects?  The error message by Hue Essentials is somewhat misleading.  Also, I see no mention of an issue in Phoscon.  It looks to be a bug in Hue Essentials.

I agree, it would be better not to expose `swconfigid` if the Hue Go doesn't support the corresponding Zigbee attribrute.  Even if it does support the attribute, `swconfigid` would be `null` just after pairing the device, before deCONZ has read the attribute for the first time.

Could you try and read the _Basic_ cluster attributes in the GUI, in particular the Hue-specific _Software Config Identifier_, 0x0041.  If that shows a value, `swconfigid` should reflect that.  
If that reports as unsupported, we would need to create separate DDF for the Hue Go, without `swconfigid`.  The DDFs from mentioned commit are based on my observation that ZB3 Hue lights all carry _Software Config Identifier_, but ZLL Hue lights don't.  Could you please provide screenshots from the GUI, in particular the mode with its endpoints and clusters, and the attributes of the _Basic_ cluster?

Thomas-Vos, 2023-01-31:
Correct Erik, the JSON is valid, but just not what Hue Essentials expects. Hue Essentials expects the `swconfigid` attribute to have a non-null value, just like the Hue API. But now this value appears with a null value in deCONZ, which breaks my app.

An update is already available for Android, iOS, macOS, and Windows, with a bug fix where this null value is ignored.

There is no need to change it in deCONZ, unless you want to match the Hue API behaviour.

Issue was originally reported here: https://community.hueessentials.com/t/cannot-connect-to-bridge-invalid-json-after-adding-hue-go/5067

ebaauw, 2023-01-31:
Thanks, Thomas.  Still wondering whether the Hue Go carries _Software Config Identifier_ and the API shows `null` because it hasn’t yet read the attribute, or whether it doesn’t and deCONZ shouldn’t expose `swconfigid` on the Hue Go.

Do you actually use `swconfigid`?  I know the Hue API exposes it, but I never figured out what it’s for.

shin10, 2023-01-31:
The Hue Essential update did the trick. However, the Hue Go is still missing in phoscon. I'll take a further look later tonight.

shin10, 2023-02-04:
OK, this is a little bit embarrassing. The battery was empty and the device was dangling around without a name in deconz. Plugging it in again, the name reappeared and so did the device in phoscon. No idea why the dated phoscon app had no issue to show it though. 🤷

Anyway, I had a look at the basic cluster as requested. Most of my devices didn't have 0x0041; four do.

Two of them (3rd gen. e27 bulbs) have
```json
        "productid": "Philips-LCT010-1-A19ECLv4",
        "swconfigid": "1D51AAA1",
```
and the other two (1st gen. hue go and a hue gradient ambience strip)
```json
        "productid": null,
        "swconfigid": null,
```
returned by the API.

Read within deconz, the last two mentioned show the keys in the list, but grayed out.
_0x0040_ `productid` seems to be an empty string
_0x0041_ `swconfigid` is _0x00000000_

Those four are my gamut type C devices. There's a fifth type C device (also a 3rd gen. e27 bulb), that probably didn't get updates for years and is missing those attributes. 😬

Hope that helps

closed 2023-02-04 by shin10