dresden-elektronik/deconz-rest-plugin #6716: Ikea Tradfri presence sensor falsely turning on all lights

olicooper, 2023-01-31:
## Describe the bug

When the Ikea motion sensor detects motion, all the lights appear to turn on in Home Assistant, however they do not actually turn on. A few seconds later they correct themselves and go back to their real state. I have looked at the logs and found that the sensor sends a button 1002 press broadcast event which might be the issue?

## Steps to reproduce the behavior

Add some lights and Tradfri motion sensor via Phoscon. Use Home Assistant Deconz integration. Add automation which turns on a light based on motion events.

## Expected behavior

Lights not related or controlled by the sensor should not change state / toggle.

## Screenshots

N/A

## Environment

 - Host system: Raspberry Pi
 - Running method: Raspbian
 - Firmware version: (26400500)
 - deCONZ version: (2.19.03)
 - Device: RaspBee I
 - Is there any other USB or serial devices connected to the host system? Wireless USB 2.0 keyboard & mouse

## deCONZ Logs

```log
21:07:38:227 [INFO] - Button 1002 - TRADFRI motion sensor, broadcast to: 0x0000, endpoint: 0x01, cluster: ONOFF (0x0006), action: On with timed off, payload: 0008070000, zclSeq: 5
21:07:38:328 ZCL attribute report 0x70AC08FFFEXXXXXX for cluster: 0x0001, ep: 0x01, frame control: 0x08, mfcode: 0x0000
21:07:38:419 emit event/tick: 0xcc86ecfffexxxxxx
21:07:38:461 SC tick --> StateCallFunction
21:07:38:463 writeTuyaData, dpid: 0x01, type: 0x10, expr: Item.val == 1 ? 1 : 0;
21:07:38:465 Tuya write expression: Item.val == 1 ? 1 : 0; --> 1
21:07:38:466 SC tick --> StateCallFunction
21:07:38:467 writeTuyaData, dpid: 0x02, type: 0x2B, expr: (Item.val / 254.0) * 1000.0;
21:07:38:468 Tuya write expression: (Item.val / 254.0) * 1000.0; --> 503.93700787401576
21:07:38:697 TY_DATA_REPORT: seq 61903, dpid: 0x01, type: 0x01, length: 1, val: 1
21:07:38:697 TY_DATA_REPORT: seq 61903, dpid: 0x01, type: 0x01, length: 1, val: 1
21:07:38:698 TY_DATA_REPORT: seq 61903, dpid: 0x01, type: 0x01, length: 1, val: 1
21:07:38:698 TY_DATA_REPORT: seq 61903, dpid: 0x01, type: 0x01, length: 1, val: 1
21:07:38:700 SC state/on: synced
21:07:38:701 SC --> StateFinished
21:07:38:701 SC state change finished: cc:86:ec:ff:fe:xx:xx:xx-01
21:07:38:736 Set sensor check interval to 100 milliseconds
21:07:38:866 TY_DATA_REPORT: seq 61904, dpid: 0x02, type: 0x02, length: 4, val: 504
21:07:38:868 TY_DATA_REPORT: seq 61904, dpid: 0x02, type: 0x02, length: 4, val: 504
21:07:38:869 TY_DATA_REPORT: seq 61904, dpid: 0x02, type: 0x02, length: 4, val: 504
21:07:38:869 TY_DATA_REPORT: seq 61904, dpid: 0x02, type: 0x02, length: 4, val: 504
21:07:38:871 SC state/bri: synced
21:07:38:872 SC --> StateFinished
21:07:38:872 SC state change finished: cc:86:ec:ff:fe:xx:xx:xx-01
21:07:39:138 [INFO] - Button 1002 - TRADFRI motion sensor, broadcast to: 0x0000, endpoint: 0x01, cluster: ONOFF (0x0006), action: On with timed off, payload: 0008070000, zclSeq: 72
21:07:39:189 send move to color temperature 312 to 0x003C84FFFEXXXXXX
21:07:39:276 delay sending request 197 dt 0 ms to 0x003C84FFFEXXXXXX, ep: 0x01 cluster: 0x0008 onAir: 1
21:07:39:277 delay sending request 198 dt 0 ms to 0x003C84FFFEXXXXXX, ep: 0x01 cluster: 0x0300 onAir: 1
21:07:39:308 delay sending request 197 dt 0 ms to 0x003C84FFFEXXXXXX, ep: 0x01 cluster: 0x0008 onAir: 1
21:07:39:309 delay sending request 198 dt 0 ms to 0x003C84FFFEXXXXXX, ep: 0x01 cluster: 0x0300 onAir: 1
21:07:39:312 ZCL attribute report 0x00158D0004XXXXXX for cluster: 0x0406, ep: 0x01, frame control: 0x18, mfcode: 0x0000
21:07:39:396    0x003C84FFFEXXXXXX force poll (2)
21:07:39:426 send move to color temperature 312 to 0x003C84FFFEXXXXXX
21:07:39:427 Replace task 9618240 type 6 in queue cluster 0x0300 with newer task 9618246 of same type. 1 runnig tasks
21:07:39:427 delay sending request 205 dt 0 ms to 0x003C84FFFEXXXXXX, ep: 0x01 cluster: 0x0300 onAir: 1
21:07:39:429 ZCL attribute report 0x003C84FFFEXXXXXX for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000
```

## Additional context

N/A

BabaIsYou, 2023-02-22:
May be take a look at this  https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/What-you-can-do-when-a-switch-turns-(almost)-all-devices-on-off

olicooper, 2023-02-24:
Hey thanks for that suggestion! I've tried unbinding the broadcast endpoint as the guide suggested but the binding doesn't exist (on the bind dropbox I get "failed: no entry"). These are battery powered devices and the guide says the 'All' group is bound for all routers (mains powered) devices.. I think it might be a bug?

All ZigBee lights and my Ikea plug are exhibiting the same problem in HomeAssistant (anything with on/off control?) and I've only just added the plug to the system so there are no automations or bindings configured.

I could reset and start again I suppose but it seems like a lot of effort if the problem is small.

olicooper, 2023-03-26:
This has not been resolved and needs to stay open

Mimiix, 2023-03-26:
> This has not been resolved and needs to stay open

Please try to keep it active before auto close next time.

closed 2023-04-25 by github-actions[bot]