dresden-elektronik/deconz-rest-plugin #6717: Adeo\Lexman door sensor

Idaho947, 2023-02-01:
## Device

- Product name: Door and vibration sensor
- Manufacturer: ADEO
- Model identifier: LEDSENK08
- Device type : Sensor

## Screenshots

[image]

### Basic
[image]
[image]

### Identify
[image]

### IAS zone
[image]

### Diagnostics
[image]
[image]

### Power
[image]

### Pull Control
[image]

BabaIsYou, 2023-02-01:
It seems that screenshots of "basic" cluster are from "Diagnostic" ?

Idaho947, 2023-02-01:
Arf. I edit my post. Thank you !

BabaIsYou, 2023-02-01:
Did not find from where the vibration sensor values are communicated. But you can try this DDF to get the Open/Close features.
```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "ADEO",
  "modelid": "LEDSENK08",
  "vendor": "Lexman",
  "product": "Lexman Door Window Sensor with Vibration",
  "sleeper": true,
  "status": "Silver",
  "subdevices": [
    {
      "type": "$TYPE_OPEN_CLOSE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0402",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0500"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/checkin"
        },
        {
          "name": "config/enrolled",
          "public": false
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lowbattery"
        },
        {
          "name": "state/open"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0500"
    }
  ]
}
```

Smanar, 2023-02-01:
Do you know how work this device ?
Have you a return for a door contact and one another for vibration, or it's the same thing ?

BabaIsYou, 2023-02-01:
> Do you know how work this device ? Have you a return for a door contact and one another for vibration, or it's the same thing ?

Seems to be determined from ias:status : [https://github.com/brantmacga/zigbee-herdsman-converters/blob/100e5c4a96846fb45367cc0e5a8354a049ff4533/devices/adeo.js](url)

Smanar, 2023-02-01:
So DDF totally experimental ^^

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "ADEO",
  "modelid": "LEDSENK08",
  "vendor": "Adeao",
  "product": "Smart Zigbee Door Window Contact and vibration",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_OPEN_CLOSE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0006"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0402",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0500"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
           "name":"state/tampered"
        },
        {
          "name": "config/checkin"
        },
        {
          "name": "config/enrolled",
          "public": false
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2"
          }
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lowbattery"
        },
        {
          "name": "state/open",
          "parse": {
            "fn": "ias:zonestatus",
            "mask": "alarm1"
          }
        }
      ]
    },
    {
      "type": "$$TYPE_VIBRATION_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0402",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0500"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
           "name":"state/tampered"
        },
        {
          "name": "config/checkin"
        },
        {
          "name": "config/enrolled",
          "public": false
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2"
          }
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lowbattery"
        },
        {
          "name": "state/vibration",
          "parse": {
            "fn": "ias:zonestatus",
            "mask": "alarm2"
          }
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 60,
          "max": 3600,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0500"
    }
  ]
}
```

Idaho947, 2023-02-01:
Ok the sensor appears in deconz :
[image]

But nothing happen when I open \ close.

Idaho947, 2023-02-01:
Ok I tried to re-include it three times without deleting it. I did it and... it works!

[image]

Idaho947, 2023-02-01:
Ah yes sorry I forget to mention that. I've already change it.

BabaIsYou, 2023-02-01:
> If I don't read it it stay at the first step and doesn't work.

Vibration seems not to be automatically cleared, that not like presence detection.

Some has changed ZHAVibration to ZHAPresence to get auto reset, then device appears as presence detection instead of Vibration like in #6270

Idaho947, 2023-02-01:
Yes vibrations works well.

BabaIsYou, 2023-02-01:
> I have to read the ias cluster to link the sensor but when the led blink green its ok

Then seems to be more a binding issue if you have to reead the cluster to have it updated.
Could you replace this part in binding of cluster 0x500 : 
```
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0500"
      "report": [
        {
          "at": "0x0002",
          "dt": "0x19",
          "min": 60,
          "max": 3600
        }
      ]
    }
```

> Yes vibrations works well.

It resets between each detection or just send another event without resseting ?

Idaho947, 2023-02-01:
It reset between détection.

Idaho947, 2023-02-02:
It stil working, you can use this ddf for this sensor.

Smanar, 2023-02-02:
>I have to read the ias cluster to link the sensor but when the led blink green its ok

Nothing to see with your procedure, but this device need an enrollement to work, perhaps this part need more time.
It's visible on log ofc but in the device json too in "config/pending", you can see a wich one phase is the sensor.

So the final DDF is

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "ADEO",
  "modelid": "LDSENK08",
  "vendor": "Adeo",
  "product": "Smart Zigbee Door Window Contact and vibration",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_OPEN_CLOSE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0006"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0402",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0500"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
           "name":"state/tampered"
        },
        {
          "name": "config/checkin"
        },
        {
          "name": "config/enrolled",
          "public": false
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2"
          }
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lowbattery"
        },
        {
          "name": "state/open",
          "parse": {
            "fn": "ias:zonestatus",
            "mask": "alarm1"
          }
        }
      ]
    },
    {
      "type": "$TYPE_VIBRATION_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0402",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0500"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
           "name":"state/tampered"
        },
        {
          "name": "config/checkin"
        },
        {
          "name": "config/enrolled",
          "public": false
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2"
          }
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lowbattery"
        },
        {
          "name": "state/vibration",
          "parse": {
            "fn": "ias:zonestatus",
            "mask": "alarm2"
          }
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 60,
          "max": 3600,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0500",
      "report": [
        {
          "at": "0x0002",
          "dt": "0x19",
          "min": 60,
          "max": 3600
        }
      ]
    }
  ]
}
```

Idaho947, 2023-02-02:
No see nothing in log. This ddf works fine

Smanar, 2023-02-03:
Ok, nice, so who wana make the PR ^^ ?

Idaho947, 2023-02-05:
You can ^^

Smanar, 2023-02-05:
PR here https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6727

closed 2023-02-06 by manup