dresden-elektronik/deconz-rest-plugin #6721: [Device Support Request] BSEED 2 Way Switch _TZ3000_e98krvvk

xADDRx, 2023-02-01:
## Device

- Product name: BSEED Zigbee 2 Way Switch
- Manufacturer: _TZ3000_e98krvvk
- Model identifier: TS0012
- Device type : Switch

## Screenshots

### Basic
[image]

### Identify
[image]

### Groups
[image]
[image]

### Scenes
[image]
[image]
[image]
[image]

### On/Off
[image]
[image]

BabaIsYou, 2023-02-01:
Please can you redo the clusters screenshots after clicking on "read" button to have attributes values updated from the device ?

xADDRx, 2023-02-01:
> Please can you redo the clusters screenshots after clicking on "read" button to have attributes values updated from the device ?

Hi, 
sorry yes of course:
[image]

BabaIsYou, 2023-02-01:
As I can see there is already a DDF for this device in #5837 but as I understand you can't manage each of the switch separatly with it ?

xADDRx, 2023-02-02:
> 

As I can see there is already a DDF for this device in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5837 but as I understand you can't manage each of the switch separatly with it ?  **Yes right**

BabaIsYou, 2023-02-03:
If it requires Tuya magic sequence it's not supported yet :  [here](https://forum.phoscon.de/t/including-standard-tuya-magic-spell-sequence-during-joining/2916/13)

BabaIsYou, 2023-02-23:
Good news,

@Smanar did a great job in doing this Tuya Magic sequence to be integrated in a [DDF](https://forum.phoscon.de/t/including-standard-tuya-magic-spell-sequence-during-joining/2916/15). That's perhaps just a workaround waiting for something more optimized but can try to add the part in the referenced issue in the existing DDF ?

closed 2023-03-25 by github-actions[bot]