dresden-elektronik/deconz-rest-plugin #6724: Device name

Sjonnie2018, 2023-02-03:
Let me start by stating I read the github recommendation and searched the issues/requests before I decided to post this toic!

## Device

- Product name: MOES Wireless Remote Switch
- Manufacturer: _TZ3000_w3c7ouru
- Model identifier: TS0043
- Device type : 3 gang switch
- Remote: remote

## Screenshots

[url=https://postimg.cc/K35F18tD][img]https://i.postimg.cc/K35F18tD/Moes-YSR-ZSW-De-CONZ.jpg[/img][/url]

[url=https://postimg.cc/sMxy2yx5][img]https://i.postimg.cc/sMxy2yx5/2023-02-03-17-39.png[/img][/url]

Device is found in DeCONZ but not in Phoscon. Because of that it is not found in my DeCONZ-Domoticz plugin which I use to access the device in Domoticz.

I know it is not very polite to "link" another user in a thread but a would like to have @Smanar have a look at this.

Sorry if I broke the rules!

BabaIsYou, 2023-02-03:
Hi,

As you don,t stick to the [template ](https://forum.phoscon.de/t/read-me-device-request-read-me/23/2)it will be difficult, event to @Smanar to give some help.

Please stick to the templte by minimally add all the requested information and screenshots.

Sjonnie2018, 2023-02-03:
Tell me what I am missing then? I would like to honor the guidelines. I walked through the template but you obviously think I left something out. Please, enlighten me!

BabaIsYou, 2023-02-03:
When you opened the issue, all was listed ... just follow the [guidelines ](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/new?assignees=&labels=Device+Request&template=device-request.md&title=Device+name)

Sjonnie2018, 2023-02-03:
[image]
[image]
[image]
[image]
[image]
There!

BabaIsYou, 2023-02-03:
Many thanks. Can you just add the screenshot that contains all the clusters (the one displayed when you click on the right grey dot) just to be sure we don't miss informations about endpoints or specifiic clusters ... Then we will certainly be in businessto help you ...  ;-)

Sjonnie2018, 2023-02-03:
I assume you mean this one:

[image]

It is a 3 gang switch as you can see )

BabaIsYou, 2023-02-03:
> I assume you mean this one:

Exactly ! This adds the needed endpoint numbers.

BTW are you sure you click on "read" before this one ?

> [image]

BabaIsYou, 2023-02-03:
BTW could try with this DDF (adapted from [other switches](devices/tuya/_TZ3000_ygvf9xzp_4gang_remote.json) but quite sure that we'll need to change also `button_map` too.

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": ["_TZ3000_ygvf9xzp", "_TZ3000_ee8nrt2l", "_TZ3000_dziaict4", "_TZ3000_mh9px7cq","_TZ3000_w3c7ouru"],
  "modelid": ["TS0044", "TS0044", "TS0044", "TS0044","TS0043"],
 "product": "MOES Switch 3 gangs",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0006"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "refresh.interval": 84000,
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "refresh.interval": 3600,
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2;",
            "fn": "zcl"
          }
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/buttonevent"
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 3600,
          "max": 86400,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0006"
    },
    {
      "bind": "unicast",
      "src.ep": 2,
      "dst.ep": 1,
      "cl": "0x0006"
    },
    {
      "bind": "unicast",
      "src.ep": 3,
      "dst.ep": 1,
      "cl": "0x0006"
    },
    {
      "bind": "unicast",
      "src.ep": 4,
      "dst.ep": 1,
      "cl": "0x0006"
    }
  ]
}
```

Sjonnie2018, 2023-02-04:
I assume you mean this one:

[image]

It is a 3 gang switch as you can see )

> > I assume you mean this one:
> 
> Exactly ! This adds the needed endpoint numbers.
> 
> BTW are you sure you click on "read" before this one ?
> 
> > [image]

Yes, I pressed read before making the screenshot. ;)

I will try to see if I can get the DDF file working. I'll let you know.

Sjonnie2018, 2023-02-04:
Hmmmm.... not so simple :|

I have DeCONZ running in a container so I am not sure where to modify the DDF. I have a persistent volume in my home folder location but I can't find out where to place the DDF. I opened DeCONZ through VNC and found the DDF editor:

[image]

I "think" I can past the DDF code you provided in here and save. But I am not sure where I can change the button_maps.json (couldn't find it).

These are the DeCONZ folders I have in my home folder:

[image]

Guidance appreciated ;)

Smanar, 2023-02-04:
>I "think" I can past the DDF code you provided in here and save

Nope, paste don't work in the editor.

It's your persistent folder on the capture ?
If yes can you take a look in the "devices" folder, you can find inside the buttonmap file and the DDF.

Sjonnie2018, 2023-02-04:
Hi Smanar, thanks for your reply!

Unfortunately the folder is empty:

[image]

I am using the community container:

`services:
  deconz:
    image: deconzcommunity/deconz:stable
    container_name: deconz
    restart: always
    privileged: true                # This is important! Without it, the deCONZ image won't be able to connect to Conbee II.
    ports: 
      - 8080:8080
      - 8443:8443
      - 5900:5900
    volumes:
      - ./opt/deCONZ:/opt/deCONZ
    devices:
      - /dev/ttyACM0                # This is the USB device that Conbee II is running on.
    environment:`

Smanar, 2023-02-05:
Ok so first, good new, it's the good place for DDF https://github.com/deconz-community/deconz-docker/issues/163#issuecomment-1368255052

Now bad new, IDK if it will work for button_maps.json.
And as you need to restart deconz for to use it, I think you need to restart the container, so lose the modifications ...

Try to see if you can use a new button_maps.json, else we will make the PR in blind mode. Your device don't seem to have something special.

Sjonnie2018, 2023-02-05:
I "think" the button_maps.json in the device folder is not working because I can find the switches in Phoscon now but they are identified as being a one gang switch:
[image]
In DeCONZ I can see the switches but they are indicating low battery. Which is strange since the batteries are brand new and have 3.14V left:
[image]
These are th files I created (showing the content, look at the address bar for location and name):
button_maps.json as provided by @BabaIsYou :
[image]
moes-ddf.json also copied from @BabaIsYou :
[image]

Wrong place for the button_maps.json maybe? I could try but my docker host is not the fastest and "rebuilding" the network takes some time.  :-S

BabaIsYou, 2023-02-05:
Hi,

normal that it’s visible as a single switch in DeConz. That’s what is described in the DDF. Only one switch type sensor with button event to track the pressed button. Not sure it’s the best approach but worked with other switch that looked similar. And then seemed to fastest way to have it working 😀

Perhaps another approach would have been to create a switch sensor for each EP based on the on/off state for each. 

Regarding the battery that’s why I asked you [earlier](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6724#issuecomment-1416186461) about the « read » button because as the battery level is null that’s normal that it reports low level. Seems that this device reports battery level differently.

Smanar, 2023-02-05:
The code showed by @BabaIsYou is just an extract, with the part to edit ^^, the complete file is here https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/button_maps.json

I think the DDF was fine but not possible to edit the button_maps.json, and I m not skilled enought to explain you how to use dockers.

Sjonnie2018, 2023-02-05:
Found the original file and edited the part @BabaIsYou indicated. Uploaded it to the device folder. Restarted the container and will now have to wait until the Zigbee mesh is fully populated with all devices.

I'll report back when the mesh is complete.

Sjonnie2018, 2023-02-05:
No luck! The mesh is (almost) complete but the switches are still indicated as a 1 gang switch.

I can only hope someone is able to tell me where the button_maps.json is supposed to be.

Smanar, 2023-02-05:
>but the switches are still indicated as a 1 gang switch

It's indicated nowhere. The only difference betwen a 1 gang and a 4 gang is the second one is able to report more values.

Have you tried to pressa button to see if the button_maps is working or not ? It's visible on the "state/buttonevent" value. On domoticz or phoscon/help/API Information/sensor
else you will have error message visible with "info" in deconz logs

Mimiix, 2023-02-05:
> I "think" the button_maps.json in the device folder is not working because I can find the switches in Phoscon now but they are identified as being a one gang switch: [image] In DeCONZ I can see the [...]

Ignore phoscon. That needs to add support on its own.

Sjonnie2018, 2023-02-05:
> Ignore phoscon. That needs to add support on its own.

OK, understood. But that does not resolve my issue in DeCONZ (or Domoticz for that matter).

And do you know, by any chance, where I should put the button_maps.json file when using the "official" community container?

Smanar, 2023-02-05:
>And do you know, by any chance, where I should put the button_maps.json file when using the "official" community container?

And if you  put it in the "devices" folder in the perisistent folder ?
Are you sure it don't work ? You still have "null" as value in "buttonevent" ?

Sjonnie2018, 2023-02-05:
I am not entirely sure what you mean by:

¨You still have "null" as value in "buttonevent" ?¨

Could you please explain what I have to do?

Thank you!

Smanar, 2023-02-05:
Take a look in the state/buttonevent value in the device json. It's the value used by third app.
You can see it in domoticz/custom/deconz/sensors or phoscon/help/API Information/sensors

Smanar, 2023-02-06:
Have made the PR https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6728
But just by curiosity, if you can try on your side to know if it's possiuble to make a custom buttonmap on docker ?

Sjonnie2018, 2023-02-06:
Hi Smanar,

Sorry for my late reply!

Is this what you are looking for?

`{
  "config": {
    "battery": 0,
    "on": true,
    "reachable": true
  },
  "etag": "cbf8d81c1c0c893318ec4be39aec8741",
  "lastannounced": null,
  "lastseen": "2023-02-06T16:58Z",
  "manufacturername": "_TZ3000_w3c7ouru",
  "mode": 1,
  "modelid": "TS0043",
  "name": "Schakelaar-Controlroom-Wand",
  "state": {
    "buttonevent": null,
    "lastupdated": "none"
  },
  "type": "ZHASwitch",
  "uniqueid": "a4:c1:38:c5:97:d0:2a:66-01-0006",
  "id": "77",
  "deviceclass": "sensors"
}`

I don't know how to make a custom button map on Docker. I could create an extra "persistent" volume if I knew which folder inside the container I need to map. I have these folders in the root of the container:
[image]

Thanks for your continued support (and PR :) )

BabaIsYou, 2023-02-06:
> Is this what you are looking for?

That's it, exactly. Can you put the modified button_maps.json in the same folder that you put the DDF file ?

if `"state": { "buttonevent": null, "lastupdated": "none" }` continues to have a `null` value when you press a switch then can you get error messages from deConz UI like @smanar asked for [above ](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6724#issuecomment-1418081237)

Smanar, 2023-02-09:
Yep the file need to be edited.
And from the comment from the docker github, it seem it's the good method and can work.

Sjonnie2018, 2023-02-10:
I have rebuilt my server and placed the ddf and the complete json in persistent volume under devices. Restarted DeCONZ but no luck. Should I wait for PR 6728?

Smanar, 2023-02-10:
>Should I wait for PR 6728?

Yes, this will be fine, the PR is small, and even not tested, 99 % of tuya switch are same, so I think it will be ok.

I m sorry, but I don't think use a custom button_maps.json is possible on docker, even on the official one.

Sjonnie2018, 2023-02-11:
I would probably not need a button_maps.json file when the MOES has been added, wouldn't you think?

One more question: should I change my container to beta releases or is the change been made in the stable version?

Thanks!

Smanar, 2023-02-11:
No the PR add the DDF and edit the button_maps.json, you will just need to re-include the device if you aren't using the DDF ATM, else you will have nothing to do.

>One more question: should I change my container to beta releases or is the change been made in the stable version?

IDK, I think the chnage will be in the next one, as it s something realy light, but I can't guess if it will be a beta or a stable.

Sjonnie2018, 2023-02-11:
OK, so I'll wait for the next release (I will change my container to the tag Latest). For now I delete the two 3 gang switches and wait for an update.

Thanks as always for your invaluable support!

Sjonnie2018, 2023-03-05:
Well I just saw a commit pass by so I hope that will be included.

Smanar, 2023-03-05:
Not yet, this issue will be closed when the PR will be valided.

Mimiix, 2023-03-14:
> I finally had the opportunity to install the switches combined with the Moes Smart dimmer 105. The dimmer works but the switches will not give the proper "signals" to DeCONZ/Domoticz. It will only register the first button press (in this case button 3) and after that only the first button registers an event in Domoticz (using the DomoticzDeCONZ plugin from a famous programmer ;-) ).
> 
> The Domoticz log shows me this:
> 
> ` 2023-03-14 18:42:53.110 ConBee2: ### WebSocket Data : {'attr': {'id': '79', 'lastannounced': None, 'lastseen': '2023-03-14T17:42Z', 'manufacturername': '_TZ3000_w3c7ouru', 'modelid': 'TS0043', 'name': 'Schakelaar-Controlroom-Wand', 'swversion': None, 'type': 'ZHASwitch', 'uniqueid': 'a4:c1:38:c5:97:d0:2a:66-01-0006'}, 'e': 'changed', 'id': '79', 'r': 'sensors', 't': 'event', 'uniqueid': 'a4:c1:38:c5:97:d0:2a:66-01-0006'} 2023-03-14 18:43:53.158 ConBee2: ### WebSocket Data : {'attr': {'id': '79', 'lastannounced': None, 'lastseen': '2023-03-14T17:43Z', 'manufacturername': '_TZ3000_w3c7ouru', 'modelid': 'TS0043', 'name': 'Schakelaar-Controlroom-Wand', 'swversion': None, 'type': 'ZHASwitch', 'uniqueid': 'a4:c1:38:c5:97:d0:2a:66-01-0006'}, 'e': 'changed', 'id': '79', 'r': 'sensors', 't': 'event', 'uniqueid': 'a4:c1:38:c5:97:d0:2a:66-01-0006'}`
> 
> And this was using the following sequence: press third button for a short time, press first button for a short time and finally press first button for a long time. I can't tell the difference from the log entries.
> 
> Maybe you can help me getting the switch to work as a switch.
> 
> Note that the switch is identified in Domoticz with the following options: B1, B2, B3, D1, D2, D3 L1, L2 ,L3. Unfortunately these options aren't registered at all.
> 
> Any help appreciated!

Are we talking 2 different devices? If that is the case, please stay on topic.

Sjonnie2018, 2023-03-14:
No, Dennis, We are not! Please re-read my post to conclude that I referenced the Moes dimmer 105 for clarity on this subject. Apparently clarity caused confusion.

To be absolutely clear: The 3 gang switch is not (fully) recognized despite the PR Smanar (I won't "call him with an @) has subjected. I can not discern the different buttons when pressed. As such the 3 gang switch is operating as a 1 gang switch. That's how it is recognized in Phoscon although that is totally irrelevant in regards to DeCONZ REST plugin.

Thank you for responding.

Smanar, 2023-03-15:
Can be a problem in the domoticz plugin.
Your log is not useful because there is not the important part, the websocket return with a "buttonevent" and the value.
You need to have a different value for all action. Else it mean the DDF is not working at all.

You have too edited the button_maps.json file ? It's not something to do on docker.

Sjonnie2018, 2023-03-15:
Hi Smanar,

I added the log from Domoticz to show there is no difference in result when pressing various buttons. ;)

I have not modified the DDF because that is rather cumbersome as I use DeCONZ in a Docker container. So it might be the plugin.

Hope you're able to help me on this!

Best regards,

Sjonnie

Smanar, 2023-03-15:
Edit a DDF is easier as you are using the official docker, you just need to create a file in the shared/persistent folder `/opt/deconz`
But as I can see on the issue you have opened on their github, it's not possible to do same for the button_maps.json file, and don't update this file can explain your incomplete logs.

And to update this file, to need to replace the file inside the docker, and it's not something easy, and it's the reason for I will never use docker.

Sjonnie2018, 2023-03-19:
I found a new version of Deconz. Unfortunately the response from the switches is the same.

Editing the button_maps.json is indeed impossible (or I don't know how to dot hat) so I will patiently wait until the image is updated.

I have been thinking to buy other 3 gang switches but they are all rather unknown in the DeCONZ universe so buying other switches is a gamble. Do you perhaps know of a brand/model that is working with DeCONZ?

TIA

Smanar, 2023-03-19:
Manup have set the milestone v2.22.0-beta  on the PR.

I don't understand the logic for numeration, but it 's the next milestone, so I think the next version.
When the PR will be valided, this PR will be closed automatically.

>Do you perhaps know of a brand/model that is working with DeCONZ?

Tuya is a problem as it's not possible to know the model you will have. But you just need to edit 2 files if your model is a new one, unfortunately just edit a simple file is a nightmare on docker.
It's the only brand on my mind that make clones (but different the first version) so fastly

> have been thinking to buy other 3 gang switches but they are all rather unknown in the DeCONZ universe

Wich one you want to try ?

Sjonnie2018, 2023-03-20:
I was thinking of this [one:](https://www.aliexpress.com/item/1005003299695211.html) but no type/model number is published. That's why it is such a gamble (and I haven't won the lottery ;) ). So I think it's better to focus on getting my current switches working.

Smanar, 2023-03-20:
>That's why it is such a gamble

Lol yes , I think it can have at least 5/6 different manufacture name.
Sorry for delay, but I think now it s the last one

Sjonnie2018, 2023-03-20:
When the DDF or Button_maps for this switch are not going to work I just might change my original plan (3 gang switches) to a different one (Aqara 2 gang D1 wireless switch). I have a number of the Aqara 2 gang switches and they are fully supported. Not ideal but a solution as a last resort ;)

Smanar, 2023-03-21:
Badly, there is ATM no project to make the Button_maps easier to edit.
DDF will be soon easier to manage, even without the GUI, but I don't see something that will help you for the buttonmap file.

Sjonnie2018, 2023-04-02:
Received deCONZ version 2.21.2 yesterday afternoon. No changes to the behaviour of the 3 gang switch. So hopefully 2.22.0 will make the 3 gang switch fully functional. :)

BabaIsYou, 2023-04-02:
You can follow pull request #6728 … current milestone is 2.22.0-beta

Sjonnie2018, 2023-04-02:
Yes! Thanks. Still trying to get used to the PR methodology. Saw that ETA for 2.20.0-beta is April and stable is also expected in April. Will refrain from extending this particular topic and follow #6728 from now on.

Sjonnie2018, 2023-04-21:
Hi All,

Received the new deCONZ Beta V2.22.0 container today and tested the MOES 3 Gang Switch (TZ3000_w3c7ouru) within deCONZ and Domoticz. The good news is that the switch is seen as a 3 gang switch (4 gang icon in Phoscon but that is probably unrelated) and single click and double click works as expected. I only have a hard time to get the long click (press ;-) ) recognized.

I had the following results in Domoticz:

Left button:
Single click -> value 10
Double click -> value 30
Long click -> no response

Middle button:
Single click -> value 40
Double click -> value 60
Long click -> no response

Right button:
Single click -> value 70
Double click -> value 90
Long click -> no rsponse.

I was expecting values of 20, 50 and 80 respectively.

Is there something I can do to make long click functional?

Note that I am using deCONZ in a container and it seems to be impossible to create a button_maps.json for this device.

Any help appreciated!

## Screenshots
Sorry that I made screenshots that are not following guidelines but I think you will see the information required. All screenshots were made after pressing the appropriate button on the switch.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Smanar, 2023-04-21:
This device, again ... ^^
I m searching information about this device "_TZ3000_w3c7ouru" but I don't find it on other zigbee project.

Can you make 2 tries.
- on domoticz, enable log ("debug info only"), and make a long press, you will perhaps see a websocket request. If no mean the problem is from deconz
- on deconz, on the GUI logs enable "info" and "info_l2", and make a long press, you will see infomation about button map, else it probably mean the device don't support long press.

Sjonnie2018, 2023-04-21:
Yes! I am sorry! This device again ;-)

I was able to catch the log entries in Domoticz. They are:
` 2023-04-21 17:14:06.292 ConBee2: ### WebSocket Data : {'attr': {'id': '79', 'lastannounced': None, 'lastseen': '2023-04-21T15:14Z', 'manufacturername': '_TZ3000_w3c7ouru', 'modelid': 'TS0043', 'name': '3 knops schakelaar Controlroom W', 'swversion': None, 'type': 'ZHASwitch', 'uniqueid': 'a4:c1:38:c5:97:d0:2a:66-01-0006'}, 'e': 'changed', 'id': '79', 'r': 'sensors', 't': 'event', 'uniqueid': 'a4:c1:38:c5:97:d0:2a:66-01-0006'}
2023-04-21 17:14:06.294 ConBee2: ### Update device (3 knops schakelaar Controlroom Wand) : {'nValue': 10, 'sValue': '10'}`

Unfortunately I can't get a screen grab from deCONZ. After selecting the required parameters the log is flushed with entries and I can't select the entries in VNC to copy them over to my computer. Is there a log *file*  I can revert to?

Smanar, 2023-04-21:
I think it's more a deconz problem.
The websocket seem incomplete.

On the deconz log, you can uncheck the "info_l2" to stop the spam when you see the request. If I m right it's the more talkative.

You need to see the same log than in domoticz 

`{'attr': {'id': '79', 'lastannounced': None, 'lastseen': '2023-04-21T15:14Z', 'manufacturername': '_TZ3000_w3c7ouru', 'modelid': 'TS0043'`

But with more information just before about button map.

Sjonnie2018, 2023-04-21:
As a wild thought: if there is a wireless 3 gang switch that *is* working with deCONZ with single press/double press/long press. I am all ears. I wouldn't mind investing in another switch if it is guaranteed to work with deCONZ.

Smanar, 2023-04-21:
IDK, it s the same code used for all tuya switch.
So strange 1 is working and not another, some have different mode enabled with press combinaison on the device.

And if you see something even empty on domoticz when pressing the button, something need to be see somewhere.

Sjonnie2018, 2023-04-21:
Tx! This is what I got:

`17:36:53:600 rule event /config/localtime: 17:36:52.587 -> 17:36:53.588 (1001ms)
17:36:53:681 poll node 14:b4:57:ff:fe:4c:35:67-01
17:36:53:682 Poll light node Dressoir A school rechts
17:36:53:733 Poll APS request to 0x14B457FFFE4C3567 cluster: 0x0006 dropped, values are fresh enough
17:36:54:588 poll node cc:cc:cc:ff:fe:9e:8c:c3-01
17:36:54:589 Poll light node Tafellamp Links
17:36:54:590 Idle timer triggered
17:36:54:591 Force read attributes for ZHASwitch SensorNode Schakelaar Trap Boven
17:36:54:592 don't create binding for attribute reporting of sensor Schakelaar Trap Boven
17:36:54:593 Force binding of attribute reporting for node Schakelaar Trap Boven
17:36:54:620 rule event /config/localtime: 17:36:53.588 -> 17:36:54.588 (1000ms)
17:36:54:736 Poll APS request to 0xCCCCCCFFFE9E8CC3 cluster: 0x0006 dropped, values are fresh enough
17:36:55:607 rule event /config/localtime: 17:36:54.588 -> 17:36:55.588 (1000ms)
17:37:04:902 ZCL attribute report 0x00158D000587D117 for cluster: 0x0000, ep: 0x01, frame control: 0x1C, mfcode: 0x115F 
17:37:04:902 0x00158D000587D117 extract Xiaomi special attribute 0xFF01
17:37:04:902 	01 battery 3215 (0x0C8F)
17:37:04:903 	03 Device temperature 21 °?C
17:37:04:903 	04 unknown 5032 (0x13A8)
17:37:04:903 	05 RSSI dB (?) 144 (0x0090)
17:37:04:903 	06 LQI (?) 4294967296 (0x0100000000)
17:37:04:904 	08 unknown 5129 (0x1409)
17:37:04:904 	0a Parent NWK 23239 (0x5AC7)
17:37:13:368 [INFO] - Button 1002 - TS0043, unicast to: 0x0000, endpoint: 0x01, cluster: ONOFF (0x0006), action: B1 short, payload: 00, zclSeq: 194
17:37:13:496 Send Tuya request 0xDC8E95FFFE5F1241 : Dp_type: 0x02, Dp_identifier 0x02, data: 000001d4
17:37:13:498 Tuya debug 10: EP: 1 ID : 13
17:37:13:499 Send Tuya request 0xDC8E95FFFE5F1241 : Dp_type: 0x01, Dp_identifier 0x01, data: 01
17:37:13:500 delay sending request 11 dt 0 ms to 0xDC8E95FFFE5F1241, ep: 0x01 cluster: 0xEF00 onAir: 1
17:37:13:612 Tuya debug 4 : Address 0xDC8E95FFFE5F1241 Payload 013102020004000001d4
17:37:13:614 Tuya debug 5 : Status: 1 Transid: 49 Dp: 514 (0x02,0x02) Fn: 0 Data 468
17:37:13:914 Tuya debug 4 : Address 0xDC8E95FFFE5F1241 Payload 01320101000101
17:37:13:915 Tuya debug 5 : Status: 1 Transid: 50 Dp: 257 (0x01,0x01) Fn: 0 Data 1
17:37:14:024 Tuya debug 4 : Address 0xDC8E95FFFE5F1241 Payload 013302020004000001d4
17:37:14:025 Tuya debug 5 : Status: 1 Transid: 51 Dp: 514 (0x02,0x02) Fn: 0 Data 468
17:37:22:537 ZCL attribute report 0x000D6F0011F9F811 for cluster: 0x0402, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
17:37:22:581 ZCL attribute report 0x000D6F0011F9F811 for cluster: 0x0405, ep: 0x02, frame control: 0x08, mfcode: 0x0000 
17:37:23:939 Bind response success for 0x000d6f0011f9f811 ep: 0x02 cluster: 0x0000
17:37:24:498 Bind response success for 0x000d6f0011f9f811 ep: 0x01 cluster: 0x0000`

Screenshot to add to the fun:
[image]

Smanar, 2023-04-22:
`2023-04-21 17:47:25.981 ConBee2: ### Update device (3 knops schakelaar Controlroom Wand) : {'nValue': 40, 'sValue': '40'}`

For me here, the device just use the previous value, because there is nothing in the previous websocket event, you need a "state" field inside and more precisely a state/buttonevent.

`17:37:13:368 [INFO] - Button 1002 - TS0043, unicast to: 0x0000, endpoint: 0x01, cluster: ONOFF (0x0006), action: B1 short, payload: 00, zclSeq: 194`

It's exaclty the needed line. but on this test you have done a short press ?

Sjonnie2018, 2023-04-22:
Hi Smanar,

Yes, I only long pressed the button to get the readings.

And indeed it is the battery operated version of the 3 gang switch.

HTH

Smanar, 2023-04-22:
>Yes, I only long pressed the button to get the readings.

So this device make short press when you make long press ? and if you make a short press ? Same result ?

`[INFO] - Button 1002 - TS0043, unicast to: 0x0000, endpoint: 0x01, cluster: ONOFF (0x0006), action: B1 short`

We need to find something different for this device on short/long press, to deconz be able to differentiate them.

Smanar, 2023-04-22:
@Sjonnie2018 can you check in the manual you haven't a special mode, enabld for exemple with 3 press on the middle button (an event mode)

[5 more replies]

closed 2023-04-23 by Sjonnie2018