dresden-elektronik/deconz-rest-plugin #6730: Tuya (QOTO) Zigbee Wireless Soil Moisture Sensor Model QT-07S

riccardo-casazza, 2023-02-07:
## Device

- Product name: Tuya (QOTO) Zigbee Wireless Soil Moisture Sensor Model QT-07S
- Manufacturer: QOTO
- Model identifier: QT-07S
- Device type : 
  - Sensor

## Description
Hey, **I have no access to deconz GUI, but I can get logs**. I'm posting the logs that I obtained by executing:

`/usr/bin/deCONZ -platform minimal --dbg-aps=2 --dbg-info=2 --dbg-error=2 | tee debug.txt`

and by clicking on "add new sensor" in http://host/pwa application and trying to pair the device.

The product is [this one](https://www.qotosmart.com/TUYA-Garden-Soil-Moisture-Probe-pd44541219.html) and is basically a sensor providing soil humidity (0-100%) and soil temperature (-20°-> 60°) and (I hope) level of battery.

Logs are here: https://pastebin.com/RnJvhdFY
Are these logs enough? If not how can I help?

Mimiix, 2023-02-07:
Hi,

Please add all screenshots. You do need the gui for that. It can't be done otherwise.

Kind regards,

closed 2023-03-09 by github-actions[bot]

Guliver, 2023-06-08:
Hi,

Can this issue be reopened, if I would provide the screenshots? Otherwise I would create a new issue.

Thanks.

Mimiix, 2023-06-08:
@Guliver  I appreciate and prefer a new issue, as its useful to have all screens on the first post :)

Guliver, 2023-06-08:
The device was already implemented with https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6731. Great work!