dresden-elektronik/deconz-rest-plugin #6731: Device name

JocWal, 2023-02-07:
## Device

- Product name: Tuya Soil Sensor 
- Manufacturer: _TZE200_myd45weu
- Model identifier: TS0601
- Device type :
  - Moisture Sensor
  - Temperature Sensor

## Screenshots
[image]
[image]

### Node Info Panel
[image]
[image]

### Basic
[image]

### Groups
[image]

### Scenes
[image]

### OTAU
[image]

### Tuya specific
[image]

BabaIsYou, 2023-02-07:
May could try this DDF. Not 100% sure about the tuya DP ...
```
{
	"schema": "devcap1.schema.json",
	"manufacturername": "_TZE200_myd45weu",
	"modelid": "TS0601",
	"vendor": "TS0601",
	"product": "Tuya Soil Sensor",
	"sleeper": true,
	"status": "Silver",
	"subdevices": [{
			"type": "$TYPE_TEMPERATURE_SENSOR",
			"restapi": "/sensors",
			"uuid": [
				"$address.ext",
				"0x01",
				"0x0402"
			],
			"items": [{
					"name": "attr/id"
				},
				{
					"name": "attr/lastannounced"
				},
				{
					"name": "attr/lastseen"
				},
				{
					"name": "attr/manufacturername"
				},
				{
					"name": "attr/modelid"
				},
				{
					"name": "attr/name"
				},
				{
					"name": "attr/swversion",
					"parse": {
						"fn": "zcl",
						"ep": 1,
						"cl": "0x0000",
						"at": "0x0001",
						"script": "tuya_swversion.js"
					},
					"read": {
						"fn": "zcl",
						"ep": 1,
						"cl": "0x0000",
						"at": "0x0001"
					}
				},
				{
					"name": "attr/type"
				},
				{
					"name": "attr/uniqueid"
				},
				{
					"name": "config/battery",
					"read": {
						"fn": "none"
					},
					"parse": {
						"dpid": 5,
						"eval": "Item.val = Attr.val / 2 ;",
						"fn": "tuya"
					},
					"default": 0
				},
				{
					"name": "config/offset"
				},
				{
					"name": "config/on"
				},
				{
					"name": "config/reachable"
				},
				{
					"name": "state/lastupdated"
				},
				{
					"name": "state/temperature",
					"read": {
						"fn": "none"
					},
					"parse": {
						"dpid": 1,
						"eval": "Item.val = Attr.val;",
						"fn": "tuya"
					},
					"default": 0,
					"awake": true
				}
			]
		},
		{
			"type": "$TYPE_HUMIDITY_SENSOR",
			"restapi": "/sensors",
			"uuid": [
				"$address.ext",
				"0x01",
				"0x0405"
			],
			"items": [{
					"name": "attr/id"
				},
				{
					"name": "attr/lastannounced"
				},
				{
					"name": "attr/lastseen"
				},
				{
					"name": "attr/manufacturername"
				},
				{
					"name": "attr/modelid"
				},
				{
					"name": "attr/name"
				},
				{
					"name": "attr/swversion",
					"parse": {
						"fn": "zcl",
						"ep": 1,
						"cl": "0x0000",
						"at": "0x0001",
						"script": "tuya_swversion.js"
					},
					"read": {
						"fn": "zcl",
						"ep": 1,
						"cl": "0x0000",
						"at": "0x0001"
					}
				},
				{
					"name": "attr/type"
				},
				{
					"name": "attr/uniqueid"
				},
				{
					"name": "config/battery",
					"read": {
						"fn": "none"
					},
					"parse": {
						"dpid": 5,
						"eval": "Item.val = Attr.val / 2;",
						"fn": "tuya"
					},
					"default": 0
				},
				{
					"name": "config/offset"
				},
				{
					"name": "config/on"
				},
				{
					"name": "config/reachable"
				},
				{
					"name": "state/humidity",
					"read": {
						"fn": "none"
					},
					"parse": {
						"dpid": 4,
						"eval": "Item.val = Attr.val;",
						"fn": "tuya"
					},
					"default": 0,
					"awake": true
				},
				{
					"name": "state/lastupdated"
				}
			]
		}
	]
}
```

JocWal, 2023-02-09:
Hi there,
Thanks for your help!
in general I got it to work. I had to change the dpid's though.
I tried the id's from the post you pointed to and the moisture and temp id's seem to be the correct ones:
[dpid: 3, 'soil_moisture'] - had to scale the raw value since putting it into tap water showed 70% moisture -> "eval": "Item.val = Attr.val * 142.857;"
[dpid: 5, 'soil temperature'] - also scaled by 100 to get the value in °C.

However I was not able the get battery level readout so far. I tried dpid 15 from the same post (see below), but it didn't work.

[9, 'temperature_unit', tuya.valueConverter.temperatureUnit],
[14, 'battery_state', tuya.valueConverter.batteryState],
[15, 'battery', tuya.valueConverter.raw],

BabaIsYou, 2023-02-09:
Well, then you have : 
Battery - dpid 5 or 15  -» not working
Humidity - dpid 3 -» Working after * 142.857 ?! Usually this number has to be adjusted by multiples of 10, not with a so strange value
Temperature - dpid 5 --» Working after * 100

That's it ?

Have you tried dpid 4 for battery ?

Smanar, 2023-02-11:
>[15, 'battery', tuya.valueConverter.raw],

Harf, I have see it recently, it seem deconz can't handle raw value, but need to make c++ code edition to correct that.
It can explain why I have ths issue on TRV too.

I think you can see log for other dpid but nothing for this one ?

`21:44:58:635 TY_DATA_REPORT: seq 66, dpid: 0x69, type: 0x02, length: 4, val: 5003`

JocWal, 2023-03-08:
> Well, then you have : Battery - dpid 5 or 15 -» not working Humidity - dpid 3 -» Working after * 142.857 ?! Usually this number has to be adjusted by multiples of 10, not with a so strange value Temperature - dpid 5 --» Working after * 100
> 
> That's it ?
> 
> Have you tried dpid 4 for battery ? But still surprised because it really seems that it's dpip 15 https://github.com/Koenkk/zigbee-herdsman-converters/blob/bcdb6b3c593cac61741cbe9f5254606de73f67f7/devices/tuya.js#L1268

I tried both 4 and 15 for the battery - neither worked unfortunately. 

There's another issue I found with humidity. The raw value seems to change resolution blow 1% resulting in an jump in the value reported by phoscon.

[image]

BabaIsYou, 2023-03-08:
> BTW it's not humidity, this device don't have it, but soil moisture, and from Z2M it's with the dp 3, not 4, it can explain the strange

But we don,t have a State/soil_moisture attribute then that's why we use state/humidity (that's only a semantic difference BTW ;-))

for dp i found another source with dp3 : https://github.com/Koenkk/zigbee-herdsman-converters/blob/bcdb6b3c593cac61741cbe9f5254606de73f67f7/devices/tuya.js#L1268 which one is the real one ?

Smanar, 2023-03-08:
And we can't use the "state/moisture" ? I can see this one in the official code ATM.

BabaIsYou, 2023-03-08:
Good news, as I didn't see it in the generic item list I thought that it didn't exist 
[image]

Smanar, 2023-03-09:
It's something recent. And effectively the json is still missing, something like

```
{
	"schema": "resourceitem1.schema.json",
	"id": "state/moisture",
	"datatype": "UInt16",
	"access": "R",
	"public": true,
	"description": "The current relative moisture in percent.",
	"parse": {
		"at": "0x0000",
		"cl": "0x0408",
		"ep": 0,
		"eval": "Item.val = Attr.val",
		"fn": "zcl"
	},
	"default": 0
}
```

psfpwsfngolwle, 2023-03-24:
Thanks for the DDF. I have a soil sensor which optically is exactly the same, just the model name is different - mine is called GXM-01.
I have tried out the DDF with dpid
3 for moisture
5 for temperature
15 for battery
and I can confirm it works.

My only question is: Can anyone explain the strange moisture * 142.857 multiplication? Why is it exactly that value?

Smanar, 2023-03-24:
>My only question is: Can anyone explain the strange moisture * 142.857 multiplication? Why is it exactly that value?

Lol, Yes, I think there is a problem here too. And other project I can't see this convertion.

What is your manufacture name ? (To add it in the PR, or if you want to make it ? )

And if you can share the final DDF with all changes.

Smanar, 2023-03-24:
Thx.
But on your DDF you are not using state/moisture too ?
Do you want I make a PR with your DDF and all files needed for the moisture one ?

BabaIsYou, 2023-03-24:
> Thanks for the DDF. I have a soil sensor which optically is exactly the same, just the model name is different - mine is called GXM-01. I have tried out the DDF with dpid 3 for moisture 5 for [...]

I think that this value was just calculated assuming that tap water is 100% relative humidity in the air compared to mesured value by device. Then 2 choices, mesured value is not humidity, or water is not 100% relative humidity in the air … I guess that it’s the second option because relative humidity is a combination of pressure/temp in the air.

Smanar, 2023-03-25:
From Z2M the value is a percent, you can just check if the value never go under 100.
But IDK if you can see it without new json file ?

BabaIsYou, 2023-03-27:
> I think it's better to create a new sensor, a ZHAMoisture

Like the idea ;-)

Smanar, 2023-03-27:
For me it's a good thing because this sensor already exist in code, so better to use it.
Just need to know if @psfpwsfngolwle want to make the PR or not ?

psfpwsfngolwle, 2023-03-27:
Thanks for your responses. Sorry, I am quite new to this - what does PR stand for?

Some further insights on comparing the values from DPID 3 on this soil sensor with the ones from a another old model I have (not using Zigbee). I have placed both of them outside next to each other. While the old one shows continiously in-/decresing values - and it outlines that is has rained during the period with values going to 70+, the Zigbee model jumps quite back and forth and in fact it stays at 60 since Mar 25 1:30 am. You can see that by opening both attached screenshots next to each other.

So I am wondering if this soil sensor is really a reliable option respectivel,y are we using the right DPID for reading the moisture?

Do you have similar observations? Are your values (100, 61.43, 60, ...) the same?

[image]
[image]

BabaIsYou, 2023-03-28:
> Need to add the ZHAMoisture in the DDF, and a state/moisture json in the "generic" folder.

And a new constant in constants.json ? ;-)

Smanar, 2023-03-28:
>could this have an impact to the way the device reports datas ?

Yes, it's exacty the problem, without the unlock sequence device simply don't send report.

>And a new constant in constants.json and a subdevice like generic/subdevices/moisture_sensor.json ? ;-)

Need to add a file called "state_moisture_item.json" in "devices\generic\items" with 

```
{
	"schema": "resourceitem1.schema.json",
	"id": "state/moisture",
	"datatype": "UInt16",
	"access": "R",
	"public": true,
	"description": "The current relative moisture in percent.",
	"parse": {
		"at": "0x0000",
		"cl": "0x0408",
		"ep": 0,
		"eval": "Item.val = Attr.val",
		"fn": "zcl"
	},
	"default": 0
}
```

psfpwsfngolwle, 2023-03-29:
I have used the definitions from the previous post, 
1) put the files into the respective folders
2) restarted deconz
3) deleted the pair sensors (humidity/ temperature)
4) Paired the soil sensor again.

The result is that I get a temperature and a moisture sensor and the temperature sensor seems to provide the right values. On the moisture sensor, I also get the values for battery, reachable, ... but the most important - the value for "moisture" is completely missing now.
Any thoughts?

BabaIsYou, 2023-03-29:
Do you have some error in log (with INFO, DDF and APS checked) for that device ?

Smanar, 2023-03-29:
Yep if you can take a look in logs ?
But its like a problem with the "state_moisture_item.json" file, but I don't see what I m doing wrong.

BEskandari, 2023-03-29:
Hello, if I can be of any help. I tried also the latest 2 files and I get the below results, the moisture is not working.

**For Temperature sensor:**
{
    "config": {
        "battery": 100,
        "offset": 0,
        "on": true,
        "reachable": true
    },
    "etag": "ea11fdc3a6f1b917aca09b0950ca45be",
    "lastannounced": null,
    "lastseen": "2023-03-29T14:21Z",
    "manufacturername": "_TZE200_myd45weu",
    "modelid": "TS0601",
    "name": "Temperature 20",
    "state": {
        "lastupdated": "2023-03-29T13:53:15.938",
        "temperature": 2500
    },
    "type": "ZHATemperature",
    "uniqueid": "a4:c1:38:af:22:d1:cc:76-01-0402"
}
**For Moisture sensor:**
{
    "config": {
        "battery": 100,
        "offset": 0,
        "on": true,
        "reachable": true
    },
    "etag": "86a1a5f226eb6b2ec46344de73b3afde",
    "lastannounced": null,
    "lastseen": "2023-03-29T14:30Z",
    "manufacturername": "_TZE200_myd45weu",
    "modelid": "TS0601",
    "name": "Moisture 21",
    "state": {
        "lastupdated": "2023-03-29T13:53:16.128",
        "moisture": 0
    },
    "type": "ZHAMoisture",
    "uniqueid": "a4:c1:38:af:22:d1:cc:76-01-0408"
}

Smanar, 2023-03-29:
>Need to add a file called "state_moisture_item.json" in "devices\generic\items" with

BTW, what is your OS ? it the system "devices" folder not the "user" one, so not possible on docker OS (or relay hard)

BEskandari, 2023-03-29:
Hope that the debug log is usable.
Raspberry Pi 4, Deconz 2.21.0, the 2 files are installed in **/home/pi/.local/share/dresden-elektronik/deCONZ/devices**:
under **./tuya** for **TS0601_Tuya_Soil_Sensor.json** (the * 100 was deleted)
under **./generic/items** for **state_moisture_item.json**

[Deconz-debug.log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/11103736/Deconz-debug.log)

BabaIsYou, 2023-03-29:
> Actually on Deconz-GUI I do not see the moisture

You won't see it DecONZ-GUI (it's "masked" under Tuya Cluster) , only thru REST-API in Phoscon or third-party software.

> Hope that the debug log is usable.

We can see reporting about battery in the log for both sensors : 
```
19:24:21:017 a4:c1:38:af:22:d1:cc:76-01-0402/config/battery expression: Item.val = Attr.val; --> 100
19:24:21:018 TY_DATA_REPORT: seq 230, dpid: 0x0F, type: 0x02, length: 4, val: 100
19:24:21:018 TY_DATA_REPORT: seq 230, dpid: 0x0F, type: 0x02, length: 4, val: 100
19:24:21:019 a4:c1:38:af:22:d1:cc:76-01-0408/config/battery expression: Item.val = Attr.val; --> 100
19:24:21:020 TY_DATA_REPORT: seq 230, dpid: 0x0F, type: 0x02, length: 4, val: 100
19:24:21:021 TY_DATA_REPORT: seq 230, dpid: 0x0F, type: 0x02, length: 4, val: 100
```
But I couldn't locate values for temp or moisture in the log. But if you have one now that's great.
Do you know what is the theoric precision for the moisture mesure of this device ? 

We can also see report of dpid 14 (?) that we don't use at this time : 
```
19:24:21:185 APS-DATA.indication srcAddr: 0x1e87, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0xEF00, lqi: 255, rssi: -64
19:24:21:186 TY_DATA_REPORT: seq 231, dpid: 0x0E, type: 0x04, length: 1, val: 2
19:24:21:187 TY_DATA_REPORT: seq 231, dpid: 0x0E, type: 0x04, length: 1, val: 2
19:24:21:187 TY_DATA_REPORT: seq 231, dpid: 0x0E, type: 0x04, length: 1, val: 2
19:24:21:187 TY_DATA_REPORT: seq 231, dpid: 0x0E, type: 0x04, length: 1, val: 2
```

BabaIsYou, 2023-03-29:
After some search, it seems that humidity accuracy for this device is fluctuant : 0~50%（±3%）,50%~100%（±5%).

Then without calibration it's difficult to say what `1` you got for moisture means, Is it 10% or 1% ? If your reference is 6% then, with the expected precision I suggest it's 10% ... but we need other measure ...

BabaIsYou, 2023-03-29:
Not really … I already got the information I need about this device and already looking at other projects to complete those that are not in reviews or seller ads. Because there is no calibration for moisture it’s difficult to be sure that the returned value is accurate or not … that’s why we just need some empiric tests just to see if the value is in a « correct » range or if we need to multiply it. Can you just add some water in you test soil just to got the variation ?

Smanar, 2023-03-30:
> 
> Hope that the debug log is usable.
> Raspberry Pi 4, Deconz 2.21.0, the 2 files are installed in /home/pi/.local/share/dresden-elektronik/deCONZ/devices:
> under ./tuya for TS0601_Tuya_Soil_Sensor.json (the * 100 was deleted)
> under ./generic/items for state_moisture_item.json

So strange it finaly work.
It's the good place for user DDF, but can't work for the "state_moisture_item.json", this one need to be in the deconz folder, you need to see other ones "state_........json" insode.

thogens, 2023-03-30:
Hi!
I have the same sensor. Can you wrap up quickly what I should do to enable it properly? So war I have just "humidity" enabled, but want to go the moisture path...
Thanks,
Thomas

Noxeus, 2023-03-30:
I've added the device, it's visible in VNC as a Temperature sensor. "Editting" the DDF shows the same DDF as I added, but it's still not visible in Phoscon, what am I doing wrong?

ebaauw, 2023-03-30:
@smanar, I assume `state/moisture` is normalised to report the percentage of water in the soil in units of as 0.01%, as per the ZCL definition of _Measured Value_ in cluster 0x0408, and consistently with `state/humidity` for 0x0405?  So values would be between 0 and 10000?

BEskandari, 2023-03-30:
As @ebaauw says maybe we have to multiply the value with *10000 instead of taking out the *100, since 3 hours I am testing the moisture and I have quit similar results between 2 devices. As of now my reference shows 26% and as you see the ZHAMoisture shows 28 it might be 280000.  @BabaIsYou & @Smanar what do you think?

{
    "config": {
        "battery": 100,
        "offset": 0,
        "on": true,
        "reachable": true
    },
    "etag": "50ddafa324c1ce6fd274f8d7bf2e6c25",
    "lastannounced": null,
    "lastseen": "2023-03-30T21:09Z",
    "manufacturername": "_TZE200_myd45weu",
    "modelid": "TS0601",
    "name": "Moisture 21",
    "state": {
        "lastupdated": "2023-03-30T21:01:21.331",
        "moisture": 28
    },
    "type": "ZHAMoisture",
    "uniqueid": "a4:c1:38:af:22:d1:cc:76-01-0408"
}

ebaauw, 2023-03-30:
By 100, so 28% is reported as 2800.

BEskandari, 2023-03-30:
@ebaauw , actually as you see the REST API reports 28, but I have no idea if this value must be 2800 or 280000 as you say. Because in the DDF actually we have:
        {
          "name": "state/moisture",
          "read": {
            "fn": "none"
          },
          "parse": {
            "dpid": 3,
            "eval": "Item.val = Attr.val ;",
            "fn": "tuya"
          },
          "default": 0,
          "awake": true
        }

psfpwsfngolwle, 2023-03-30:
I have tried several approaches now all with the same result. The moisture sensor gets created but without a value for moisture - as mentioned before. I figure out that the "humidity" sensor which was initially created after pairing it with the first DDF where I used humidity gets recreated after I reboot deconz even though I had deleted it via the API before. That might be a reason for the weird behaviour on my side but I am also not clear what I else I can do. If you have any experience with that, please let me know.

BabaIsYou, 2023-03-31:
As @BEskandari seems to have right value I think that DDF is correct. May be we can multiply it by 100 get compliant with other devices.

For situation of @psfpwsfngolwle, it looks like symptoms of a fantom -FF endpoint. Please @psfpwsfngolwle could you get once more json of your sensors about this device ? Just to be sure about your situation ?

Noxeus, 2023-03-31:
Hi everyone, 
I think if you put the `generic/items/state_moisture_item.json` in the persistent `/mnt/data/supervisor/addons/data/core_deconz/.local/share/dresden-elektronik/deCONZ/devices` folder it actually loads. It seems to do for me anyway.

I get the following from the REST-API:

```

  |  
-- | --
lastannounced | "2023-03-30T15:21:08Z"
lastseen | "2023-03-31T06:55Z"
manufacturername | "_TZE200_myd45weu"
modelid | "TS0601"
name | "Moisture 60"
state |  
lastupdated | "2023-03-31T06:37:05.169"
moisture | 3800
type | "ZHAMoisture"
```
so that seems to be in working order, right?

BabaIsYou, 2023-03-31:
Yes it seems good 👍

Smanar, 2023-03-31:
So if values go from 0 to 10000, it's fine for me too.
Someone want to make a PR with the DDF and the item json ?

BabaIsYou, 2023-03-31:
> So if values go from 0 to 10000, it's fine for me too. Someone want to make a PR with the DDF and the item json ?

And adding new constant in constants.json and a subdevice like generic/subdevices/moisture_sensor.json ? ;-)

psfpwsfngolwle, 2023-03-31:
An update from my side:
I finally got it to work and the API requests do show me proper values now. In order to have it in my home automation solution I also had to adapt/ enhance the logic to create Zigbee devices as "moisture" is not known there at this point.
I would definitely recommend to have the * 100 multiplication as this would give more consistency overall - to my knowledge, e.g. temperature and humidity are also having a 0-10000 range in the API and therefore it would make sense to do it the same way for moisture.

thogens, 2023-04-01:
HI! I followed the seps above and was able to get it running. Many thanks! I deleted the *100, as I like to have it at a percentage scale between 0-100. Works good for me.
Thomas

Smanar, 2023-04-02:
>I deleted the *100, as I like to have it at a percentage scale between 0-100

You need to have a value between 0>10000, not 0>100, it's your third app that need to divide the value by 100.
We are doing that to make the code more consistent, moisture will work like humidity.

thogens, 2023-04-02:
Hmm... I have a couple of aqara humidity sensors and all I get over the API is a value between 0-100%. On moisture I don't get percent now, but a number instead. That is why I adjusted the DDF to make it more look like humidity in my case...

ebaauw, 2023-04-02:
Then they're exposed wrongly.  Which models?  The `lumi.weather` and `lumi.airmonitor.acn01` correctly report `state.humidity` in units 0.01%, so values between 0 and 10000.

thogens, 2023-04-02:
I was referring to this [multisensor](https://www.aqara.com/us/temperature_humidity_sensor.html). Using Deconz to fetch the API. The value was automatically mapped to percentages, and the values are 0-100%. Actually for humidity, I would expect exactly that and nothing else.

ebaauw, 2023-04-02:
That's the `lumi.weather`, by the looks of it?  I don't understand what you mean by "Using deCONZ to fetch the API".  How are you interacting with the API.  See https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/deCONZ-for-Dummies to understand the difference between deCONZ, the API, and the Phoscon web app.

> The value was automatically mapped to percentages, and the values are 0-100%.

By whom?  Where are you looking?  The API doesn't include a `%` sign in its response.

psfpwsfngolwle, 2023-04-03:
I am getting back to the values which the sensor provides and still look weird on my end. I assume it has nothing to do with DDF but maybe with the hardware itself. I have placed this soil sensor next to an older one (not based on Zigbee) in an outdoor setup. We had quite some rain over the last days and the mositure is quite high. The old one reports continiously changes and as we are back to sun the soil starts to dry and this is reflected. The Tuya soil sensor does also show a decrease but with up & down hops of the moisture value - this happens since 3 days for me now. Following the log excerpt below you can see that there is even a pattern: Every ~20 seconds the  moisture value jumps to a higher level to then decrease again to the "right" level 2 seconds later. With "right" I mean a comparable level to the other sensor I have.
Does anyone of you have a similar behaviour or any idea why this is happening?

[image]

thogens, 2023-04-03:
> That's the `lumi.weather`, by the looks of it? I don't understand what you mean by "Using deCONZ to fetch the API". How are you interacting with the API. See [...]

Sorry for not being clear. I'm using ioBroker with the respective deCONZ plugin to fetch any data from the API. The % is the unit that the plugin adds for "humidity", but it could be set to anything else. It doesn't matter. What matters is the value itself. Could very well be that the plugin does some conversion itself, I can't see that. The only thing I see is that humidity is shown as a value between 0-100 (%). I can work with any data, don't get me wrong. But having a value between 0-100 (%) for humidity makes so much sense to me. And I don't want to fork this thread either!
Thomas

BabaIsYou, 2023-04-03:
> Same should happen for `state/moisture`, so DDF need to multiply the value reported by the sensor by 100, so the deCONZ REST API reports values 0 - 10000.

It's the way the PR have already been made.

Smanar, 2023-04-03:
>Does anyone of you have a similar behaviour or any idea why this is happening?

Good question ....

If you enable log in deconz (Flag DDF + info + info_l2), you will be able to see the decrypted tuya request and the DDf action. Can be usefull to see if the issue is from the device or the code.

Nothing similar on other zigbee project.

BEskandari, 2023-04-03:
@psfpwsfngolwle, I have also compared with a bluetooth hardware and for example I have a temperature of 8° on BT and 12° on the zigbee and for moisture it is even worse I have 18% on BT and 38% on zigbee.

Smanar, 2023-04-04:
There is no report for humidity in your logs (need the dpid 0x03)
But its like this one (for dpid = 0x0F = battery)

```
20:22:09:237 TY_DATA_REPORT: seq 30, dpid: 0x0F, type: 0x02, length: 4, val: 90
20:22:09:238 a4:c1:38:af:22:d1:cc:76-01-0408/config/battery expression: Item.val = Attr.val; --> 90
```

The first line is the raw value send by the device.
The second line is the "magic" done by deconz to update the field.

BEskandari, 2023-04-05:
@Smanar thanks for your hint, below a new debug file with:

**Moisture**
```
20:57:16:550 TY_DATA_REPORT: seq 107, dpid: 0x03, type: 0x02, length: 4, val: 33
20:57:16:551 a4:c1:38:af:22:d1:cc:76-01-0408/state/moisture expression: Item.val = Attr.val * 100; --> 3300
```
and

**Temperature**
```
20:57:16:287 TY_DATA_REPORT: seq 106, dpid: 0x05, type: 0x02, length: 4, val: 12
20:57:16:288 a4:c1:38:af:22:d1:cc:76-01-0402/state/temperature expression: Item.val = Attr.val * 100; --> 1200
```
Still some doubt on the precision of that hardware.

[TuyaSoilSensor-NewDebug.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/11162234/TuyaSoilSensor-NewDebug.txt)

Smanar, 2023-04-06:
Better logs, but not enought ^^.

In fact you need to wait for 2 successives values. If you see for exemple 

`20:57:16:550 TY_DATA_REPORT: seq 107, dpid: 0x03, type: 0x02, length: 4, val: 33`

And just after

`20:57:16:550 TY_DATA_REPORT: seq 107, dpid: 0x03, type: 0x02, length: 4, val: 11`

It will be device fault, deconz just read values send by the device

But if you see similar value and something different for 

`20:57:16:551 a4:c1:38:af:22:d1:cc:76-01-0408/state/moisture expression: Item.val = Attr.val * 100; --> 3300`

The device will send correct value, but deconz fail with making "magic"

Smanar, 2023-04-07:
Not a problem.
It's device report, not customisable, so if the device want to send them twice .... It's not deconz that make a poll.

>but for the temperature and moisture value I have the read once in 1 hour!

But on your capture here https://github,.com/dresden-elektronik/deconz-rest-plugin/issues/6731#issuecomment-1493807219 You have a value every 2 mn sometime ?

BEskandari, 2023-04-07:
@Smanar thank you for your followup.

I know, by using Homebridge Deconz I have the temperature every 10 minutes in Eve App, but no Moisture because for the time being the ZHAMoisture is not yet implemented.

But even for the temperature the strange thing is that in the Deconz debug log I have it just once in one hour.

Also the **strange point is that for the firmware version I have "0"**.

This might be because of my implementation temporary implementation. 
Anyway looking forward to have the latest Deconz beta for checking the implementation.

Smanar, 2023-04-08:
>Also the strange point is that for the firmware version I have "0".

You mean on the field "swversion" in the device json ? Where have you put the DDF ?

But it's not logic you have a websocket event to the homebridge and no update visible on log in deconz .....
You need to have them in same time like here

```
20:57:16:552 TY_DATA_REPORT: seq 107, dpid: 0x03, type: 0x02, length: 4, val: 33
20:57:16:553 Websocket 192.168.2.56:54852 send message: {"e":"changed","id":"21","r":"sensors","state":{"lastupdated":"2023-04-05T18:57:16.551","moisture":3300},"t":"event","uniqueid":"a4:c1:38:af:22:d1:cc:76-01-0408"} (ret = -1095410824)
```

BEskandari, 2023-04-08:
@Smanar, as previously reported, I have Raspberry Pi 4, Deconz 2.21.2.

The 2 files are installed in **/home/pi/.local/share/dresden-elektronik/deCONZ/devices**:
**under ./tuya** for TS0601_Tuya_Soil_Sensor.json (the * 100 was deleted)
**under ./generic/items** for state_moisture_item.json

You metioned that the ./generic/items would not work, but if I put them under the original generic/items it is deleted at the reboot.
And for swversion, I presume that as we do reference to *"script": "tuya_swversion.js"* this later file is not under my personal tuya directory but only on the original directory, this might be the cause of **swversion=0**

Anyway as @ebaauw has made the ZHAMoisture now to Homebridge Deconz, I will be able to report the values from the Eve App.

Let me check this and come back to you.

[98 more replies]

PR #6839 (2023-04-13): DDF for Tuya soil sensor (_TZE200_myd45weu / TS0601)
Adding BTW some generic json fles for moisture as suggested in #6731 
Thus this close #6731 

Thanks to everybody involved in this process ;-)
PR #6897 (2023-04-17): Update Tuya Soil Sensor DDF
Change reused "config/configured" item by new "config/tuya_unlock" and add "lowbattery" forgotten in previous PR #6842 

Definitely close #6731
closed 2023-04-13 by manup