dresden-elektronik/deconz-rest-plugin #6733: MOES Tuya Zigbee BHT-002 wall thermostat TZE200_aoclfnxz

ltspicer, 2023-02-08:
[image]

[image]

### Basic
[image]

### DDF editor
[image]

I can't find this device in the compatibility list.

Pairing with Phoscon app works and the device is displayed in the deCONZ GUI.
The device works.
In the ioBroker adapter I see it as a sensor named “Thermostat 11”.
But the device is not listed in the Phoscon app.

I use RaspBee2 on RPI3 with Bullseye.

DDF:
`{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZE200_aoclfnxz",
  "modelid": "TS0601",
  "product": "Tuya_THD BTH-002 Thermostat",
  "sleeper": false,
  "status": "Draft",
  "subdevices": [
    {
      "type": "$TYPE_THERMOSTAT",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0201"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0051",
        "endpoint": "0x01",
        "in": [
          "0xEF00",
          "0x0000",
          "0xEF00"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/productid",
          "public": false,
          "description": "Product identifier of the device.",
          "refresh.interval": 86400
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/heatsetpoint",
          "description": "Target temperature of a thermostat.",
          "default": 0
        },
        {
          "name": "config/locked",
          "description": "Locks or unlocks the device. This item has to be understood as child-lock.",
          "default": false
        },
        {
          "name": "config/offset",
          "description": "Relative offset to the main measured value.",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/preset",
          "description": "Sets the operating mode for Tuya thermostats."
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "config/schedule",
          "description": "On-device heating schedules."
        },
        {
          "name": "config/schedule_on",
          "description": "Determines if on-device schedules for setting the heatsetpoint are currently used or if the thermostat is operated manually.",
          "default": false
        },
        {
          "name": "config/setvalve"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/on",
          "description": "True when device is on; false when off.",
          "refresh.interval": 5
        },
        {
          "name": "state/temperature",
          "description": "The current temperature in °C &times; 100.",
          "default": 0
        }
      ]
    }
  ]
}`

Mimiix, 2023-02-08:
Missing screenshots from all clusters.

Mimiix, 2023-02-08:
> Missing screenshots from all clusters.

Additionaly, make sure to add the first paragraph containing the model id , manufacturer name and other info

ltspicer, 2023-02-08:
Now I find the device there:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3304

closed 2023-02-08 by Mimiix