dresden-elektronik/deconz-rest-plugin #6734: Tuya Photoelectric Smoke Detector _TZE200_m9skfctm

BabaIsYou, 2023-02-09:
## Device

- Product name: Tuya Photoelectric Smoke Detector [Link](https://www.aliexpress.com/item/1005003970944017.html?spm=a2g0o.order_list.order_list_main.10.747f1802aTUNEp)
- Manufacturer:  _TZE200_m9skfctm
- Model identifier:TS0601
- Device type 
  - Sensor

## Screenshots

[image]

### Basic
[image]

### Groups
[image]
[image]
[image]

### Scenes
[image]

Information found [here](https://github.com/Koenkk/zigbee-herdsman-converters/blob/bcdb6b3c593cac61741cbe9f5254606de73f67f7/devices/tuya.js#L2660)

BabaIsYou, 2023-02-09:
Testing following DDF

```
{
	"schema": "devcap1.schema.json",
	"manufacturername": "_TZE200_m9skfctm",
	"modelid": "TS0601",
	"vendor": "Tuya",
	"product": "Photoelectric Smoke sensor",
	"sleeper": true,
	"status": "Silver",
	"subdevices": [{
		"type": "$TYPE_FIRE_SENSOR",
		"restapi": "/sensors",
		"uuid": [
			"$address.ext",
			"0x01",
			"0xef00"
		],
		"items": [{
				"name": "attr/id"
			},
			{
				"name": "attr/lastannounced"
			},
			{
				"name": "attr/lastseen"
			},
			{
				"name": "attr/manufacturername"
			},
			{
				"name": "attr/modelid"
			},
			{
				"name": "attr/name"
			},
			{
				"name": "attr/swversion",
				"refresh.interval": 86400,
				"parse": {
					"fn": "zcl",
					"ep": 1,
					"cl": "0x0000",
					"at": "0x0001",
					"script": "tuya_swversion.js"
				},
				"read": {
					"fn": "zcl",
					"ep": 1,
					"cl": "0x0000",
					"at": "0x0001"
				}
			},
			{
				"name": "attr/type"
			},
			{
				"name": "attr/uniqueid"
			},
			{
				"name": "config/battery",
				"awake": true,
				"read": {
					"fn": "none"
				},
				"parse": {
					"dpid": 15,
					"eval": "Item.val = Attr.val / 2;",
					"fn": "tuya"
				},
				"default": 0
			},
			{
				"name": "config/enrolled",
				"public": false
			},
			{
				"name": "config/on"
			},
			{
				"name": "config/pending"
			},
			{
				"name": "config/reachable"
			},
			{
				"name": "state/fire",
				"awake": true,
				"read": {
					"fn": "none"
				},
				"parse": {
					"dpid": 1,
					"eval": "Item.val = (Attr.val == 1);",
					"fn": "tuya"
				},
				"default": false
			},
			{
				"name": "state/test",
				"read": {
					"fn": "none"
				},
				"parse": {
					"dpid": 101,
					"eval": "Item.val = Attr.val;",
					"fn": "tuya"
				},
				"default": 0
			},
			{
				"name": "state/errorcode",
				"read": {
					"fn": "none"
				},
				"parse": {
					"dpid": 11,
					"eval": "Item.val = String(Attr.val);",
					"fn": "tuya"
				},
				"default": "0"
			},
			{
				"name": "state/lastupdated"
			}
		]
	}]
}
```

closed 2023-02-09 by BabaIsYou

BabaIsYou, 2023-02-10:
Will be able to add another attribute regarding smoke ppm when PR [#6673](https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6673 ) will be merged

        {
          "name": "state/airqualityco2",
          "awake": true,
          "read": {
            "fn": "none"
          },
          "parse": {
            "dpid": 2,
            "eval": "Item.val = Attr.val ;",
            "fn": "tuya"
          },
          "default": 0
        },

BabaIsYou, 2023-02-14:
Hi. A bit weird because mine is fully functional with that DDF and a test return an event with test:true.

Do you mean that you have a fire alarm but the device does not make a at noise ?

I don’t see any version information in your request. Then I suppose that you don’t have the tuya_version.js under your local directory. Can you make a screenshot from DeConz GUI basic cluster to show the version please (may be a new firmware ?)

Bernhard-Bodensee, 2023-02-14:
[image]
 
When I press the butten the device make the beep beep beep... until I press again. Looks normal.
I never installed any tuya stuff, how can I check that?
The versions of conbee/phoscon and homebridge are the latest.
I also tried to remove everything, strange to me: the sensor still exists. Can there be some caching problem, some old stuff is used somewhere?
I am using a browser w/o caching and I have cleared the homebridge cache already. But I dont know how to clear caches in deconz or phoscon???

Bernhard-Bodensee, 2023-02-14:
My  configuration:
raspberry 3B mit Linux raspberrypi 5.10.60-v7+ #1449 SMP Wed Aug 25 15:00:01 BST 2021 armv7l GNU/Linux
Homebridge [v1.6.0]
homebridge-hue v0.13.51
deCONZ_ConBeeII_0x26780700.bin.GCF
phoscon/deconz:
[image]

BabaIsYou, 2023-02-14:
> When I press the butten the device make the beep beep beep... until I press again. Looks normal.

Yes that's the test button. Then you should have an event in Phoscon with `"test": true`

> I also tried to remove everything, strange to me: the sensor still exists.

Where is it still present ? In Deconz, Phoscon, Homebridge ? Everywhere ? How did you delete it ?

Bernhard-Bodensee, 2023-02-14:
[image]

still the same, when I hit the test button, "test": true.
but in every record it shows "fire":true , this is probably the reason for showing the alarm in homekit and eve permanently.
I have not found any info how to factory reset the device, when I hit the button for longer it seems to go into pairing mode only.

Regarding "removal" and start all over: This is how I did it:
I removed battery from device, deleted the node in deconz, deleted the DDF-File (local), stopped deconz, restarted deconz and homebridge and homekit - then the device was still in all of them with alarm on!

BTW: The device has never been visible in Phoscon but I learned that that is normal for temporary DDFs.

What would be the correct way/sequence of actions to remove the 2 smoke devices currently paired and start all over ?

BabaIsYou, 2023-02-14:
> "phoscon-GW: /sensors/19: warning: unknown ZHAFire sensor etc..."

Got exactly the same ... I don't really care about Phoscon for this device... I sometimes used it just to have a friendly gui to access some information like json ;-)

It's weird that your device present `"fire": true ` because the default in DDF is false. Than perhaps means that your device seems to present really an alarm and it's may be faulty ?

To delete a device, it's seems you did it like it could be done. Perhaps you could try to delete the device before remove battery and restart deconz bfore putting the battery back and repairing the device.

It's strange that the device stay after delete. You say that you have two of them, is it the previous "deleted" one and the second the device after rejoining ? Do they have different sensor id ? And how could Deconz handle the same `uniqueid` on two difference sensor ?

Could you provide for both what you called API-Info [previously](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6734#issuecomment-1429523314) ?

Bernhard-Bodensee, 2023-02-14:
[image]
[image]
[image]

The 2 devices are different, got different Mac/NWK, there is no mix, deconz and all other components process that correctly. I paired fire-20 this morning for the first time just to see if it behaves the same as fire-19.
I would assume that it is unlikely that 2 devices got the same fault, but still possible. I will try more things tomorrow - I would hate my neighbors calling fire brigade caused by my test alarms...

BabaIsYou, 2023-02-15:
Ok I understand. And you’re right, chances to get two faulty devices at the same time … hmmm 🤔

Bernhard-Bodensee, 2023-02-15:
[image]

I could not reproduce that the device is still visible somewhere after removal in deconz. Maybe I fell into a trap with data from cache.
Every removal and repairing of the same device results in a new sequence number fire-19, fire-20, fire 21 etc- with the same uniqueid - that looks normal.
However, Result in API-Info/events  is always the same:
fire = true
If I hit the test button that is shown inaddition to fire = true.

I played around a bit in the DDF setting the default to different values like "0" and "false" or empty etc - no change.
I am lost now, I hope you got some idea on how to proceed.

BabaIsYou, 2023-02-15:
> Every removal and repairing of the same device results in a new sequence number fire-19, fire-20, fire 21 etc- with the same uniqueid - that looks normal.

Yes, it is !

> I hope you got some idea on how to proceed.

Not he idea of the century for sure : did you try to click in basic cluster on exec button to reset to factory default ? In the mean time could you get some log with just INFO and DDF checked ? it's sufficient to get DDF evals and Tuya request/response ?

Bernhard-Bodensee, 2023-02-15:
I hit the exec button. But nothing really happened. I would expect that the device would require repairing after factory reset but it just continued working and spreading "fire: true".
How do I get the log info you requested :"get some log with just INFO and DDF checked ? it's sufficient to get DDF evals and Tuya request/response ?"

BTW: swversion is null or that line is missing in event records. 
```
   "22:51:39:526": {
        "attr": {
            "id": "22",
            "lastannounced": "2023-02-15T21:43:28Z",
            "lastseen": "2023-02-15T21:51Z",
            "manufacturername": "_TZE200_m9skfctm",
            "modelid": "TS0601",
            "name": "Fire 22",
            "swversion": null,
            "type": "ZHAFire",
            "uniqueid": "a4:c1:38:b3:eb:2a:c7:e9-01-ef00"
        },
        "e": "changed",
        "id": "22",
        "r": "sensors",
        "t": "event",
        "uniqueid": "a4:c1:38:b3:eb:2a:c7:e9-01-ef00"
    }
}
```
                   
In the DDF it shows:
```
name": "attr/swversion",
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001"}
```
I don´t see a script file "tuya_swversion.js" in my Home dir. Could that be a problem?

BabaIsYou, 2023-02-16:
> Could that be a problem?

It shouldn't. You can make a copy of /usr/share/deCONZ/devices/tuya/tuya_swversion.js to the same folder of your DDF if you want to be sure, but IFIAK it doesn't really matter.

> How do I get the log info you requested :"get some log with just INFO and DDF checked ? it's sufficient to get DDF evals and Tuya request/response ?"

In fact you already gave it previously [here](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6734#issuecomment-1429523314). Can you propuce it again with just INFO and DDF checked ?
[image]

BabaIsYou, 2023-02-16:
To be very honest, I based my DDF on other plateform information (ie : https://github.com/Koenkk/zigbee-herdsman-converters/commit/1ca479c45e804eb6827cc2bac912ec0091be71d2) but I have some doubt about what does `tuya.valueConverter.true0ElseFalse` function means. 
Could it be that it is mine that is faulty and that the default correct value would be 1 instaed of 0 ?

Bernhard-Bodensee, 2023-02-16:
Thanks a lot for your great support and fast replies - and the patience with me...

Step by step - I hope I am not presenting this in a too confusing way. My simple observation:

1. Tuya version still wrong even after copying tuya_swversion.js into $HOME and .local...device where the DDF resides. But that is not the main problem. Lets forget about that.

2. This is the debug view with "eval": "Item.val = (Attr.val == 1);",like in the master DDF-file,  after insertion of battery (shows the dpid 0x01 =1 for true)
```
18:58:28:184 Bind response success for 0x60a423fffe118350 ep: 0x0B cluster: 0x0006
18:58:28:239 ZCL configure reporting rsp seq: 64 0x60A423FFFE118350 for ep: 0x0B cluster: 0x0006 attr: 0x0000 status: 0x00
18:58:47:491 DeviceAnnce of SensorNode: 0xA4C138B3EB2AC7E9 [1]
18:58:49:179 Bind response success for 0x60a423fffe118350 ep: 0x0B cluster: 0x0006
18:58:49:180 skip configure report for cluster: 0x0006 attr: 0x0000 of node 0x60A423FFFE118350 (seems to be active)
18:58:52:142 TY_DATA_REPORT: seq 93, dpid: 0x65, type: 0x01, length: 1, val: 0
18:58:52:143 TY_DATA_REPORT: seq 93, dpid: 0x65, type: 0x01, length: 1, val: 0
18:58:52:143 TY_DATA_REPORT: seq 93, dpid: 0x65, type: 0x01, length: 1, val: 0
18:58:52:144 a4:c1:38:b3:eb:2a:c7:e9-01-ef00/state/test expression: Item.val = Attr.val; --> false
18:58:52:145 TY_DATA_REPORT: seq 93, dpid: 0x65, type: 0x01, length: 1, val: 0
18:58:52:286 TY_DATA_REPORT: seq 94, dpid: 0x01, type: 0x04, length: 1, val: 1
18:58:52:287 TY_DATA_REPORT: seq 94, dpid: 0x01, type: 0x04, length: 1, val: 1
18:58:52:288 a4:c1:38:b3:eb:2a:c7:e9-01-ef00/state/fire expression: Item.val = (Attr.val == 1); --> true
18:58:52:289 TY_DATA_REPORT: seq 94, dpid: 0x01, type: 0x04, length: 1, val: 1
18:58:52:290 TY_DATA_REPORT: seq 94, dpid: 0x01, type: 0x04, length: 1, val: 1
18:58:52:410 TY_DATA_REPORT: seq 95, dpid: 0x02, type: 0x02, length: 4, val: 0
18:58:52:411 TY_DATA_REPORT: seq 95, dpid: 0x02, type: 0x02, length: 4, val: 0
18:58:52:411 TY_DATA_REPORT: seq 95, dpid: 0x02, type: 0x02, length: 4, val: 0
18:58:52:412 TY_DATA_REPORT: seq 95, dpid: 0x02, type: 0x02, length: 4, val: 0
18:58:52:553 a4:c1:38:b3:eb:2a:c7:e9-01-ef00/config/battery expression: Item.val = Attr.val; --> 84
18:58:52:553 TY_DATA_REPORT: seq 96, dpid: 0x0F, type: 0x02, length: 4, val: 84
18:58:52:554 TY_DATA_REPORT: seq 96, dpid: 0x0F, type: 0x02, length: 4, val: 84
18:58:52:555 TY_DATA_REPORT: seq 96, dpid: 0x0F, type: 0x02, length: 4, val: 84
18:58:52:555 TY_DATA_REPORT: seq 96, dpid: 0x0F, type: 0x02, length: 4, val: 84
18:58:52:676 TY_DATA_REPORT: seq 97, dpid: 0x0B, type: 0x01, length: 1, val: 0
18:58:52:677 a4:c1:38:b3:eb:2a:c7:e9-01-ef00/state/errorcode expression: Item.val = String(Attr.val); --> false
18:58:52:678 TY_DATA_REPORT: seq 97, dpid: 0x0B, type: 0x01, length: 1, val: 0
18:58:52:678 TY_DATA_REPORT: seq 97, dpid: 0x0B, type: 0x01, length: 1, val: 0
18:58:52:679 TY_DATA_REPORT: seq 97, dpid: 0x0B, type: 0x01, length: 1, val: 0
18:58:52:830 ZCL attribute report 0xA4C138B3EB2AC7E9 for cluster: 0x0000, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
18:58:52:849 ZCL attribute report 0x60A423FFFE118350 for cluster: 0x0006, ep: 0x0B, frame control: 0x18, mfcode: 0x0000 
18:59:57:707 0x60a423fffe118350 found group 0xFFF0
18:59:57:708 0x60a423fffe118350 found group 0x0001
18:59:58:454 0xec1bbdfffeaed1f8 found group 0xFFF0
18:59:58:455 0xec1bbdfffeaed1f8 found group 0x0001
```
3. This is the debug view with "eval": "Item.val = (Attr.val == 0);",  as you recommended in your last posting, after insertion of battery (shows the dpid 0x01 =1 for true), same as above. Dont be confused by the system time shown, I tried that first
```
18:52:36:267 DeviceAnnce of SensorNode: 0xA4C138B3EB2AC7E9 [1]
18:52:40:944 TY_DATA_REPORT: seq 119, dpid: 0x65, type: 0x01, length: 1, val: 0
18:52:40:945 TY_DATA_REPORT: seq 119, dpid: 0x65, type: 0x01, length: 1, val: 0
18:52:40:946 TY_DATA_REPORT: seq 119, dpid: 0x65, type: 0x01, length: 1, val: 0
18:52:40:947 a4:c1:38:b3:eb:2a:c7:e9-01-ef00/state/test expression: Item.val = Attr.val; --> false
18:52:40:948 TY_DATA_REPORT: seq 119, dpid: 0x65, type: 0x01, length: 1, val: 0
18:52:41:088 TY_DATA_REPORT: seq 120, dpid: 0x01, type: 0x04, length: 1, val: 1
18:52:41:090 TY_DATA_REPORT: seq 120, dpid: 0x01, type: 0x04, length: 1, val: 1
18:52:41:092 a4:c1:38:b3:eb:2a:c7:e9-01-ef00/state/fire expression: Item.val = (Attr.val == 0); --> false
18:52:41:093 TY_DATA_REPORT: seq 120, dpid: 0x01, type: 0x04, length: 1, val: 1
18:52:41:094 TY_DATA_REPORT: seq 120, dpid: 0x01, type: 0x04, length: 1, val: 1
18:52:41:219 TY_DATA_REPORT: seq 121, dpid: 0x02, type: 0x02, length: 4, val: 0
18:52:41:220 TY_DATA_REPORT: seq 121, dpid: 0x02, type: 0x02, length: 4, val: 0
18:52:41:221 TY_DATA_REPORT: seq 121, dpid: 0x02, type: 0x02, length: 4, val: 0
18:52:41:221 TY_DATA_REPORT: seq 121, dpid: 0x02, type: 0x02, length: 4, val: 0
18:52:41:349 a4:c1:38:b3:eb:2a:c7:e9-01-ef00/config/battery expression: Item.val = Attr.val; --> 84
18:52:41:350 TY_DATA_REPORT: seq 122, dpid: 0x0F, type: 0x02, length: 4, val: 84
18:52:41:351 TY_DATA_REPORT: seq 122, dpid: 0x0F, type: 0x02, length: 4, val: 84
18:52:41:353 TY_DATA_REPORT: seq 122, dpid: 0x0F, type: 0x02, length: 4, val: 84
18:52:41:354 TY_DATA_REPORT: seq 122, dpid: 0x0F, type: 0x02, length: 4, val: 84
18:52:41:583 TY_DATA_REPORT: seq 123, dpid: 0x0B, type: 0x01, length: 1, val: 0
[... cut ...]
18:33:38:451 ZCL attribute report 0x60A423FFFE118350 for cluster: 0x0006, ep: 0x0B, frame control: 0x18, mfcode: 0x0000 
18:34:59:640 GW firmware version is up to date: 0x26780700
18:36:28:690 ZCL attribute report 0xEC1BBDFFFEAED1F8 for cluster: 0x0006, ep: 0x0B, frame control: 0x18, mfcode: 0x0000 
18:38:41:322 ZCL attribute report 0x60A423FFFE118350 for cluster: 0x0006, ep: 0x0B, frame control: 0x18, mfcode: 0x0000 
18:38:59:637 GW firmware version is up to date: 0x26780700
18:39:30:248 ZCL attribute report 0xEC1BBDFFFEAED1F8 for cluster: 0x0006, ep: 0x0B, frame control: 0x18, mfcode: 0x0000 
```

Bernhard-Bodensee, 2023-02-16:
I did the above without pairing again - just restarted deconz. Is that sufficient?
From my understanding a DDF change is active when either "Hot reload" is pressed or after restart of deconz.
I hate to run into problems of many renumberings and running out of numbers - if that is a real problem?

Bernhard-Bodensee, 2023-02-16:
I think I got the solution, in addition to setting value you need to change default to 1 - then it works.
I am testing now... found no problem so far.

```
          "name": "state/fire",
          "awake": true,
          "read": {
            "fn": "none"
          },
          "parse": {
            "dpid": 1,
            "eval": "Item.val = (Attr.val == 0);",
            "fn": "tuya"
          },
          "default": true
        },
```

Bernhard-Bodensee, 2023-02-16:
I think we entered our comments at the same time. 
Problem solved as I mentioned above with
Attr.val == 0
and
default": true

Strange, normal people would think 1 if fire - but in fact for tuya 0 seems to be fie alarm.
Regardsless of my understanding:
I tested that solution with the 2 devices I got. 
When I hit test, alarm goes on and is displayed on all Home Apps as Notification (iPhone, iPad, Mac).
Homebridge still has that warning as mentioned but that can be disregarded.
Phoscon shows the correct values in API-Info and alerts - when test is pressed it shows both, fire and test are true.
When the device is ling it shows both as false.

Thanks a lot for your great support. Greetings from the Lake of Constance

BabaIsYou, 2023-02-16:
> in fact for tuya 0 seems to be fie alarm.

Yes for this kind of device it seems to be that way internaly, but for HA systems,  fire race is when fire attribute is true. That's why I think that default to true seems very weird.

Bernhard-Bodensee, 2023-02-16:
I will try your settings tomorrow and get back to you.
```
Could you try this other formula and keep "default": false ?
"eval": "Item.val = (Attr.val == 0 ? true : false);",
```

Bernhard-Bodensee, 2023-02-17:
works. No problems found. Tested essential funktions (idle, alarm, battery change), everything fine.

Looks probably better than default =true
If I understand your fix it looks like a workaround for the stupid device reaction ? if Attr.val is false set to true!
Let me know if you need any more analysis data.

BabaIsYou, 2023-02-17:
Many thx 🙏 
PR updated