dresden-elektronik/deconz-rest-plugin #6737: Lidl HG06338 still switches all plugs on at once

programmieramt, 2023-02-10:
## Describe the bug

## Steps to reproduce the behavior

## Expected behavior

## Environment

## Additional context
it is this device:
https://www.zigbee2mqtt.io/devices/HG06338.html
-->
It is also covered by this bug: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5493 but it was closed

BabaIsYou, 2023-02-10:
And what more do you expect by opening this query than what is already covered by #5493 ?

programmieramt, 2023-02-10:
> And what more do you expect by opening this query than what is already covered by #5493 ?

mmmh the other one is closed and I found no way to reopen it. So I assume no one will work on it.

Mimiix, 2023-02-10:
> > And what more do you expect by opening this query than what is already covered by #5493 ?
> 
> mmmh the other one is closed and I found no way to reopen it. So I assume no one will work on it. We can close this one if you offer a way to reopen the old one.

You did what we always ask users to do: Ping in there and ask for a reopen.

However, you can't expect that we reply within 3 hours.

Closing.

Mimiix, 2023-02-10:
> thank for the answer. I was not aware that I post in a closed issue someone will be notified. Sorry for that.
> 
> Thanks for the quick answer anyway.

No worries :)

I get emails of everything. I normally check a few times a day.

closed 2023-02-10 by Mimiix