dresden-elektronik/deconz-rest-plugin #6742: Aubess Human Presence Sensor mmWave _TZE200_v6ossqfy

BabaIsYou, 2023-02-11:
## Device

- Product name: Aubess/Tuya ZigBee Human Presence Sensor mmWave Radar Detection PIR Motion Sensor [Link](https://www.aliexpress.com/item/1005004697597000.html?spm=a2g0o.order_list.order_list_main.5.21ef1802aOXaEb)
- Manufacturer: _TZE200_v6ossqfy
- Model identifier: TS0601
- Device type : Please remove all unrelated device types. 
  - Sensor
[image]

## Screenshots
[image]

### Basic
[image]

### Groups
[image]
[image]
[image]

### Scenes
[image]

BabaIsYou, 2023-02-11:
Found some information for this device, with other clones BTW. Then trying following DDF. But not quite sure about : 
-> dp 105 is functionnal, have to test
-> dt for dp 103, don't remeber dt for boolean ;-)
```
{
  "schema": "devcap1.schema.json",
  "manufacturername": ["_TZE200_v6ossqfy","_TZE200_mx6u6l4y","_TZE200_0u3bj3rc"],
  "modelid": ["TS0601","TS0601","TS0601"]
  "product": "2.4GHz mmWave radar w/LED",
  "sleeper": false,
  "status": "Silver",
    "subdevices": [
    {
      "type": "$TYPE_PRESENCE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0xef00"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "refresh.interval": 86400,
          "read": {
            "at": "0x0001",
            "cl": "0x0000",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0001",
            "cl": "0x0000",
            "ep": 1,
            "fn": "zcl",
            "script": "tuya_swversion.js"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/delay",
          "description": "Shows the presence duration in minutes",
          "read": {
            "fn": "none"
          },
          "parse": {
            "dpid": 101,
            "eval": "Item.val = Attr.val;",
            "fn": "tuya"
          },
          "default": 1
        },
        {
          "name": "config/duration",
          "description": "Shows the duration of absence in minutes",
          "read": {
            "fn": "none"
          },
          "parse": {
            "dpid": 102,
            "eval": "Item.val = Attr.val;",
            "fn": "tuya"
          }
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "config/ledindication",
          "description": "Turn the onboard LED On or Off",
          "read": {
            "fn": "none"
          },
          "write": {
            "dpid": 103,
            "dt": "0x2b",
            "eval": "Item.val;",
            "fn": "tuya"
          },
          "parse": {
            "dpid": 103,
            "eval": "Item.val = Attr.val;",
            "fn": "tuya"
          },
          "default": 0
        },
        {
          "name": "config/triggerdistance",
          "description": "Far detection sensitivity",
          "read": {
            "fn": "none"
          },
          "write": {
            "dpid": 105,
            "dt": "0x2b",
            "eval": "Far = ['0.5','1','1.5','2','2.5','3','3.5','4','4.5','5']; Attr.val = Math.max(0,Far.indexOf((Item.val-0.5)*2);",
            "fn": "tuya"
          },
          "parse": {
            "dpid": 105,
            "eval": "Item.val = String((Attr.val * 0.5)  + 0.5);",
            "fn": "tuya"
          }
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/presence",
          "read": {
            "fn": "none"
          },
          "parse": {
            "dpid": 1,
            "eval": "Item.val = Attr.val;",
            "fn": "tuya"
          },
          "default": false
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0xEF00"
    }
  ]
}
```

closed 2023-02-20 by BabaIsYou

BabaIsYou, 2023-03-01:
Just to let other people now ... This device is so flooding my Zigbee network that I have to unplug it ;-(

gorynychzmey, 2023-03-14:
Is it ok that device is recognised as Smart Plug and shows no any sensor in web-interface?

BabaIsYou, 2023-03-14:
It only means that DDF is probably not loaded.

gorynychzmey, 2023-03-14:
On pairing I see for short time that device is recognised as a presence (seems to be that DDF is loaded), but after that it turns into Smart Plug.
It's also possible, that the device is malfunctioning, because I haven't seen yet that the light turns red - it's always green.

BabaIsYou, 2023-03-15:
Just have to check "Silver"  to pick up these level of not "fully tested" DDF
[image]

I don't know any other way of resetting the device than unplug it (lost the led configuration) or pressing reset button to re-pair.