dresden-elektronik/deconz-rest-plugin #6743: Lonsonho 16A plug  _TZ3000_dksbtrzs

wvuyk, 2023-02-11:
## Device

- Product name: Lonsonho 16A plug
- Manufacturer:  _TZ3000_dksbtrzs
- Model identifier: TS011F
- Device type : Please remove all unrelated device types. 
  - Light (plug)
  - Sensor (power/consumption/voltage)
 
Light/Plug infor is in the JSON. The sensor information for power is missing.

## Screenshots

[image]

[image]

Basic info part 1
[image]
Basic info part 2
[image]

Identify
[image]

Groups
[image]

Scenes
[image]

On/Off
[image]

Simple Metering 
[image]

Electrical Measurement part 1
[image]

Electrical Measurement part 2
[image]

E000 Unknown and empty
E001 Unknown and empty

OTAU
[image]

BabaIsYou, 2023-02-11:
Try to create a DDF file from this [DDF](https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/lonsonho/_TZ3000_okaz9tjs_plug.json) one after changing theses lines 
```
  "manufacturername": "_TZ3000_okaz9tjs",
  "modelid": "TS011F",
  "product": "Smart US Plug 20A Power Monitor",
```
By these : 

```
  "manufacturername": ["_TZ3000_okaz9tjs", " _TZ3000_dksbtrzs "],
  "modelid": ["TS011F", "TS011F"],
  "product": "Smart US Plug Power Monitor",
```

wvuyk, 2023-02-12:
I tested it and could not get it at work first. Once I replaced the existing file in the devices\lonsonho directory it started working for me. This was on a Windows installation, does the temporary install not work for Windows?  Net very importatn, the file works!

This what I am having at this moment (also changed the product name as this plug is for the EU:

`{
  "schema": "devcap1.schema.json",
  "manufacturername": ["_TZ3000_okaz9tjs", " _TZ3000_dksbtrzs "],
  "modelid": ["TS011F", "TS011F"],
  "product": "Smart Plug Power Monitor",
  "vendor": "Lonsonho",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_SMART_PLUG",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "refresh.interval": 86400,
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "../tuya/tuya_swversion.js"},
          "read": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert",
          "default": "none"
        },
        {
          "name": "state/on",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_POWER_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
[... 210 lines cut ...]
        {
          "at": "0x0505",
          "dt": "0x21",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        },
        {
          "at": "0x0508",
          "dt": "0x21",
          "min": 1,
          "max": 300,
          "change": "0x00000064"
        },
        {
          "at": "0x050B",
          "dt": "0x29",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        }
      ]
    }
  ]
}`

BabaIsYou, 2023-02-12:
> This was on a Windows installation, does the temporary install not work for Windows?
It’s supposed to work also… may be not in the right directory ?

BTW if the DDF works could you do a PR to be included in a next release ? Also I can do it for you.

wvuyk, 2023-02-12:
If you could do this, it would be a great help... I do not really know mucht about the PR's I have to confess :-)

Thanks,
Wim

BabaIsYou, 2023-02-12:
I’ll do it in a few hours ;-)

BabaIsYou, 2023-02-12:
Done !

closed 2023-03-14 by github-actions[bot]