dresden-elektronik/deconz-rest-plugin #6747: Smart Plug Gen2

Jacksonbm1, 2023-02-14:
## Device

I'm specifically trying to get power monitoring values

- Product name: Smart Plug Gen2
- Manufacturer: Third Reality, Inc
- Model identifier: 3RSP02028BZ
- Device type : Please remove all unrelated device types. 
  - Light
  - Other:  Not fully sure, I didn't see instructions on where to get this, but it's a smart plug

## Screenshots

  Required screenshots:
[image]

### Basic

[image]

### Identify

[image]

### Groups

[image]

[image]

### Scenes

[image]
[image]
[image]
[image]

### On/Off

[image]

[image]

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

[image]

[image]

[image]

BabaIsYou, 2023-02-15:
Could you please redo the screenshots after clicking on [read] button on each to gather values and supported attributes ?

Jacksonbm1, 2023-02-15:
oops:

Basic
[image]

Identify:

[image]

Groups:

[image]

Scenes:

[image]

On/Off

[image]

(Rest of image)
[image]

Electrical Measurement:

[image]
(rest of image)
[image]

OTAU
[image]
(Rest of image)
[image]

Smanar, 2023-02-15:
Hello can you try this DDF ? Di you know how to use them ?

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "Third Reality, Inc",
  "modelid": "3RSP02028BZ",
  "vendor": "Third Reality",
  "product": "Smart plug",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_PLUGIN_UNIT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert"
        },
        {
          "name": "state/on",
          "refresh.interval": 360
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_POWER_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0b04"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "none"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/current",
          "refresh.interval": 360
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/power",
          "read": {
            "at": "0x050b",
            "cl": "0x0b04",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x050b",
            "cl": "0x0b04",
            "ep": 1,
            "eval": "Item.val = Attr.val / 10"
          },
          "refresh.interval": 360
        },
        {
          "name": "state/voltage",
          "read": {
            "at": "0x0505",
            "cl": "0x0b04",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0505",
            "cl": "0x0b04",
            "ep": 1,
            "eval": "Item.val = Attr.val / 10"
          },
          "refresh.interval": 360
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0006",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 1,
          "max": 300
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0B04",
      "report": [
        {
          "at": "0x0505",
          "dt": "0x21",
          "min": 10,
          "max": 300,
          "change": "0x00000001"
        },
        {
          "at": "0x0508",
          "dt": "0x21",
          "min": 10,
          "max": 300,
          "change": "0x00000064"
        },
        {
          "at": "0x050B",
          "dt": "0x29",
          "min": 10,
          "max": 300,
          "change": "0x00000001"
        }
      ]
    }
  ]
}
```

The tension is strange 124 V ?
It seem this device don't support consumption ?

BabaIsYou, 2023-02-15:
I guess that voltage is more 1204 / 10 (attr 0x0601) then 120.4 ;-)

SwoopX, 2023-02-15:
Haven't we had the issue here that the comma in the manufacturer name wasn't playing nicely (data is read into a string list with comma as separator)? Could be that the `, Inc` must be skipped?

BabaIsYou, 2023-02-16:
> Could be that the `, Inc` must be skipped?

We already have it an already existing DDF from this manufacturer : [https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/third_reality/3RMS16BZ_motion_sensor.json](https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/third_reality/3RMS16BZ_motion_sensor.json)

Jacksonbm1, 2023-02-16:
"Hello can you try this DDF ? Di you know how to use them ?"

Sorry, I don't! Could you link documentation on what to do here?

Smanar, 2023-02-16:
> Haven't we had the issue here that the comma in the manufacturer name wasn't playing nicely (data is read into a string list with comma as separator)? Could be that the `, Inc` must be skipped?

It happen if you use the editor if you set "Third Reality, Inc" as manufacture name you will in DDF ["Third Reality", "Inc"], but no problem if you use a text editor.

>Sorry, I don't! Could you link documentation on what to do here?

What if your OS ? To resume you just need to create a text file called what_youy_want.json in a "devices" folder.

> All DDF provided directly with deCONZ typically reside in /usr/share/deCONZ/devices/ on a Linux system and are loaded first. However, files residing in the home directory of the user running deCONZ (e.g. /home/<DECONZUSER>/.local/share/dresden-elektronik/deCONZ/devices) will override the pre-packaged files to allow users to amend and keep their own files if desired.

For classic OS
If you have HA > https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-home-assistant-using-text-editor/1839
If you have the offical docker, there is too a persistant folder.

Jacksonbm1, 2023-02-17:
Ok I think I understand, so Just create a json file in my the deconz/devices directory? Does it have to have any special naming convention? What do I do after that? I haven't noticed any changes to the device.

BabaIsYou, 2023-02-17:
> For the tension 120 V is still strange

It's a US plug then voltage is 120V ;-)

BabaIsYou, 2023-02-17:
> Ok I think I understand, so Just create a json file in my the deconz/devices directory? Does it have to have any special naming convention? What do I do after that? I haven't noticed any changes to the device.

Under `/home//.local/share/dresden-elektronik/deCONZ/devices` or under `/usr/share/deCONZ/devices/` then delete and re-pair the device or restart DeCONZ. If DDF is correctly loaded values will be populated accordingly, and you could see modify DDF from DDF editor

sadpixel, 2023-02-25:
I own several of these new devices, and can confirm that the DDF above is working! =D

What's the process for getting it included by default in future releases of Deconz?

Smanar, 2023-02-25:
So convertion are fine ?
You can make a PR with the DDF or ask for someone make it (here) if you can't.

Jacksonbm1, 2023-02-26:
I've been trying to make sure my network is stable rn, but thanks for doing that @sadpixel. If the PR makes it through, does it automatically get updated on our end or do we need to get a software update pushed through?

Smanar, 2023-02-27:
If the PR is done (and valided) it will be in official version so nothing more to do.
I just need to know if you want to make the PR yourself or if you want I make it ^^ ?

Jacksonbm1, 2023-02-27:
Interestingly, it looks like the values are being populated already. Does that mean it's been submitted as a PR already? if not, it would be super helpful if you could!

Smanar, 2023-02-27:
>Interestingly, it looks like the values are being populated already

It's because you are using the DDF ?

PR done https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6768

closed 2023-03-02 by manup