dresden-elektronik/deconz-rest-plugin #6749: Wattle Door Lock SPL Smart

Gyldenlow, 2023-02-15:
- Product name: Wattle Door Lock SPL Smart
- Manufacturer: Home Control AS
- Model identifier: HC-SLM-1
- Device type :
  - Lock

## Screenshots

### Basic

[image]

### Power Configuration

[image]

### Identify 

[image]

### Door lock 
[image]

### Node Info 

[image]

Gyldenlow, 2023-02-16:
@Smanar This is funny, I found that exact code you posted on another issue only 10 mins ago, and got my lock working 2 mins before reading this! Happy to report that it works :) For any future readers - after hot reloading you need to remove and re-insert the batteries for it to update in Home Assistant.

Smanar, 2023-02-16:
Lol, so the working mode is ok, can make a PR ? (or if you want to make it ?)

Gyldenlow, 2023-02-16:
Yes, you can make a PR. I am very new to this (and I dont know how to do it)

Smanar, 2023-02-16:
Done https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6754

closed 2023-02-20 by manup