dresden-elektronik/deconz-rest-plugin #6750: 3A Smart Home LXN-2S27LX1.0 Shutter Control

Febus123, 2023-02-15:
## Device

- Product name: 3A Shutter Control
- Manufacturer: 3A Smart Home DE
- Model identifier: LXN-2S27LX1.0
- Device type : Please remove all unrelated device types. 
  - Other: Window Covering Device (Generic Device Type "Unspecified")

## Screenshots

[image]
[image]

### Basic
[image]
[image]

### Identify
[image]

### Alarms
./.
### Device Temperature
./.
### Groups
[image]
[image]
[image]

### Scenes
[image]
[image]
[image]
[image]

### On/Off
[image]
[image]
[image]

### Level Control
[image]
[image]

### Color Control
./.
### Simple Metering
./.
### Diagnostics
./.
### Other clusters that are not mentioned above
[image]
[image]

### OTAU
[image]
[image]

ebaauw, 2023-02-15:
It’s a controller, not a window covering device.  It looks like a wired switch with two outputs (presumably for up and down).  Since the device types on the endpoints are _On/Off Light_, I assume the _Level Control_ clusters are there merely to support _Move to Level (with On/Off)_, and they probably only report _Current Level_ at 0 or 254.  If so, it should be exposed as two on /off light `/lights` resources, without `state/bri`, rather than as two dimmable lights.

You cannot expose this as a window covering device, as it doesn’t seem to report the current position of the shutter.

Smanar, 2023-02-16:
But it s not working natively, used a dimmable light ?

Febus123, 2023-02-16:
It works when using the dimmable lights (as soon as one is above 0% it goes either up or down). I was hoping to get this somehow converted into a shadowing device. I managed to do this via FHEM (which calculates positions via timing) but would prefer to do this directly via deCONZ…

closed 2023-03-18 by github-actions[bot]