dresden-elektronik/deconz-rest-plugin #6755: Müller-Licht tint Remote control

twobeass, 2023-02-16:
## Device

- Product name: https://forum.phoscon.de/t/read-me-device-request-read-me/23/2 
- Manufacturer: _TZ3000_bdbb0fon
- Model identifier: TS1001
- Device type : Please remove all unrelated device types. 
  - Remote
 

## Screenshots

[image]
[image]

### Basic
[image]

### Power Configuration
[image]

### Identify
[image]

### Groups
[image]
[image]

### Color Control
[image]
[image]

### OTAU
[image]

BabaIsYou, 2023-02-21:
Just to be sure ... Did you click on "read" before Power Configuration screenshot ?

twobeass, 2023-02-21:
I just double checked it and pressed „read power descriptor“ again! Still the same values… can i somehow provide more information ??

twobeass, 2023-02-21:
Unwanted close on mobile device, sorry…

BabaIsYou, 2023-02-21:
Screenshots seems more "normal" now ! It's normal that for a battery device you could have to read several time or play with buttons while you read to awake the device reporting.
Can you add also ZLL commissioning cluster screenshot ? I don't see it.
Thx

twobeass, 2023-02-21:
[image]

sure i thought these informations are not necessary as there is a count of 0 in that cluster...

twobeass, 2023-02-21:
I just updated my first post, as i recognized that there were a lot of values not written properly.. Thanks @BabaIsYou for pointing me in the right direction...

As i searched the logs i found out that there were missing button mappings:
`22:27:43:526 [INFO] - No button map for: TS1001, broadcast to: 0x0000, endpoint: 0x01, cluster: BASIC (0x0000), command: 0x02, payload: 05402001, zclSeq: 72`
i am hoping that this makes it easier to create a proper ddf.. if i could provide more information to you devs, just let me know...

closed 2023-03-23 by github-actions[bot]