dresden-elektronik/deconz-rest-plugin #6756: Ikea Vindstyrka

vinyli85, 2023-02-18:
## Device

- Product name: Ikea Vindstyrka
- Manufacturer: Ikea of Sweden
- Model identifier: VINDSTYRKA
- Device type :
  - Sensor

## Screenshots

### Basic
[image]

### Identify
[image]

### Groups
[image]
[image]

### Temperature Measurment
[image]

### Relative humidity measurment
[image]

### Ikea specific
[image]

### UNKNOWN (3 times)
Empty Cluster Info

ebaauw, 2023-02-18:
Where did you get this device?  I would love to get my hands on one, but It's only expected to come out earliest in April, here in the Netherlands.

Supporting the temperature and humidity sensors is rather straighforward, but we need to reverse engineer the custom clusters for air quality support.  For that we would have to pair the device to an IKEA gateway, and sniff the Zigbee traffic to see what commands the gateway sends on what app actions.  The Startvind uses cluster 0xFC7D for Air Purfier control and reports PM2.5 density on attribute 0x0004 on that cluster.  deCONZ computes the quality label ("good", "fair", ...) from that.  The Vindstyrka should report both PM2.5 density and TVOC level, if I read the documentation correctly.  I wonder if it reports the coloured label or if we need to compute that ourselves.  And if there's separate labels for PM2.5 vs TVOC.

vinyli85, 2023-02-18:
Hi Erik,

got mine today at an Ikea store in Germany. There were maybe 2-3 left. 
The colored label is „fixed“. That means the color is like printed on the display and only two arrows on the side show the quality. The TVOc looks similar, only an arrow which shows the level. 

I don‘t have an Ikea hub to help you out with more data.

ebaauw, 2023-02-18:
I checked ikea.de, but couldn’t find it there.  Only in the physical store?

vinyli85, 2023-02-18:
Yes, it’s only available local not online. You want one? :)

ebaauw, 2023-02-18:
Yes, would appreciate if you could get one for me.  Happy to pay for it and for shipping costs.  Please PM me on Discord to exchange address and bank account details.

Cluster 0x042A has actually been defined in revision 8 of the ZCL spec, as _PM2.5 Measurement_, but it hasn't been defined yet in `general.xml`.  It should have the same attributes as _Relative Humidity Measurement_: _Measured Value_, _Min Measured Value_, _Max Measured Value_, and _Tolerance_.  And we need to check whether it reports density (in µg/m³), or level (in ppb).

The client _Fan Control_ cluster suggests you can pair it to a Starkvind or similar to control the fan speed based on this sensor.  Wonder if the IKEA app will set this up.

sadpixel, 2023-02-24:
I’m also looking forward to grabbing this device when it launches in the US. 😊 I think it’s trickling out into some European stores at the moment, but seems to sell out immediately. As already noted, the German website doesn’t list it yet — but I found the product page live on the Hungarian website (though currently sold out of all stock, as of this post):

https://www.ikea.com/hu/hu/p/vindstyrka-legminoseg-erzekelo-okos-eszkoez-40498234/

vinyli85, 2023-02-25:
I created a DDF with some basic data that works fine for me right now. But I think there is some more data we can get from that sensor. Maybe there are some mistakes due to just guessing the right values for AirQuality

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "$MF_IKEA",
  "modelid": "VINDSTYRKA",
  "vendor": "IKEA of Sweden",
  "product": "VINDSTYRKA",
  "sleeper": false,
  "status": "Gold",
  "path": "/devices/vindstyrka.json",
  "subdevices": [
    {
      "type": "$TYPE_TEMPERATURE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0402"
      ],
      "items": [
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/temperature",
          "refresh.interval": 15,
          "read": {
            "at": "0x0000",
            "cl": "0x0402",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0000",
            "cl": "0x0402",
            "ep": 1,
            "eval": "Item.val = Attr.val",
            "fn": "zcl"
          },
          "default": 0
        }
      ]
    },
    {
      "type": "$TYPE_HUMIDITY_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0405"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/offset",
          "description": "Relative offset to the main measured value.",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/humidity",
          "refresh.interval": 15,
          "read": {
            "at": "0x0000",
            "cl": "0x0405",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0000",
            "cl": "0x0405",
            "ep": 1,
            "eval": "Item.val = Attr.val",
            "fn": "zcl"
          },
          "default": 0
        },
        {
          "name": "state/lastupdated"
        }
      ]
    },
    {
      "type": "$TYPE_AIR_QUALITY_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0000"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/airquality",
          "parse": {
            "fn": "numtostr",
            "op": "le",
            "srcitem": "state/airqualityppb",
            "to": [
              "10",
              "excellent",
              "20",
              "good",
              "25",
              "moderate",
              "50.poor.75",
              "unhealthy",
              "65535",
              "out of scale"
            ]
          },
          "default": "unknown"
        },
        {
          "name": "state/airqualityppb",
          "refresh.interval": 15,
          "read": {
            "at": "0x0000",
            "cl": "0x042a",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0000",
            "cl": "0x042a",
            "ep": 1,
            "eval": "Item.val = Attr.val",
            "fn": "zcl"
          }
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ]
}
```

ebaauw, 2023-03-03:
Got the Vindstyrka, thanks to @vinyli85.

IKEA app only displays temperature, humidity, and PM2.5.  No apparent settings.  The help text in the app confirms that PM2.5 is in µg/m³ (which the reported values already suggested).  They also mention levels to 12 µg/m³ are healthy, and above 35 µg/m³ unhealthy.  I don't understand how that relates to the colour coding, where 0-35 is Good (green), 35-85 Okay (yellow), and above 85 Poor (red).  The resolution of the temperature and humidity is rather coarse: full °C or %.  Not just on the display, but also as reported over Zigbee.

It's a router, permanently powered by USB.  In can indeed be paired directly with the Startvind according to the documentation, but I haven't tried that.  Pairing to the Dirigera is straightforward.  The hub sends an _Identify_, but that doesn't seem to do too much.  It does request the descriptors, and sends a _Discover Attributes_ for each server cluster (thank you, IKEA!), except 0xFC57.

As mentioned the 0x42a cluster is PM2.5 measurement, as described in v8 of the ZCL spec.  The known clusters show no surprises.  0xFC7C only contains 0xFFFD (the cluster version); 0xFC7E has the same attributes as 0x042A and the other measurement clusters.  Need to read it when paired to deCONZ, to see if it reports TVOC.  The hub sets up attribute reporting for temperature (0/1800/50), humidity (0/1800/100), pm2.5 (5/5/1) and 0xFC7E/0x0000 (5/5/50).  Looking at the reported values, 0xFC7E is something else than PM2.5 density.  I'm seeing values between 95 and 105.

It looks like the device settings (bridghtness of the display backlight, auto off of the display backlight, Celsius vs Fahenheit) are exposed over Zigbee.

ebaauw, 2023-03-03:
[image]
[image]

ebaauw, 2023-03-03:
I think it makes most send to expose the sensor as four resources, corresponding to clusters 0x0402, 0x0405, 0x042a, and 0xfc7e.  So two ZHAAirQuality resources, one with `state/pm2_5`, and one with `state/airqualityppb` (which should have been named `state/tvoc`.  Both have the text `state/airquality`, based on the corresponding quantity.

vinyli85, 2023-03-03:
> I think it makes most send to expose the sensor as four resources

That Sounds good. How did you get that deCONz Shows the cluster names? Will this be a Part of the next Update?

ebaauw, 2023-03-03:
> How did you get that deCONz Shows the cluster names? Will this be a Part of the next Update?

That's the updated to `general.xml`.

ebaauw, 2023-03-03:
Here's the resources created by the DDF:
```
  "81": {
    "config": {
      "offset": 0,
      "on": true,
      "reachable": true
    },
    "etag": "e5eedff5d79e458750bfde7a212b4298",
    "lastannounced": null,
    "lastseen": "2023-03-03T14:09Z",
    "manufacturername": "IKEA of Sweden",
    "modelid": "VINDSTYRKA",
    "name": "Vindstyrka",
    "state": {
      "lastupdated": "2023-03-03T14:06:06.000",
      "temperature": 2000
    },
    "swversion": "1.0.010",
    "type": "ZHATemperature",
    "uniqueid": "94:34:69:ff:fe:bb:f4:92-01-0402"
  },
  "82": {
    "config": {
      "offset": 0,
      "on": true,
      "reachable": true
    },
    "etag": "0b9c743033012a1ede70c51d970f6e50",
    "lastannounced": null,
    "lastseen": "2023-03-03T14:09Z",
    "manufacturername": "IKEA of Sweden",
    "modelid": "VINDSTYRKA",
    "name": "Vindstyrka",
    "state": {
      "humidity": 4500,
      "lastupdated": "2023-03-03T14:09:02.439"
    },
    "swversion": "1.0.010",
    "type": "ZHAHumidity",
    "uniqueid": "94:34:69:ff:fe:bb:f4:92-01-0405"
  },
  "83": {
    "config": {
      "on": true,
      "reachable": true
    },
    "etag": "57f1e7b49b3f861b008116ca48813591",
    "lastannounced": null,
    "lastseen": "2023-03-03T14:09Z",
    "manufacturername": "IKEA of Sweden",
    "modelid": "VINDSTYRKA",
    "name": "Vindstyrka",
    "state": {
      "airquality": "excellent",
      "lastupdated": "2023-03-03T14:09:05.398",
      "pm2_5": 5
    },
    "swversion": "1.0.010",
    "type": "ZHAAirQuality",
    "uniqueid": "94:34:69:ff:fe:bb:f4:92-01-042a"
  },
  "84": {
    "config": {
      "on": true,
      "reachable": true
    },
    "etag": "967608f8c57fea04be1d0d96545b8552",
    "lastannounced": null,
    "lastseen": "2023-03-03T14:09Z",
    "manufacturername": "IKEA of Sweden",
    "modelid": "VINDSTYRKA",
    "name": "Vindstyrka",
    "state": {
      "airquality": "good",
      "airqualityppb": 96,
      "lastupdated": "2023-03-03T14:10:11.632"
    },
    "swversion": "1.0.010",
    "type": "ZHAAirQuality",
    "uniqueid": "94:34:69:ff:fe:bb:f4:92-01-fc7e"
  }
```

ebaauw, 2023-03-03:
Looks like it's now listed on the Dutch IKEA website and available in the store and onllne.
https://www.ikea.com/nl/nl/p/vindstyrka-luchtkwaliteitsensor-smart-00498231/

vinyli85, 2023-03-04:
> Here's the resources created by the DDF:

I don‘t get the airquality PM2.5 value, and both quality Indexes for tVOC and PM2.5 with your created DDF.

ebaauw, 2023-03-04:
> I don‘t get the airquality PM2.5 value

Did you install `general.xml`?

> quality Indexes for tVOC and PM2.5

Did you complie the API plugin with the `numtostr` fix?

vinyli85, 2023-03-04:
> Did you install `general.xml`?
 Yes.
> Did you complie the API plugin with the `numtostr` fix?
No, no idea how to do so

ebaauw, 2023-03-04:
> No, no idea how to do so

Then you'll have to wait for the next version.  As mentioned above, until then `airquality` will remian `null`.

For the PM2.5 value: can you read the cluster from the GUI?  If that causes the API to report `state/pm2_5`, there's probably an issue with the binding and/or attribute reporting of the cluster.  Not sure, but I suspect deCONZ might have an issue with setting up attribute reporting for `float` values.

closed 2023-03-04 by manup

javiges, 2023-04-04:
I have the vindstyrka device, how can I integrate it into ha?

Meyckoo, 2023-04-10:
Hi @ebaauw,

I need your assistant if it's possible :).

I have install my Ikea Vindstyrka but i can't show the values  pm2_5 and tVOC on HomeAssistant. While i can read the value on deCONZ.

deCONZ Configuration with the values : 
[image]

[image]

Values i can see on HA
[image]

[image]

Do you have any idea what I need to do?

Best Regards,

Mimiix, 2023-04-10:
> Hi @ebaauw,
>
> I need your assistant if it's possible :).
>
> I have install my Ikea Vindstyrka but i can't show the values pm2_5 and tVOC on HomeAssistant. While i can read the value on deCONZ.
>
> deCONZ [...]

This is a question for the forums, not here.

Locking this up.