dresden-elektronik/deconz-rest-plugin #6757: Xiaomi Aqara relay controller lumi.relay.c2acn01 stopped working

monpelaud, 2023-02-19:
## Describe the bug

Hi,
For months I have been using an aqara llkzmk11lm wireless relay with the deCONZ plugin for HA.
Everything was working fine until deCONZ restarted after an HA core update.
Now this device appears in deCONZ but not in Phoscon and is no longer visible in HA.
In deCONZ UI the device is connected and everything works fine. No errors in deCONZ logs, communication with the RaspBee II module seems to work well.
But I no longer see the device in Phoscon and HA.
## Steps to reproduce the behavior

See discussion with @Smanar on Phoscon forum:
https://forum.phoscon.de/t/device-appear-in-deconz-but-not-in-phoscon/3298
## Expected behavior

DDF file proposed by @Smanar in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6512#issuecomment-1426980875 fixe this issue.
## Screenshots

[image]
[image]
## Environment

 - Host system: (Raspberry Pi / PC / NAS)
 - Running method: (Raspbian / Ubuntu / Home Assistent deCONZ Add-on / Marthoc Docker container / Windows / Virtual Machine)
 - Firmware version: (26xxyy00)
 - deCONZ version: (2.xx.yy)
 - Device: (ConBee I / ConBee II / RaspBee I / RaspBee II)
 - Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? If so: Which?

## deCONZ Logs

## Additional context

Smanar, 2023-02-19:
From the last logs from the phoscon forum, all seem working except current and power, perhaps you haven't load during your tests ?

And as you are seing event, it mean you have it in the API, so the device need to be visible on HA (Phoscon it's random).

monpelaud, 2023-02-19:
The device is visible in Phoscon:
[image]
deCONZ Electrical_Measurement without load:
[image]
deconZ Electrical_Measurement in load:
[image]
It's the same !!!!

SwoopX, 2023-02-19:
As this is obviously unrelated to the REST API, I'll close this issue. Feel free to debate any Phoscon or 3rd party related topics/issues in the forum.

closed 2023-02-19 by SwoopX

monpelaud, 2023-02-19:
@SwoopX, I don't understand why you closed my issue, if it's not a REST API issue what is it? Phoscon stops working and it works again with a new DDF file!
It's no longer works because REST API seems to no longer show data to Phoscon or HA. With the new DDF file REST API show again data to Phoscon and it's work again.
Is DDF file not in the code of deconz-rest-plugin?