dresden-elektronik/deconz-rest-plugin #6759: Device name

EduardoHerreraM, 2023-02-21:
## Device

- Product name: Tuya ZigBee Smart Knob
- Manufacturer:  _TZ3000_qja6nq5z
- Model identifier: TS004F
- Device type :
  - Switch
  - Dimmer

## Screenshots

### Basic

[image]

### Identify

[image]

### Groups

[image]

### On/Off

[image]

### Color Control

Huge list (50 items)

### Power Configuration

[image]

## Comments

I've followed the instractions that appear here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6031. 

I've modified the DDF, the current one is as follows:

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_qja6nq5z",
  "modelid": "TS004F",
  "vendor": "Tuya",
  "product": "Smart Knob",
  "sleeper": true,
  "status": "Gold",
  "path": "/devices/backup/tuya/_TZ3000_qja6nq5z_smart_knob.json",
  "subdevices": [
    {
      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x1000"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "refresh.interval": 7300,
          "read": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1
          },
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2"
          },
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/angle",
          "default": 0
        },
        {
          "name": "state/bri",
          "refresh.interval": 5,
          "read": {
            "fn": "none"
          },
          "parse": {
            "cl": "0x0008",
            "ep": 1,
            "fn": "zcl",
            "script": "_TZ3000_qja6nq5z_smart_knob_buttons.js"
          }
        },
        {
          "name": "state/buttonevent"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/on",
          "awake": true,
          "refresh.interval": 5,
          "read": {
            "fn": "none"
          },
          "parse": {
            "cl": "0x0006",
            "ep": 1,
            "fn": "zcl",
            "script": "_TZ3000_qja6nq5z_smart_knob_buttons.js"
          }
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "groupcast",
      "config.group": 0,
      "src.ep": 1,
      "cl": "0x0006"
    },
    {
      "bind": "groupcast",
      "config.group": 0,
      "src.ep": 1,
      "cl": "0x0008"
    },
    {
      "bind": "groupcast",
      "config.group": 0,
      "src.ep": 1,
      "cl": "0x0300"
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 300,
          "max": 600,
          "change": "0x00000001"
        }
      ]
    }
  ]
}
```

I don't have the original DDF as I didn't make a copy (whoops).

The logs shows this kind of logs:

```
[INFO] - No button map for: TS004F, broadcast to: 0x0000, endpoint: 0x01, cluster: ONOFF (0x0006), command: TOGGLE (0x02), payload: None, zclSeq: 34
[INFO] - No button map for: TS004F, broadcast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), command: STEP (0x02), payload: 010D0100, zclSeq: 35
```

Buttons mapping:

Mode ? (I don't know how to identify the mode)

- Single Click -> cluster: ONOFF (0x0006), command: TOGGLE (0x02), payload: None
- Double Click -> Nothing
- Rotate Clockwise -> cluster: LEVEL_CONTROL (0x0008), command: STEP (0x02), payload: 000D0100
- Rotate Counter-Clockwise -> cluster: LEVEL_CONTROL (0x0008), command: STEP (0x02), payload: 01190100
- Push and Rotate Clockwise ->cluster: COLOR_CONTROL (0x0300), command: STEP_COLOR_TEMPERATURE (0x4C), payload: 01340002009900F401
- Push and Rotate Counter-Clockwise -> cluster: COLOR_CONTROL (0x0300), command: STEP_COLOR_TEMPERATURE (0x4C), payload: 03560002009900F401

Mode ?
- Single Click -> cluster: ONOFF (0x0006), command: LIDL (0xFD), payload: 00
- Double Click -> cluster: ONOFF (0x0006), command: LIDL (0xFD), payload: 01
- Rotate Clockwise -> cluster: ONOFF (0x0006), command: 0xFC, payload: 00
- Rotate Counter-Clockwise -> cluster: ONOFF (0x0006), command: 0xFC, payload: 01
- Push and Rotate Clockwise ->Nothing
- Push and Rotate Counter-Clockwise -> Noting

BabaIsYou, 2023-02-22:
I imagine that _TZ3000_qja6nq5z_smart_knob_buttons.js is the same than https://github.com/manup/deconz-rest-plugin/blob/2563c6666e0ebca11359942e506eebc4c130cc1c/devices/tuya/_TZ3000_4fjiwweb_smart_knob_buttons.js ?

EduardoHerreraM, 2023-02-23:
> I imagine that _TZ3000_qja6nq5z_smart_knob_buttons.js is the same than https://github.com/manup/deconz-rest-plugin/blob/2563c6666e0ebca11359942e506eebc4c130cc1c/devices/tuya/_TZ3000_4fjiwweb_smart_knob_buttons.js ?

Yes!! Sorry I didn't clarify that. Is the same, but renamed

EduardoHerreraM, 2023-03-04:
I tried using this DDF, still no luck:

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_qja6nq5z",
  "modelid": "TS004F",
  "product": "TS004F",
  "sleeper": false,
  "status": "Gold",
  "path": "/devices/tuya/_TZ3000_qja6nq5z_smart_knob.json",
  "subdevices": [
    {
      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x1000"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "default": 0
        },
        {
          "name": "config/group",
          "default": "auto"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/buttonevent",
          "description": "The last received button event."
        },
        {
          "name": "state/lastupdated"
        }
      ]
[... 172 lines cut ...]
        },
        "XIAOYAN": {
            "COMMAND_0": 0,
            "ATTRIBUTE_REPORT": 10,
            "COMMAND_20": 20
        }
	},
	"maps": {
        "TuyaSmartKnob": {
            "vendor": "Tuya",
            "doc": "Smart Knob",
            "modelids": ["_TZ3000_qja6nq5z"],
            "map": [
                [1, "0x01", "ONOFF", "TOGGLE", "0", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "Toggle"],
                [1, "0x01", "ONOFF", "0xfd", "0", "S_BUTTON_1", "S_BUTTON_ACTION_DOUBLE_PRESS", "Double"],
                [1, "0x01", "LEVEL_CONTROL", "STEP", "0", "S_BUTTON_2", "S_BUTTON_ACTION_SHORT_RELEASED", "Level Clockwise"],
                [1, "0x01", "LEVEL_CONTROL", "STEP", "1", "S_BUTTON_3", "S_BUTTON_ACTION_SHORT_RELEASED", "Level Anticlockwise"],
                [1, "0x01", "COLOR_CONTROL", "STEP_COLOR_TEMPERATURE", "1", "S_BUTTON_4", "S_BUTTON_ACTION_SHORT_RELEASED", "Color Clockwise"],
                [1, "0x01", "COLOR_CONTROL", "STEP_COLOR_TEMPERATURE", "3", "S_BUTTON_5", "S_BUTTON_ACTION_SHORT_RELEASED", "Color Anticlockwise"]
            ]
        }
    }
}
```
Events are shown in HA as stated in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6031, but not really usefull, I don't know how to take advante of those events.

harrydaws, 2023-07-14:
hi i just bought this and have the same problem. have you managed to figure it out?

EduardoHerreraM, 2023-07-14:
> hi i just bought this and have the same problem. have you managed to figure it out?

Stoped using deconz. I'm using zigbee2mqtt instead :)

harrydaws, 2023-07-16:
do you recommend switching and how long did it take you

spudje, 2023-07-24:
Would love to see this thing natively supported as well. Deconz is starting to lag behind to other Zigbee stacks. Not really looking forward to migrate to something else, with the amount of devices I have.

LukeDefeo, 2023-10-25:
I just added this device and I don't see any events coming through in deconz_events in home assistant

coyote9999, 2023-10-31:
I'm also starting to wonder why Deconz doesn't support such great devices as this Knob, and also why there is no response to device requests

Mimiix, 2023-12-15:
> I'm also starting to wonder why Deconz doesn't support such great devices as this Knob, and also why there is no response to device requests

Because those are done by volunteers :) 

Happy to re open. Perhaps @Smanar wants to help out?

Smanar, 2023-12-15:
It seem the DDF is partially working https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6759#issuecomment-1454736791

There is a similar one here https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6242

closed 2024-01-15 by github-actions[bot]