dresden-elektronik/deconz-rest-plugin #6761: Left and Right Buttons of Ikea STYRBAR remote control are not working

Gloomyeye, 2023-02-23:
## Describe the bug

The left and right buttons of the Ikea STYRBAR remote control are not working with Home Assistant. No deconz_event will be redirected to Home Assistant for these buttons. The up and down buttons are working fine.

## Steps to reproduce the behavior

  1. Activate the deconz_event listening in Home Assistant
  2. Use the left or right button of the Ikea STYRBAR remote control 
  3. Check the event log under Home Assistant development tools

## Expected behavior

The left and right buttons are working like the up and down buttons and throw deconz_events to Home Assistant.

## Screenshots

[image]

[image]

## Environment

 - Host system: Raspberry Pi 4
 - Running method: Raspbian with Docker container for Deconz and Home Assistant
 - Firmware version: 26780700
 - deCONZ version: 2.20.01
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs

## Additional context

vbaros, 2023-03-12:
I've had the same issue for some time. First update everything to the latest versions if not already. There was also a fix in home assistant for mixing up the events, but that's unrelated to this issue.

The problem is that if you hold left or right button for loner time both left s nd right buttons stop sending events. To fix it you have to do it again. Just hold one of the left or right buttons for more than 5 seconds then try again. It should work.

Gloomyeye, 2023-03-22:
Hi, I updated Deconz and Home Assistant to the latest version.
If I hold the left or the right button for longer than 5 seconds, then a deconz event 1002 occurs in Home Assistant... I would expect a 2002 or 4002 event.

Gloomyeye, 2023-04-13:
This is still an issue.

Gloomyeye, 2023-04-17:
I deleted the remote from phoscon / deconz and readded it. Issue with the left and right button still exists. They are not working. With a long press on these two buttons, a deconz event 1002 will be sent to Home Assistant, no 3002 or 4002 event.
@ebaauw Maybe you can help me :-) You helped with the issue #4572 which was also regarding the button events.

Gloomyeye, 2023-04-17:
This is the DDF binding of my configuration (I did't change the binding from my side):
```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "$MF_IKEA",
  "modelid": "Remote Control N2",
  "product": "Remote Control N2",
  "sleeper": false,
  "status": "Draft",
  "subdevices": [
    {
      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x1000"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0820",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x1000"
        ],
        "out": [
          "0x0006",
          "0x0008",
          "0x0005"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "description": "The current device battery level in 0&ndash;100&thinsp;%.",
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "cppsrc": "power_configuration.cpp:193",
            "ep": 1,
            "eval": "Item.val = Attr.val"
          },
          "default": 0
        },
        {
          "name": "config/group",
          "description": "comma seperated list of groups which a device controls."
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/buttonevent"
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0000"
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 300,
          "max": 2700,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "groupcast",
      "config.group": 127,
      "src.ep": 1,
      "cl": "0x0006"
    }
  ]
}
```

ebaauw, 2023-04-17:
Looking at my notes in the other issue, you want to do the group cast binding first, and then the unicast binding.  Not sure why you’re binding the _Basic_ cluster?  Also, you might need group cast bindings for _Level Control_ (up/down buttons) and/or _Scenes_ (left/right buttons).  My notes say it didn’t need this, but IKEA kinda have a track record changing this logic with each firmware upgrade.

Best run deCONZ with APS logging enabled at level 2, to see which commands the remote sends.  Alternatively use a Zigbee sniffer.  And double-check that those commands are matched in `button_map.json`.

I do have the device lying around somewhere, but not currently online.  And too many other priorities before finding the time to look into this.

TilBumble, 2023-04-17:
All four Buttons of my STYRBAR ("Remote Control N2") were initially working:
I assigned the function of toggling two light groups via Hue Essentials to the "Left" and "Right" buttons.

Out of boredom/curiosity/stupidity i updated the firmware of the STYRBAR to the version 2.4.5. Since then, i could only receive 1002 and 2002 events from the "dim up" and "dim down" - buttons.
Only these events were generated and only from these buttons. No "release after long press" or "double press".

The Ikea-changelog states:
New features and changes in Accessories:
STYRBAR Remote control (V-2.4.5)
◆ Low battery indication fixed.
◆ Power reporting fixes.
◆ Fixed issue with recall scene command that is not coming through when initiating remote sync.

So, i think @ebaauw is right in assuming that something broke with the new Ikea-firmware.

Holding one of the left/right buttons for more than 5 seconds did NOT solve this issue for me.

While writing this text, i re-paired the STYRBAR by enabling permit join in deconz and pressing the connect-button on the back of the remote.
After this, no button events at all were generated! :-(
I re-paired via Phoscon again and now i have the same situation as @Gloomyeye (1002 and 2002 from "dim up/down" and 1002 after long press of "left/right" (without release))

No lines in the debug view are immediately generated when pressing a button (APS_L2), but I saw 
22:02:11:279 Delay APS request id: 12 to 0xF381, profile: 0x0000 cluster: 0x0021 node already has busy 2
coming in blocks of several lines. I don't know if this is related.

I am using deconz v2.21.2-beta, but i paired the STYRBAR with a previous version.
I am not using a DDF for this device.

This is not an urgent problem for me and I hope, i could possibly help in in solving the problem.
Many thanks to whoever can fix this.

ebaauw, 2023-04-17:
> No lines in the debug view are immediately generated when pressing a button

Then deCONZ isn't receiving anything, most likely because the remote isn't sending anything, due to lack of bindings (from the client clusters).

The delay APS request means deCONZ cannot send a command to the device (is 0xF381 indeed the NWK address of the remote?), presumably because it's asleep.  Profile 0x0000 is ZDP, so this could very well be the bind request (I don't know by heart what command ZDP cluster 0x0021 is).

I've seen a number of IKEA devices that timeout very quickly after joining the network, and turn off their radio too early.  If deCONZ wasn't successful creating a client binding in that time, there's no way to wake the device, as it thinks it doesn't need to send anything on a button press.  In other words: you want to create the client group cast bindings before the unicast for reporting the battery percentage.

I know of no remedy, other than restting the device and retrying to pair it.  You might try and bind the _On/Off_ cluster manually in the GUI, directly after pairing (when the node appears, or the existing node blinks blue indicating traffic).

TilBumble, 2023-04-17:
Thank you for your quick response.
Yes, 0xF381 is the NWK address of the remote.
Just to be clear "No lines in the debug view are immediately generated when pressing a button" referred to APS_L2.

Debug level "Info" shows:
22:36:58:268 [INFO] - Button 2002 - Remote Control N2, unicast to: 0x0000, endpoint: 0x01, cluster: ONOFF (0x0006), action: On, payload: None, zclSeq: 15
22:36:58:273 rule event /sensors/71/state/lastupdated: 0 -> 0
22:36:58:275 trigger rule 227 - Rule TOOGLE_OFF
22:36:58:278 rule event /groups/29/state/any_on: 0 -> 29
22:37:00:245 Bind response success for 0xb4e3f9fffecb5ecd ep: 0x01 cluster: 0x0000
22:37:10:203 [INFO] - Button 1002 - Remote Control N2, unicast to: 0x0000, endpoint: 0x01, cluster: ONOFF (0x0006), action: On, payload: None, zclSeq: 16
22:37:10:207 rule event /sensors/71/state/lastupdated: 0 -> 0
22:37:10:209 trigger rule 223 - Rule TOOGLE_OFF
22:37:10:445 ZCL attribute report 0x54EF4410002F9731 for cluster: 0x0006, ep: 0x02, frame control: 0x18, mfcode: 0x0000 
and the blue indicator shows up in deconz, but only for dim up/down.

I will delete the rules in Hue Essentials and delete and pair the sensor again and see what happens.

ebaauw, 2023-04-17:
Did I mix my debug levels?  Maybe it’s ZCL Level 2?  I want to see the `asdu` on the line after the message, with the payload of the command.

Does it still show a single endpoint with the clusters as in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4572#issue-830391066?

TilBumble, 2023-04-19:
Sorry for the late response.
I am grateful for your help.

Is this the information, yióu were asking for?
[image]
No line was generated when i pressed a button.

I did not delete and re-pair the remote yet.
First i will install 2.22.0-beta (Although i do not expect this to repair my STYRBAR)
I am still astonished that the firmware update of the remote broke it.

Gloomyeye, 2023-04-23:
In addition to the left/right issue, I noticed that the battery level is not reporting anymore after repairing the device.

Gloomyeye, 2023-05-15:
This is still an issue with the latest IKEA firmware and Deconz version.

adrspa, 2023-05-16:
Hi all,

I can confirm that I'm seeing the exact same issue as Gloomyeye. 

My setup is:
ConBee 2 connected to RPi 4, no extension cable. 
Deconz v2.21.2
HA 2023.5.3
Supervisor 2023.04.1
Operating System 10.1

IKEA Strybar running 2.4.5 FW. 

I initially had my Styrbar connected to the IKEA DIRIGERA hub, which of course had it updated to the latest FW version. Pressing the dim up and dim down buttons generate these entries in the deconz debug view:
[image]

Whereas holding in the left and right buttons generate the same button no, 1002:
[image]

I'm happy to do tests to support having this issue resolved. I'm really keen for this to work as the Styrbar remotes would be very useful if all buttons worked!

Best regards,
Adrian

adrspa, 2023-05-17:
Hi

I saw the same issue reported for zigbee2mqtt and they seem to have fixed the issue in their environment. Maybe their solution could help out here?

https://github.com/Koenkk/zigbee2mqtt/issues/15725

ebaauw, 2023-05-17:
Found my Styrbar at the bottom of some drawer.  Pairing with deCONZ v2.22.0-beta succceeds at first attempt, and I see button events for all four buttons.  The Styrbar's _SW Build ID_ is 1.0.024;  file version 0x00010024.  `ikea-ota-download.py` doesn't seem to download any newer firmware.  After pairing it to the DIRIGERIA hub, the IKEA Home app initially reported the firmware as: "1.0.024 Up-to-date", but now as "1.0.024".  The app gives no clue, but the sniffer reveals that it's updating its firmware to 0x02040005.  To be continued.

ebaauw, 2023-05-17:
Updated to v2.4.5.  Still connected to the Dirigera, the sniffer reveals it's still sending the same unicast commands to the coordinator as under the previous firmware version (see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4572#issuecomment-804398474), except that on Hold, the 0x07 is replaced by a _Recall Scene_ of 0x0000/0x00.

Pairing with deCONZ again succeeds on the first attempt.  The Styrbar  now reports _Battery Percentage Remaining_ (0x0001/0x0021) in units of 0.5%.  IKEA are really upping their adherence to the standard, nice!

Indeed, no button events on the _Left_ and _Right_ buttons, but also not on holding the _DimUp_ and _DimDown_ buttons.  Bindings from the _Level Control_ and _Scenes_ clusters are missing.  The device seems to ignore any groupcast bindings, but after adding unicast bindings for these clusters to the coordinator, I get the same button events as with the previous firmware.  Only 2003 seems to be missing.  The device actually sends _Stop_, but the button map looks for _Stop (with On/Off)_.  I suppose IKEA corrected that in the firmware update as well?

I removed the group cast bindings - the Styrbar won't send group casts, no matter what I tried.  Bloody GUI doesn't want to send _Binding Table Request_, but after restarting the deCONZ it now does.  The code also created unicast bindings from 0x0019 (_OTAU_) and 0x0020 (_Poll Control_).  Next, there's unicast bindings from 0x0000 (_Basic_), 0x0001 (_Power Configuration_), 0x0006 (_On/Off_), _Level Control_ (0x0008), _Scenes_ (0x0005).

So it looks like we need to adopt `buttonmap.json` and create a DDF without `config/group` and with unicast bindings from 0x0000, 0x0001 (with reporting for 0x0021), 0x0006, 0x0008, and 0x0005, and with the default `parse` for `config/battery`.

adrspa, 2023-05-17:
Hi ebaauw

That is excellent progress! Thank you for spending time in this issue, much appreciated! I had actually just concocted a plan to go to ikea and buy new Styrbar and NEVER pair them with Dirigera and hope they are on an older FW version still compatible with the existing implementation. Maybe I don't have to do that now... 

Best regards 
Adrian

closed 2023-05-21 by manup