dresden-elektronik/deconz-rest-plugin #6762: TMZ-02 Mini a.k.a Tmz02L-16A-W ( Manu: _TZ3000_txpirhfq, Model: TS0011 )

elpablillo, 2023-02-26:
## Device

- Product name: TMZ-02 Mini / Tmz02L-16A-W
- Manufacturer: _TZ3000_txpirhfq
- Model identifier: TS0011
- Device type :
  - Switch

## Screenshots

### Node Info panel
[image]

### Basic
[image]

### Identify
[image]

### Groups
[image]
[image]

### Scenes
[image]
[image]
[image]

### On/Off
[image]
[image]

Smanar, 2023-02-26:
Hello, can you try this DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": ["_TZ3000_txpirhfq","_TZ3000_qmi1cfuq", "_TZ3000_ji4araar"],
  "modelid": ["TS0011", "TS0011", "TS0011"],
  "product": "1 gang switch module without neutral",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_OUTPUT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",                   
          "refresh.interval": 86400,
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on"
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ]
}
```

elpablillo, 2023-02-26:
Hi, I use HassOS... so that part to copy and paste it is complicated... but I was able to create using DDF aditor. 
[image]
I have removed the other manufacturername "_TZ3000_qmi1cfuq", "_TZ3000_ji4araar", because in the preview, it didn't look same as you. 
[image]

After the "Hot reload" and reloading the integration... the device is shown into HA 
[image]

Smanar, 2023-02-26:
And all is working ? No need binding/report ?

(Just for information, you have a special folder to put DDF on deconz plugin docker on HA, but I m agree, not easy)

elpablillo, 2023-02-26:
Yes, all is working... this kind of switch it is quite simple just on/off, no need binding. I've just done a hardware reboot of the Raspberry... it's still woking 😀 and taking the .json from the folder: 
[image]

Smanar, 2023-02-26:
Ok, have made a PR with the 2 clones https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6764

closed 2023-03-02 by manup