dresden-elektronik/deconz-rest-plugin #6765: ShenZhen_Homa 0301 HMS823

crx-michi, 2023-02-26:
## Device

- Product name: HMS823-Z-XX-WP-OTA-3.0
- Manufacturer: ShenZhen_Homa
- Model identifier: HOMA0301
- Device type :
  - Sensor

## Screenshots

### Basic
[image]

### Identify
[image]

### Alarms

### Device Temperature

### Groups

### Scenes

[image]

### On/Off
[image]

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above
[image]

[HMS823-Z-XX-WP-OTA-3 0 specification.pdf](https://github.com/dresden-elektronik/deconz-rest-plugin/files/10834469/HMS823-Z-XX-WP-OTA-3.0.specification.pdf)

BabaIsYou, 2023-02-27:
Hi. Can you please complete cluster screenshots with Groups and Illuminance ? 
And be sure to click on "read" button before taking screenshots.

Thx

crx-michi, 2023-02-28:
Hi,

thanks for your answer, her are the screenshots from group and illuminance, but its only a motion sensor without illuminance measurement.

hope this helps - Thanks a lot!
[image]
[image]

BabaIsYou, 2023-03-02:
Seems weird that a motion sensor doesn't contains relative cluster and that it contains illuminance measurement if it's not that type of sensor. In addition, hope that it's not using 0xFFEE specific cluster.

Can you try this DDF BTW, but would be surprise it works. May be we need some log after that.

{
  "schema": "devcap1.schema.json",
  "manufacturername": "ShenZhen_Homa",
  "modelid": "HOMA0301",
  "vendor": "ShenZhen HOMA Technology Co",
  "product": "HMS823",
  "status": "Silver",
  "sleeper": true,
  "subdevices": [
    {
      "type": "$TYPE_LIGHT_LEVEL_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0400"
      ],
      "items": [
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "config/tholddark"
        },
        {
          "name": "config/tholdoffset"
        },
        {
          "name": "state/lightlevel"
        },
        {
          "name": "config/battery"
        },
        {
          "name": "state/dark"
        },
        {
          "name": "state/daylight"
        },
        {
          "name": "state/lux"
        },
        {
          "name": "state/lastupdated"
        }
      ]
    },
    {
      "type": "$TYPE_PRESENCE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0406"
      ],
      "items": [
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/presence"
        },
        {
          "name": "config/battery"
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ]
}

closed 2023-03-31 by github-actions[bot]