dresden-elektronik/deconz-rest-plugin #6766: Associate Off and transitiontime

OUARZA, 2023-02-26:
Hello

I see that it is possible to associate an Off command with the transitiontime parameter on a group but not on a light directly.
Can you fix the problem so that we can use the Off command and the transitiontime parameter on light-only commands?
Thanks in advance.

Work:
[image]

Don't work:
[image]

Deconz version : 2.20.01 

Mathieu

OUARZA, 2023-03-28:
Not solved

ebaauw, 2023-03-28:
Make sure the light is exposed using a DDF that includes the `cap/bri/move_with_onoff` item.  Note that this item is not visible in the REST API.

Background: The Zigbee _Off_ command doesn't take a _Transition Time_.  So you need to specify both `"on": false` and `"bri": 0`, to send _Move to Level (with On/Off)_ command, which does honour _Transition Time_.  However, not all lights implement this command correctly.  We used to whitelist the lights that do, hardcoded in C++, falling back to _Off_ for those that don't.  Recently the whitelist has changed to specifying the above mentioned capability in the DDF.  When that capability is present, the REST API plugin will send _Move to Level (with On/Off)_; when not, it will send a plain _Off_.

As we have no logic to analyse what types of lights are in a group, setting a group action of "on": false` and `"bri": 0` results in a _Move to Level (with On/Off)_ unconditionally, accepting the risk that some lights might not turn off.

OUARZA, 2023-04-01:
Hello @ebaauw ,
I cannot find the reference of my Philips lamps under the file /usr/share/deCONZ/devices/philips
LST002 and LCT007
Mathieu

ebaauw, 2023-04-01:
LST002 is in `light_zll_C.json` and LCT007 is in `light_zll_B.json`.  These should be included in v2.20.1, and should be used if you have _DDF Mode_ set to _Basic_ with _Gold_ enabled.  Select the light node in the GUI, right click and select _Edit DDF_: the popup window should show Status Gold.

OUARZA, 2023-04-01:
@ebaauw 
I already have the correct version.
[image]

I looked in the 2 files, they have the right parameter
[image]

Mathieu

ebaauw, 2023-04-01:
Then they should turn off slowly over 20s when PUTting `{ "on": false, "bri": 0, "transitiontime" 200 }`.

What's your setting for _DDF_ in the _Control_ panel?

OUARZA, 2023-04-01:
Where can I find this information?
I am under Jeedom with the Deconz / Phoscon plugin.

ebaauw, 2023-04-01:
In the _Control_ panel in the deCONZ GUI.

OUARZA, 2023-04-01:
I don't have deconz GUI

OUARZA, 2023-04-05:
@ebaauw Can't check from command line?

Mimiix, 2023-04-05:
> @ebaauw Can't check from command line?

Nope.

closed 2023-05-05 by github-actions[bot]