dresden-elektronik/deconz-rest-plugin #6767: v2.20.1-beta - syslog error: deCONZ[2619]: "No carrier" and database locked

TungstenE2, 2023-02-27:
I get following error in syslog: deCONZ[2619]: "No carrier"

Raspberry Pi 2b 

Raspbian Bullseye

deCONZ v2.20.1-beta

```
Feb 27 11:19:15 raspberrypi deCONZ-WIFI2.sh[19059]: Error: database is locked
Feb 27 11:19:15 raspberrypi deCONZ-WIFI2.sh[19060]: Error: database is locked
Feb 27 11:19:24 raspberrypi deCONZ[2619]: "No carrier"
Feb 27 11:19:34 raspberrypi deCONZ[2619]: "No carrier"
Feb 27 11:19:44 raspberrypi deCONZ[2619]: "No carrier"
Feb 27 11:19:54 raspberrypi deCONZ[2619]: "No carrier"
Feb 27 11:20:04 raspberrypi deCONZ[2619]: "No carrier"
Feb 27 11:20:14 raspberrypi deCONZ[2619]: "No carrier"
Feb 27 11:20:24 raspberrypi deCONZ[2619]: "No carrier"
Feb 27 11:20:34 raspberrypi deCONZ[2619]: "No carrier"
Feb 27 11:20:44 raspberrypi deCONZ[2619]: "No carrier"
Feb 27 11:20:54 raspberrypi deCONZ[2619]: "No carrier"
Feb 27 11:21:04 raspberrypi deCONZ[2619]: "No carrier"
Feb 27 11:21:14 raspberrypi deCONZ[2619]: "No carrier"
Feb 27 11:21:24 raspberrypi deCONZ[2619]: "No carrier"
Feb 27 11:21:34 raspberrypi deCONZ[2619]: "No carrier"
Feb 27 11:21:44 raspberrypi deCONZ[2619]: "No carrier"
Feb 27 11:21:54 raspberrypi deCONZ[2619]: "No carrier"
Feb 27 11:22:04 raspberrypi deCONZ[2619]: "No carrier"
Feb 27 11:22:14 raspberrypi deCONZ[2619]: "No carrier"
Feb 27 11:22:24 raspberrypi deCONZ[2619]: "No carrier"
Feb 27 11:22:34 raspberrypi deCONZ[2619]: "No carrier"
Feb 27 11:22:44 raspberrypi deCONZ[2619]: "No carrier"
Feb 27 11:22:54 raspberrypi deCONZ[2619]: "No carrier"
Feb 27 11:23:04 raspberrypi deCONZ[2619]: "No carrier"
Feb 27 11:23:14 raspberrypi deCONZ[2619]: "No carrier"
Feb 27 11:23:24 raspberrypi deCONZ[2619]: "No carrier"
Feb 27 11:23:34 raspberrypi deCONZ[2619]: "No carrier"
```

[image]

Any advice to solve this?

Mimiix, 2023-02-27:
Hi

As this is not a bug report as described in #5113, i am closing this issue.

Please open a topic on the forums.

Kind regards,
Dennis

closed 2023-02-27 by Mimiix

TungstenE2, 2023-02-27:
@Mimiix all need details are provided, why are you closing this? If you are no longer supporting Github all topics should be closed. I searched the internet and forum before I opened the ticket.

Mimiix, 2023-02-27:
> @Mimiix all need details are provided, why are you closing this? If you are no longer supporting Github all topics should be closed. I searched the internet and forum before I opened the ticket.

User questions are handled on the forums , not GitHub.