dresden-elektronik/deconz-rest-plugin #6772: _TZ3000_yxmafzmd - TS0002 switched on and off as a group

jhubig, 2023-03-02:
## Describe the bug
I have bought the following switch to be able to control two light switches: https://de.aliexpress.com/item/1005003051876791.html (AVATTO Tuya Smart Zigbee Switch Module 2 Way Control 2gang N-ZWSM01).

[image]

The pairing through Phoscon was easy and device was recognized as:

Hersteller: _TZ3000_yxmafzmd
Modell: TS0002

With the physical switches everything works normal and I can switch on the lights independently from each other. In the Phoscon Interface, if I switch on one of the lights, then both are switched on. Same happens when I switch on one light through the the REST-API (I know it should be the same happening when I switch it on the Phoscon UI).

```
curl -s http://192.168.178.226:80/api/26097226B6/lights/13

{"etag":"-","hascolor":false,"lastannounced":"2023-03-02T12:27:11Z","lastseen":"2023-03-02T15:58Z","manufacturername":"_TZ3000_yxmafzmd","modelid":"TS0002","name":"On/Off light 13","state":{"alert":"none","on":false,"reachable":true},"swversion":null,"type":"On/Off light","uniqueid":"-"}%

curl -s http://192.168.178.226:80/api/26097226B6/lights/14

{"etag":"-","hascolor":false,"lastannounced":"2023-03-02T12:27:11Z","lastseen":"2023-03-02T15:58Z","manufacturername":"_TZ3000_yxmafzmd","modelid":"TS0002","name":"On/Off light 14","state":{"alert":"none","on":false,"reachable":true},"swversion":null,"type":"On/Off light","uniqueid":"-"}%

curl -X PUT -d '{"on":true}' -s "http://192.168.178.226:80/api/26097226B6/lights/13/state"
[{"success":{"/lights/13/state/on":true}}]%

curl -s http://192.168.178.226:80/api/26097226B6/lights/13

{"etag":"-","hascolor":false,"lastannounced":"2023-03-02T12:27:11Z","lastseen":"2023-03-02T15:58Z","manufacturername":"_TZ3000_yxmafzmd","modelid":"TS0002","name":"On/Off light 13","state":{"alert":"none","on":true,"reachable":true},"swversion":null,"type":"On/Off light","uniqueid":"-"}%

curl -s http://192.168.178.226:80/api/26097226B6/lights/14

{"etag":"-","hascolor":false,"lastannounced":"2023-03-02T12:27:11Z","lastseen":"2023-03-02T15:58Z","manufacturername":"_TZ3000_yxmafzmd","modelid":"TS0002","name":"On/Off light 14","state":{"alert":"none","on":true,"reachable":true},"swversion":null,"type":"On/Off light","uniqueid":"-"}%
```

## Steps to reproduce the behavior
see above

## Environment

 - Host system: (Proxmox)
 - Running method: (Debian with deCONZ as Virtual Machine)
 - Firmware version: (26780700)
 - deCONZ version: (2.17.01 / 1.7.2022)
 - Device: (ConBee II)
 - Do you use an USB extension cable: (yes)
 - Is there any other USB or serial devices connected to the host system? No

BabaIsYou, 2023-03-02:
Hi,

Seems to be a device request and not a bug. Could you please follow this model ?

Thx

jhubig, 2023-03-02:
Hi,

okay. The device seems to exist, or? But not behaving how it should.

That's why had flagged it initially as a bug. So means I need to rewrite with the information you have linked although the device is already available? Sorry if I'm wrong. First issue I write in this rep.

Thanks. 

Cheers,
Johannes

BabaIsYou, 2023-03-02:
The device seems to have been partially recognized but the fact that you couldn't command each line individually via REST_API (and then Phoscon) means that it certainly need more stuff to be descibed to REST-API to address the situation, and perhaps to provide a Tuya magic sequence to unlock it.

This should be done with a DDF file (I hope so) but to be sure you have to provide further informations using the r`equest device` template I referenced.

jhubig, 2023-03-02:
Okay. Copied. Is it possible to provide the information without having access to deconz-Gui?

BabaIsYou, 2023-03-02:
> Okay. Copied. Is it possible to provide the information without having access to deconz-Gui

I afraid not ...

jhubig, 2023-03-02:
Okay. I have my deconz instance running on a Proxmox Virtual machine. I'll try to setup GNOME and VNC to be able to access the deconz-GUI.

Mimiix, 2023-03-02:
Hi,

As this is not a issue as described in #5113 i am closing this issue.

Please open a proper device request.

Kind regards,

closed 2023-03-02 by Mimiix